moveit/moveit #3453: Contrller is talking too long execute trajectory

SARAWIN-BK-7, 2023-06-21:
### Description
Hello, I'm having some problem, not sure about move group or not, I tried testing on ur5 yesterday it worked. But today there was a problem.
### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
[image]

simonschmeisser, 2023-06-21:
I cannot read anything, please provide error messages as text (ideally formatted as code block).

SARAWIN-BK-7, 2023-06-21:
Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 10.964647 seconds). Stopping trajectory.

SARAWIN-BK-7, 2023-06-22:
> I cannot read anything, please provide error messages as text (ideally formatted as code block).

`Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 10.964647 seconds). Stopping trajectory.`

zp2546265641, 2024-07-11:
> > I cannot read anything, please provide error messages as text (ideally formatted as code block).
> 
> This problem was caused by my "sudo apt update" `Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 10.964647 seconds). Stopping trajectory.`
hello, today i also meet this too, yesterday is normal. why sudo apt update cause it? and what do you update?

rhaschke, 2024-07-11:
The problem is surely not related to `apt update`. Your controller is simply too slow to follow the trajectory.
If you don't care about the exact timing of your trajectory, you can disable duration monitoring via the ROS parameter `execution_duration_monitoring = false`, which can also be adapted via dynamic reconfigure.

zp2546265641, 2024-07-11:
> The problem is surely not related to `apt update`. Your controller is simply too slow to follow the trajectory.
> If you don't care about the exact timing of your trajectory, you can disable duration [...]
hello,hello, and which file context can i  set the parameter? i am using ur5 ros_driver