moveit/moveit #3455: Generating IKfast plugin for UR5 robot from URDF file throws error raise InverseKinematicsError(u'wrong rate %f > 0!'%wrongrate)

devvaibhav455, 2023-06-21:
## Background

I am trying to generate IKFast plugin for moveit using the tutorial [https://ros-planning.github.io/moveit_tutorials/doc/ikfast/ikfast_tutorial.html](url)

## Instructions
1) Follow the tutorial up until to the point where we ned to convert the xacro file to urdf
2) I used below command to generate URDF files.
Input xacro file: [https://github.com/ros-industrial/universal_robot/blob/melodic-devel/ur_description/urdf/ur5.xacro]() 
`rosrun xacro xacro -o /tmp/ur5.urdf /home/dev/Documents/catkin_ws/src/universal_robot/ur_description/urdf/ur5.xacro`
After running this command, output file looks like this 
[ur5.urdf.txt](https://github.com/ros-planning/moveit/files/11826500/ur5.urdf.txt)
 (I just the file extension from urdf to txt for the sake of uploading it to git) which seems OK to me.

3) Go to `~/Documents/catkin_ws/src` and run `rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D /tmp/$MYROBOT_NAME.urdf manipulator base_link tool0`

This fails with the below output. It is just a part of the error as it is very big. For full error, please see here 
[ur5_ikfast_error.txt](https://github.com/ros-planning/moveit/files/11826491/ur5_ikfast_error.txt)

> Try converting urdf to Collada directly
> 
> Document successfully written to /tmp/ikfast.1WgcWV/robot.dae
> Building docker image
> Sending build context to Docker daemon  19.05MB
> Step 1/3 : FROM personalrobotics/ros-openrave
>  ---> 3470beb7a703
> Step 2/3 : RUN apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654 &&     apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116 &&     apt-get update &&     apt-get install -y --no-install-recommends python-pip build-essential liblapack-dev ros-indigo-collada-urdf &&     apt-get clean && rm -rf /var/lib/apt/lists/*
>  ---> Using cache
>  ---> c7b477b3f884
> Step 3/3 : RUN pip install git+https://github.com/sympy/sympy.git@sympy-0.7.1
>  ---> Using cache
>  ---> 7ee34f093d05
> Successfully built 7ee34f093d05
> Successfully tagged fixed-openrave:latest
> Successfully built docker image.
> Running openrave0.9.py --database inversekinematics --robot=/input/wrapper.xml --iktype=Transform6D --iktests=1000
> could not import scipy.optimize.leastsq
> 2023-06-21 21:41:11,885 openrave [WARN] [plugindatabase.h:577 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name fcl_, interface collisionchecker
> openravepy.databases.inversekinematics: generate, Generating inverse kinematics for manip ur5_robot: Transform6D [0, 1, 2, 3, 4, 5], precision=8, maxcasedepth=3 (this might take up to 10 min)
> openravepy.databases.inversekinematics: generate, creating ik file /input/.openrave/kinematics.e0ef30b8b8c4665fe9ec33866bd3c85a/ikfast0x10000049.Transform6D.0_1_2_3_4_5.cpp
> openravepy.ikfast: forwardKinematicsChain, moved translation [0, 0, 823/10000] to right end
> openravepy.ikfast: forwardKinematicsChain, moved translation [0, 0, 0] to left end
> openravepy.ikfast: forwardKinematicsChain, moved translation on intersecting axis [0, 0, 0] to left
> openravepy.ikfast: generateIkSolver, [[-1, 0, 0, 0],[0, -1, 0, 0],[0, 0, 1, 89159/1000000]]
> openravepy.ikfast: generateIkSolver, [[cos(j0), -sin(j0), 0, 0],[sin(j0), cos(j0), 0, 0],[0, 0, 1, 0]]
> openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, -1, 0],[0, 1, 0, 0]]
> openravepy.ikfast: generateIkSolver, [[cos(j1), -sin(j1), 0, 0],[sin(j1), cos(j1), 0, 0],[0, 0, 1, 0]]
> openravepy.ikfast: generateIkSolver, [[1, 0, 0, -17/40],[0, 1, 0, 0],[0, 0, 1, 0]]
> openravepy.ikfast: generateIkSolver, [[cos(j2), -sin(j2), 0, 0],[sin(j2), cos(j2), 0, 0],[0, 0, 1, 0]]
> openravepy.ikfast: generateIkSolver, [[1, 0, 0, -1569/4000],[0, 1, 0, 0],[0, 0, 1, 2183/20000]]
> openravepy.ikfast: generateIkSolver, [[cos(j3), -sin(j3), 0, 0],[sin(j3), cos(j3), 0, 0],[0, 0, 1, 0]]
> openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, -1, -1893/20000],[0, 1, 0, 0]]
> openravepy.ikfast: generateIkSolver, [[cos(j4), -sin(j4), 0, 0],[sin(j4), cos(j4), 0, 0],[0, 0, 1, 0]]
> openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 0, 1, 0],[0, -1, 0, 0]]
> openravepy.ikfast: generateIkSolver, [[cos(j5), -sin(j5), 0, 0],[sin(j5), cos(j5), 0, 0],[0, 0, 1, 0]]
> openravepy.ikfast: generateIkSolver, [[1, 0, 0, 0],[0, 1, 0, 0],[0, 0, 1, 823/10000]]
> openravepy.ikfast: solveFullIK_6D, ikfast 6d: [j0, j1, j2, j3, j4, j5]
> openravepy.ikfast: iterateThreeNonIntersectingAxes, found 3 consecutive non-intersecting axes links[2:7], vars=[j1, j2, j3]
> openravepy.ikfast: iterateThreeNonIntersectingAxes, found 3 consecutive non-intersecting axes links[4:9], vars=[j2, j3, j4]
> openravepy.ikfast: iterateThreeNonIntersectingAxes, found 3 consecutive non-intersecting axes links[0:5], vars=[j0, j1, j2]
> openravepy.ikfast: iterateThreeNonIntersectingAxes, found 3 consecutive non-intersecting axes links[6:11], vars=[j3, j4, j5]
> openravepy.ikfast: solveFullIK_6D, try first group 0/4
> openravepy.ikfast: buildRaghavanRothEquations, computed in 1.153035s
> openravepy.ikfast: buildRaghavanRothEquations, computed in 0.396517s
> openravepy.ikfast: solveLiWoernleHiller, attempting li/woernle/hiller general ik method
> openravepy.ikfast: solveLiWoernleHiller, allowed indices: [0]
> openravepy.ikfast: solveLiWoernleHiller, assuming equation htj1 is non-zero, dividing by Poly(-py*cj0*htj1 + px*sj0*htj1 - 2183/20000*htj1, cj2, sj2, cj3, sj3, cj0, sj0, cj5, sj5, htj1, domain='QQ[px,py]')
> openravepy.ikfast: solveLiWoernleHiller, factors 1 reduced not unique: -cj0*py + px*sj0 - 2183/20000
> openravepy.ikfast: solveLiWoernleHiller, assuming equation htj1 is non-zero, dividing by Poly(r11*cj0*cj5*htj1 + r10*cj0*sj5*htj1 - r01*sj0*cj5*htj1 - r00*sj0*sj5*htj1, cj2, sj2, cj3, sj3, cj0, sj0, cj5, sj5, htj1, domain='ZZ[r00,r01,r10,r11]')
> openravepy.ikfast: solveLiWoernleHiller, factors 1 reduced not unique: cj0*cj5*r11 + cj0*r10*sj5 - cj5*r01*sj0 - r00*sj0*sj5
> openravepy.ikfast: solveLiWoernleHiller, assuming equation htj1 is non-zero, dividing by Poly(-cj2*cj3*htj1**3 + cj2*cj3*htj1 - 2*cj2*sj3*htj1**2 - 2*sj2*cj3*htj1**2 + sj2*sj3*htj1**3 - sj2*sj3*htj1 - r21*cj5*htj1**3 - r21*cj5*htj1 - r20*sj5*htj1**3 - r20*sj5*htj1, cj2, sj2, cj3, sj3, cj0, sj0, cj5, sj5, htj1, domain='ZZ[r20,r21]')
[... 21 lines cut ...]
> raw ik command: 0.5710601071317917 -0.04468919819207223 0.8196909354186956 -0.02044300007411198 0.7993978318832151 -0.1968094080003497 -0.5676523260790596 0.05908716443249935 0.1866908150473442 0.9794227547323483 -0.07666555347388243 0.03159087178206872 
> 
> 2023-06-21 21:41:47,973 openrave [WARN] [ikfastmodule.cpp:1064 DebugIK] FindIKSolution: No ik solution found, i = 125
> jointvalues=[1.579074459472172, -0.5853106265156733, 0.5008849703467955, -0.8520233001877795, 0, -0.7123342810422724, ]
> ikparameterization="1728053249 0.02464078502603745 -0.7064491274003515 -0.03048984253540604 0.70667731788331 -0.1982425596445072 0.8197313204292865 0.3009359178900994 "
> raw ik command: -0.0006449242171652214 0.008252877561944698 -0.9999657364553557 -0.1982425596445072 0.07790521305290218 -0.9969264024309337 -0.0082780381310028 0.8197313204292865 -0.996960561833785 -0.077907882451418 -2.051038451491433e-10 0.3009359178900994
> 
> openravepy.databases.inversekinematics: testik, success rate: 0.001000, wrong solutions: 0.993000, no solutions: 0.006000, missing solution: 0.001000
> Traceback (most recent call last):
>   File "/usr/bin/openrave0.9.py", line 126, in <module>
>     database.run(args=args)
>   File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1123, in run
>     InverseKinematicsModel.RunFromParser(*args,**kwargs)
>   File "/usr/lib/python2.7/dist-packages/openravepy/_openravepy_0_9/databases/inversekinematics.py", line 1111, in RunFromParser
>     raise InverseKinematicsError(u'wrong rate %f > 0!'%wrongrate)
> openravepy._openravepy_0_9.databases.inversekinematics.InverseKinematicsError: Inverse Kinematics Error
> wrong rate 0.993000 > 0!
> 'docker run --rm --user 1000:1000 -v /tmp/ikfast.1WgcWV:/input --workdir /input -e HOME=/input fixed-openrave:latest openrave0.9.py --database inversekinematics --robot=/input/wrapper.xml --iktype=Transform6D --iktests=1000' failed with exec code 1:

Surprisingly I am able to generate the IKFast plugin for Panda robot from this URDF 
[panda.urdf.txt](https://github.com/ros-planning/moveit/files/11826507/panda.urdf.txt)

I tried comparing these two URDF's, tried to remove transmission, gazebo tags from UR5's urdf but it didn't help.

Does anyone have any idea?

rhaschke, 2023-06-23:
This is a failure in OpenRAVE. Try to reduce precision as stated in the tuorial:

> Often floating point issues arise in converting a URDF file to Collada file, so a script has been created to round all the numbers down to x decimal places in your .dae file. Its probably best if you skip this step initially and see if IKFast can generate a solution with your default values, but if the generator takes longer than, say, an hour, try the following:

```
export IKFAST_PRECISION="5"
cp "$MYROBOT_NAME".dae "$MYROBOT_NAME".backup.dae  # create a backup of your full precision dae.
rosrun moveit_kinematics round_collada_numbers.py "$MYROBOT_NAME".dae "$MYROBOT_NAME".dae "$IKFAST_PRECISION"
```

devvaibhav455, 2023-06-23:
@rhaschke : Thanks for replying. I tried this part of the tutorial as well but it still produces the same error. I came across [this urdf](https://github.com/dairal/ur5-joint-position-control/blob/main/urdf/ur5_jnt_pos_ctrl.urdf) on the internet and this builds successfully. But the plugin generated from this does not seem to work in the rviz (I move the visualization marker, but it doesn't move whereas the default plugin works). So, I tried to compare the difference between this working urdf and my failing urdf. I found that it is the same mostly except for the rpy and xyz values in the `joint` tag. So, I copy-pasted the working xyz, rpy values into my urdf, and the plugin generated as well. But it does not work in moveit demo because I believe that the urdf which `ur5_moveit_config` package uses has different xyz, rpy values from the working urdf. Essentially, I am still at the location where I was initially.

rhaschke, 2023-06-23:
Copy-pasting xyz/rpy fields from one urdf to another usually isn't a good idea. This defines the kinematics of your robot joints and if you change those values, the joint will be differently oriented. I cannot help further as both your urdf and the openrave generator are out of MoveIt's scope.

simonschmeisser, 2023-06-24:
If you are actually using a UR5 there is also an analytical model for it (it ignores the calibration so if you need "high" accuracy it might be not be applicable). See ur_kinematics