moveit/moveit #3460: [move_group-5] process has died. exit code -11

Ecurbliz, 2023-06-26:
Dear Moveit developers,
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Source or Binary build? Source build
* If binary, which release version? 
* If source, which branch? noetic-devel

### Steps to reproduce

build from source and git clone ompl in the same src folder.

cd ../ompl/src/ompl/CMakeList.txt
${CMAKE_INSTALL_LIBDIR}
 
changed to
 
"/opt/ros/noetic/lib/x86_64-linux-gnu"

## This step is to install the library to the source
## After the change, the code should look like this :
# install the library
install(TARGETS ompl
    DESTINATION "/opt/ros/noetic/lib/x86_64-linux-gnu/"

and run this 
cd ~/ws_moveit
 
catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release
 
sudo catkin build
# because it builds into the source, needs sudo

after build, go here
~/ws_moveit/src/ompl/src/ompl/geomatric/planner/rrt

creat new file called xxx.h and new src/xxx.cpp 

simply copy all the code from algorithms rrt and change the name to xxx

now goto this path （~/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src）
modify this file
**planning_context_manager.cpp**

include the header file of xxx (hint: include all the path e.g:</home/usr/ws_moveit/src/ompl/src/ompl/geometric/planners/rrt/xxx.h>

and change **avoid ompl_interface::PlanningContextManager::registerDefaultPlanners()** accordingly

also, change here in the yaml file import our new planner xxx. **panda_moveit_config/config/ompl_planning.yaml**

after all the change, run sudo catkin build again in ~/ws_moveit path

now. **roslaunch panda_moveit_config demo.launch** 

### Expected behaviour

try to use our new planner xxx. it shows in ompl list but can't work .. please help 

### Actual behaviour

expect the planner xxx should work as rrt planner.

### Backtrace or Console output

[move_group-5] process has died [pid 106833, exit code -11, cmd /home/usr/ws_moveit/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/usr/.ros/log/5049d486-142a-11ee-8ed8-e52e0d7856dc/move_group-5.log].
log file: /home/usr/.ros/log/5049d486-142a-11ee-8ed8-e52e0d7856dc/move_group-5*.log

rhaschke, 2023-06-30:
Sorry, but I cannot follow the steps to reproduce your issue? I guess you tried to implement a new OMPL planner, but first by simply copying the RRT code. Why do you need to build OMPL from source for this? Can you reproduce that with the default OMPL installation?
Usually, problems arise if you use headers not matching to the linked library (or linking two different versions of the library).

Ecurbliz, 2023-07-01:
> Sorry, but I cannot follow the steps to reproduce your issue? I guess you tried to implement a new OMPL planner, but first by simply copying the RRT code. Why do you need to build OMPL from source [...]

Hi rhaschke, 

thanks for your reply.
Apologize for the inconvenience, and YES , I'm trying to implement a new OMPL planner and test it in Moveit. 
did you follow the steps and everything works fine? 

i did try the default installation with OMPL, unforturally it doesn't work.
after few tries, the easiest way to reproduce the issue is doing follows...

1. went through this installation:
https://moveit.ros.org/install/source/

2. git clone ompl from main branch in the same catkin workspace/src

3. catkin build again,

4. source the devel/setup.bash  and roslaunch panda_moveit_config demo.launch (nothing is changed)

and the move_group -5 error shows when i try to plan with ompl planners ..... after click plan&excuate

thanks for you help

rhaschke, 2023-07-01:
Rebuilding again (via `catkin build`) with a changed set of sources doesn't work reliable with catkin. While the MoveIt source probably still uses the old, default OMPL installation during the second build, at runtime, the new OMPL libraries will be used. However, they don't match the headers used during build, thus causing the crash.
Always build a catkin workspace from a clean state: `catkin clean`.

Ecurbliz, 2023-07-01:
> Rebuilding again (via `catkin build`) with a changed set of sources doesn't work reliable with catkin. While the MoveIt source probably still uses the old, default OMPL installation during the [...]

Hi rhaschke,

thanks for your reply. what you said is absolutely correct, this issue has been solved.

closed 2023-07-01 by Ecurbliz