moveit/moveit #3463: PlanningRequestAdapterChain merge the added path indices in the wrong order

Hugal31, 2023-06-28:
### Description

When multiple planning request adapters (PRA) add states to the trajectory, the indices of the outer PRA can be incremented if the inner PRA added a state with a lower index. If I understand this correctly, this is wrong because the inner PRA indices should be incremented and not the opposite.

### Your environment

* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Binary build 1.1.12-1focal.20230524.152908

### Steps to reproduce

* Have at least two PRAs.
* Have the inner one (the latter in the pipeline PRA list) add a state with a low index. 
* Have the outer one (the first in the pipeline PRA list) add a state with a higher index.

For example, given the following pipeline:

```
my_pra/A
my_pra/B
my_pra/C
```

* A runs `planner()`
  * B runs `planner()`
    * C runs `planner()`
      * The planner output a 4 state long trajectory
    * C appends a state to the trajecotry, adds `4` to its `added_path_index`. The trajectory is now 5 state long.
  * B prepends a state to the trajectory, adds `0` to its `added_path_index`. The trajectory is now 6 state long.
* A retrieves B's trajectory and appends a state. It adds `6` to its `added_path_index`. The trajectory is now 7 state long.

So far so good. Now the `PlanningRequestAdapterChain` [merges the different `added_path_index`](https://github.com/ros-planning/moveit/blob/69ca441bc343c51544b19dde9c5a095481a3d906/moveit_core/planning_request_adapter/src/planning_request_adapter.cpp#L153). The goal is to update the indices if other states were added before them.

### Expected behaviour

The index of the state added by C is updated to 5 to reflect that it's now the before-the-last state.
The final merged `added_path_index` is `0, 5, 6`.

### Actual behaviour

The indices are merged and incremented in the wrong order.
1. A index `6` is added.
2. B index `0` is added. Because it is lower than the 6, the 6 is incremented by 1.
3. C index `4` is added. Because it its lower than the 7, the 7 is incremented by 1.

The final merged `added_path_index` is `0, 4, 8`. `4` doesn't correspond the the state added by `C` and `8` isn't even a valid index.

Hugal31, 2023-06-28:
This also causes the #3462 bug.

rhaschke, 2023-09-28:
Fixed via #3464.

PR #3464 (2023-09-25): Fix the added path index vector merge in PlanningRequestAdapterChain
### Description

When merging the `added_path_index` for the different planning request adapters, the `PlanningRequestAdapterChain` was adding them in the wrong order, causing wrong incrementation of the indices.

Closes #3463.

### Checklist
- [X] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [M [... cut]
closed 2023-09-28 by rhaschke