moveit/moveit #3468: catkin build fails to build chomp_motion_planner

JustinYu2003, 2023-06-28:
### Description
After installing ROS Noetic and the linked tutorial: https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html#
When I get to running catkin build, it fails to build chomp_motion_planner.

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Source build

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
The build should succeed without errors or abandoning any packages.

### Actual behaviour
It outputs these errors and abandons the rest of the build.
_______________________________________________________________________________________________________________
Errors     << chomp_motion_planner:make /home/ajile/ws_moveit/logs/chomp_motion_planner/build.make.000.log     
In file included from /usr/include/eigen3/Eigen/Geometry:37,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:46,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:48,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/include/chomp_motion_planner/chomp_trajectory.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/include/chomp_motion_planner/chomp_cost.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/src/chomp_cost.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/EulerAngles.h: In member function ‘Eigen::Matrix<typename Eigen::internal::traits<T>::Scalar, 3, 1> Eigen::MatrixBase<Derived>::eulerAngles(Eigen::Index, Eigen::Index, Eigen::Index) const’:
/usr/include/eigen3/Eigen/src/Geometry/EulerAngles.h:39:3: error: there are no arguments to ‘EIGEN_USING_STD_MATH’ that depend on a template parameter, so a declaration of ‘EIGEN_USING_STD_MATH’ must be available [-fpermissive]
   39 |   EIGEN_USING_STD_MATH(atan2)
      |   ^~~~~~~~~~~~~~~~~~~~
/usr/include/eigen3/Eigen/src/Geometry/EulerAngles.h:39:3: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/usr/include/eigen3/Eigen/src/Geometry/EulerAngles.h:39:30: error: expected ‘;’ before ‘EIGEN_USING_STD_MATH’
   39 |   EIGEN_USING_STD_MATH(atan2)
      |                              ^
      |                              ;
   40 |   EIGEN_USING_STD_MATH(sin)
      |   ~~~~~~~~~~~~~~~~~~~~        
In file included from /usr/include/eigen3/Eigen/Geometry:41,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:46,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:48,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/include/chomp_motion_planner/chomp_trajectory.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/include/chomp_motion_planner/chomp_cost.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/src/chomp_cost.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/Rotation2D.h: In member function ‘Eigen::Rotation2D<Scalar>& Eigen::Rotation2D<Scalar>::fromRotationMatrix(const Eigen::MatrixBase<OtherDerived>&)’:
/usr/include/eigen3/Eigen/src/Geometry/Rotation2D.h:178:3: error: there are no arguments to ‘EIGEN_USING_STD_MATH’ that depend on a template parameter, so a declaration of ‘EIGEN_USING_STD_MATH’ must be available [-fpermissive]
  178 |   EIGEN_USING_STD_MATH(atan2)
      |   ^~~~~~~~~~~~~~~~~~~~
/usr/include/eigen3/Eigen/src/Geometry/Rotation2D.h:178:30: error: expected ‘;’ before ‘static_assert’
  178 |   EIGEN_USING_STD_MATH(atan2)
      |                              ^
      |                              ;
/usr/include/eigen3/Eigen/src/Geometry/Rotation2D.h: In member function ‘Eigen::Rotation2D<Scalar>::Matrix2 Eigen::Rotation2D<Scalar>::toRotationMatrix() const’:
/usr/include/eigen3/Eigen/src/Geometry/Rotation2D.h:190:3: error: there are no arguments to ‘EIGEN_USING_STD_MATH’ that depend on a template parameter, so a declaration of ‘EIGEN_USING_STD_MATH’ must be available [-fpermissive]
  190 |   EIGEN_USING_STD_MATH(sin)
      |   ^~~~~~~~~~~~~~~~~~~~
/usr/include/eigen3/Eigen/src/Geometry/Rotation2D.h:190:28: error: expected ‘;’ before ‘EIGEN_USING_STD_MATH’
  190 |   EIGEN_USING_STD_MATH(sin)
      |                            ^
      |                            ;
  191 |   EIGEN_USING_STD_MATH(cos)
      |   ~~~~~~~~~~~~~~~~~~~~      
/usr/include/eigen3/Eigen/src/Geometry/Rotation2D.h:194:31: error: ‘sinA’ was not declared in this scope; did you mean ‘sinh’?
  194 |   return (Matrix2() << cosA, -sinA, sinA, cosA).finished();
      |                               ^~~~
      |                               sinh
In file included from /usr/include/eigen3/Eigen/Geometry:42,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:46,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:48,
[... 192 lines cut ...]
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/include/chomp_motion_planner/chomp_cost.h:39,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/src/chomp_cost.cpp:37:
/usr/local/include/eigen3/Eigen/src/Core/GlobalFunctions.h:89:3: note: candidate: ‘template<class Derived> const Eigen::CwiseUnaryOp<Eigen::internal::scalar_sqrt_op<typename Derived::Scalar>, const Derived> Eigen::sqrt(const Eigen::ArrayBase<Derived>&)’
   89 |   EIGEN_ARRAY_DECLARE_GLOBAL_UNARY(sqrt,scalar_sqrt_op,square root,\sa ArrayBase::sqrt DOXCOMMA MatrixBase::cwiseSqrt)
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/eigen3/Eigen/src/Core/GlobalFunctions.h:89:3: note:   template argument deduction/substitution failed:
In file included from /usr/include/eigen3/Eigen/Geometry:42,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:46,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:48,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/include/chomp_motion_planner/chomp_trajectory.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/include/chomp_motion_planner/chomp_cost.h:40,
                 from /home/ajile/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner/src/chomp_cost.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:792:15: note:   mismatched types ‘const Eigen::ArrayBase<Derived>’ and ‘Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3> >::Scalar’ {aka ‘double’}
  792 |       t = sqrt(mat.coeff(i,i)-mat.coeff(j,j)-mat.coeff(k,k) + Scalar(1.0));
      |           ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/chomp_motion_planner.dir/build.make:63: CMakeFiles/chomp_motion_planner.dir/src/chomp_cost.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:966: CMakeFiles/chomp_motion_planner.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/ajile/ws_moveit/build/chomp_motion_planner; catkin build --get-env chomp_motion_planner | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2023-06-29:
The error message states that `EIGEN_USING_STD_MATH` is not defined. However, it should be defined within `Eigen/Core` which is correctly included from `Eigen/Geometry`. As [our CI builds compile without issues](https://github.com/ros-planning/moveit/actions?query=branch%3Amaster), I think your setup of the `Eigen` library is screwed somehow.
Try to clean your catkin workspace and rebuild: `catkin clean`.

JustinYu2003, 2023-07-04:
Hello, thanks for the response.

catkin clean doesn't seem to fix it, however installing the binary seems to work fine.
For now I'll work with the binary.

Thanks,
Justin

closed 2023-07-04 by rhaschke