moveit/moveit #3471: How to limit joint rotation while planning

HedgeHao, 2023-07-03:
I am using c++ moveitGroupInterface.plan() to move the robot. I am awared that there might be multiple planning results giving the same target pose.

For example, giving the same target pose (x,y,z,w),

I want the robot looks like this

[image]

but I got this

[image]

How can I limit the last joint to ‘face down’ ?

Robotawi, 2023-07-03:
Hi @HedgeHao 

The arm has 7 DOF, which adds more to this undesirable behavior. If there is a range of joints that you do not want the robot to operate in, according to your application, you can specify the joint position limits when defining the joints (in the urdf file) or in the `robot_moveit_config/joint_limits.yaml` file. I worked with a robot of a similar structure and ended up limiting this joint to almost zero range because the application did not need it. It became like a 6 DOF robot.

[image]

We are working on [pick_ik](https://github.com/PickNikRobotics/pick_ik) kinematic solver for MoveIt 2 and considering cost functions to satisfy secondary objectives when solving the IK problem. Related to your question, when solving IK, reach the goal pose while keeping joints biased toward a specific desirable range.

rhaschke, 2023-07-03:
Of course, redundant robots add to the problem of IK solution ambiguities. However, the original problem is that the goal orientation is not valid at all. Are you sure you are constraining orientation? Does your IK solver consider orientation (or are you using e.g. KDL's position-only solver)?

HedgeHao, 2023-07-04:
> Of course, redundant robots add to the problem of IK solution ambiguities. However, the original problem is that the goal orientation is not valid at all. Are you sure you are constraining orientation? Does your IK solver consider orientation (or are you using e.g. KDL's position-only solver)?

@rhaschke I am new to moveit. I simply use all the default solver and parameters. Give the target pose and get the result.

I'll try what @Robotawi says. See if it works.

Innomer, 2023-07-07:
Have you tried using Joint or Orientation Constraints to prevent this? I believe Joint Constraints can prevent such rotation of joints but might lead to a failed execution. Sorry if this does not work for you. I am not too well versed in this as well.

HedgeHao, 2023-08-11:
After a lot of tries. Still not working. If I set the joint limit, the plan fails most of the time. Still not able to found a good solution

Innomer, 2023-08-11:
> After a lot of tries. Still not working. If I set the joint limit, the plan fails most of the time. Still not able to found a good solution
> 

It seems to be a goal state orientation problem. Have you tried playing around with the orientation of the goal pose that you are providing? You could try using the tf transformations module to have the goal pose to face the opposite direction.

ZZWang21, 2023-09-15:
I am using GAZEBO+Moveit on a UR5 robot, I have the similar problem. I use python code to use MOVEIT to plan the movement. But even with joint constraints, it was not working. Can anybody help? Thanks.

jide07, 2024-08-08:
You need to pass orientation with your pose, you can try this
    pose.orientation.x = -1.0;
    pose.orientation.y = 0.00;
    pose.orientation.z = 0.00;
    pose.orientation.w = 0.00;
    
    this will limit the rotation angle.