moveit/moveit #3472: CPU usage of rviz increases when running for long periods

nyxrobotics, 2023-07-03:
### Description
Since the [814908b](https://github.com/ros-planning/moveit/commit/814908bbf3dcfb193c4f9d848e8e4aeb876a1713) pull request, the cpu usage of rviz has been increasing over time
This is about twice the CPU usage in an hour
I went back and verified the commits one by one, it seems to be caused by commit [d98a529](https://github.com/ros-planning/moveit/tree/d98a529a28ffe9bc283abe70828388a5b5b48c59)
Revert this commit will resolve it
I have read the commit logs and code but could not find the cause of the bug
I don't know what this commit is needed for

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04 (with nvidia GPU)
* Source
  *  [1a95deb](https://github.com/ros-planning/moveit/commit/1a95deb9ccf8c1f8da65f6afcdc25878c5800f8f)

### Steps to reproduce
I am using xarm
```bash
roslaunch xarm_gazebo xarm7_beside_table.launch add_gripper:=true
```

```bash
roslaunch xarm7_gripper_moveit_config xarm7_gripper_moveit_gazebo.launch 
```

### Expected behaviour
rviz cpu utilization is essentially constant

### Actual behaviour
rviz cpu utilization will increase forever over time

### Backtrace or Console output
nothing

nyxrobotics, 2023-07-24:
In my environment, I solved the problem by reverting the commit of [d98a529](https://github.com/ros-planning/moveit/tree/d98a529a28ffe9bc283abe70828388a5b5b48c59). Is it the same in your environment?

rhaschke, 2023-07-26:
I can't yet reproduce the issue. And I don't really understand how reverting d98a529 can give any improvement in performance. This commit just added two simple boolean queries, which shouldn't take much time (and they should reduce the probability of executing the more time consuming code in the `if` branch.)

Could you please provide an rviz config reproducing this issue?
I would also like to see some concrete numbers regarding CPU usage.

nyxrobotics, 2023-07-26:
Thanks for your help. Here are the details of what I have tried.

### ROS package I use
* [xarm_ros](https://github.com/xArm-Developer/xarm_ros) (clone master branch)
* [moveit](https://github.com/ros-planning/moveit) (clone noetic-devel branch)
* [moveit_visual_tools](https://github.com/ros-planning/moveit_visual_tools) (clone noetic-devel branch)
* [moveit_msgs](https://github.com/ros-planning/moveit_msgs) (clone master branch)

### Steps to reproduce
```bash
roslaunch xarm_gazebo xarm7_beside_table.launch add_gripper:=true
```
```bash
roslaunch xarm7_gripper_moveit_config xarm7_gripper_moveit_gazebo.launch 
```
Then, the rviz config is [moveit.rviz](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm7_gripper_moveit_config/launch/moveit.rviz).

### Rviz CPU usage
- Just after launch -> 10.1%
- 30 minuts later -> 37.8%

Could you try the above method to see what happens?

rhaschke, 2023-07-26:
Could you please check without Gazebo as well? I'm just running `roslaunch panda_moveit_config demo.launch` and it works like a charm. Gazebo itself is very CPU-hungry and probably the overall CPU load is very high. This might cause rviz not being able to process all incoming messages fast enough, which leads to message queuing and thus higher CPU usage over time.
Please monitor memory usage. Increasing memory usage over time indicates message queuing.
Please also verify the frequency of messages processed by rviz, most prominently `/joint_states`:
`rostopic hz /joint_states`

Still, it is very puzzling to me, that reverting d98a529a28ffe9bc283abe70828388a5b5b48c59 fixes the problem for you.

rhaschke, 2023-07-27:
I can confirm your issue running the xarm setup. d98a529a28ffe9bc283abe70828388a5b5b48c59 introduces regular updates of the query states, even if they are not shown. The update rate was as low as 4 Hz or so. I don't yet understand why CPU load and memory usage increase with these updates.

simonschmeisser, 2023-07-27:
Maybe some sort of Ogre entity is leaked?

rhaschke, 2023-07-27:
Yes, looks like `RobotInteraction` has a memory leak. Unfortunately, I don't have time to dig into that. Maybe the increase in CPU usage is due to missing cache hits if memory usage increased too much?

rhaschke, 2023-07-27:
@nyxrobotics, please try PR #3478 as a quick fix. This partially reverts d98a529a28ffe9bc283abe70828388a5b5b48c59.
However, the fundamental memory leakage in RobotInteraction should be analyzed...

nyxrobotics, 2023-07-28:
> @nyxrobotics, please try PR #3478 as a quick fix. This partially reverts [d98a529](https://github.com/ros-planning/moveit/commit/d98a529a28ffe9bc283abe70828388a5b5b48c59). However, the fundamental memory leakage in RobotInteraction should be analyzed...

Your PR has solved the problem of high CPU usage in my environment. Thank you so much.