moveit/moveit #3482: Problems when trying to install older version of MoveIt

Renzo-Mendoza, 2023-08-01:
Hello everyone, I am trying to install a previous version of MoveIt (version 1.1.12-1) using binary installation on Ubuntu 20.04. However, apparently this version is not available. I am using the following command in terminal:

`sudo apt-get install ros-noetic-moveit=1.1.12-1`

I would appreciate if you could help with a solution to install this version. Is there any other way to work with this specific MoveIt version?

rhaschke, 2023-08-01:
This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here only.
Actually, this question was already asked before: 
https://answers.ros.org/question/308566/apt-get-to-install-ros-package-with-specific-version!

Note, 1.1.12 is the latest officially released MoveIt version. 
However, the corresponding package version is: `1.1.12-1focal.20230526.061512`. Using that one works as expected: 
`sudo apt-get install ros-noetic-moveit=1.1.12-1focal.20230526.061512`

closed 2023-08-01 by rhaschke

Renzo-Mendoza, 2023-08-01:
Hi, thanks for the help, I apologize for asking this question here. When using the following command in the terminal:
`sudo apt-get install ros-noetic-moveit=1.1.12-1focal.20230526.061512`
I got this:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '1.1.12-1focal.20230526.061512' for 'ros-noetic-moveit' was not found
```
When using:
`apt show ros-noetic-moveit`
Only shows version 1.1.13-1focal.20230728.162553

rhaschke, 2023-08-02:
You are using [ROS `testing` repo](http://wiki.ros.org/TestingRepository), which already has the newer release.
In any case, the ROS repos only keep the last version of a package. Hence, you can only install this last version.

simonschmeisser, 2023-08-02:
There is of course also http://snapshots.ros.org/noetic/ which contains snapshots of the whole ROS repo at certain times (cannot seem to answer on answers.ros.org any more)