moveit/moveit #3488: JointConstraint causes sudden arm movement and robot crash

BrendanWooAdmetal, 2023-08-25:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
Binary build
Release version: 1.1.13
Platform: Jetson AGX

### Steps to reproduce
When commanding a 6-joint robot arm with a gripper to move into position to pick up an object directly in front of it, the arm had a tendency to sweep the arm around instead of simply reaching forwards to grab it. 
To deal with this, we attempted to add some constraints using the following code:

<pre>forward_traj_constraints = moveit_msgs.msg.Constraints()
forward_traj_constraints.name = "forward_traj_constraints"
joint1_constraint = moveit_msgs.msg.JointConstraint(
    joint_name = "joint1",
    position = 0,
    tolerance_above = pi/2,
    tolerance_below = pi/2,
    weight = 1
)
<b>#The code works fine when this constraint is not included.
joint2_constraint = moveit_msgs.msg.JointConstraint(
    joint_name = "joint2",
    position = 0,
    tolerance_above = pi/2,
    tolerance_below = 0.1,
    weight = 0.95
)</b>
joint6_constraint = moveit_msgs.msg.JointConstraint(
    joint_name = "joint6",
    position = 0,
    tolerance_above = pi/3,
    tolerance_below = pi/3,
    weight = 0.92
)
joint4_constraint = moveit_msgs.msg.JointConstraint(
    joint_name = "joint4",
    position = 0,
    tolerance_above = pi/2,
    tolerance_below = pi/2,
    weight = 0.99
)
forward_traj_constraints.joint_constraints.extend([joint1_constraint, joint2_constraint, joint4_constraint, joint6_constraint])
traj_constraints = moveit_msgs.msg.TrajectoryConstraints()
traj_constraints.constraints.append(forward_traj_constraints)
# self.rm.arm_group.set_trajectory_constraints(traj_constraints)
self.rm.arm_group.set_path_constraints(forward_traj_constraints)</pre>

### Expected behaviour

The arm is expected to reach straight forwards, following the constraints which prevent it from sweeping around.
The expected behaviour occurs when the second constraint is removed, and the issue does not appear.
The second constraint affects the second joint on the robot (second closest to the base of the arm).
This line that seems to cause the error is bolded above.

The arm's movement is normally very slow, taking several seconds to reach the position.

### Actual behaviour

The error occurs when this code is run as shown above.
The arm jerks forwards quickly a short distance, then stops abruptly, and our Jetson AGX running the code shuts down.
This occurred during two consecutive tests.

It is notable that before shutdown, warnings appear about the second constraint that seems to be causing the error.
They note that the second constraint is violated, as the starting position of the arm has the affected joint starting at an angle that violates that constraint.
These lines have been bolded in the output below.

While general advice about constraints is appreciated, the main concern is the sudden and violent movement of the arm, and the Jetson AGX crashing.
It is difficult to determine which is the cause.

It could be that a serious error is encountered that causes the AGX to crash while MoveIt is controlling the arm, and this results in the abnormal arm movement.

My co-worker also pointed out the AGX and arm are drawing from the same battery, and suggested instead that the arm's sudden movement caused a power draw that made the AGX shutdown.

However, regardless of which is the cause, either is concerning.
The arm's movement is normally very slow, and it is unclear why it so suddenly moves so quickly when its speed should be a very slow, safe speed.
We are also concerned about the AGX crashing. It has happened before while running tests of MoveIt and our robot arm, but until now it has happened too inconsistently to find a cause.

### Backtrace or Console output
<pre>[INFO] [1692996702.700928]: [Move to Pose] Arm Moving to Pose
[WARN] [1692996702.720535]: [Move to Pose] Sent goal to move_group
[ INFO] [1692996702.748394718]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1692996702.748917079]: Planning attempt 1 of at most 1
[ INFO] [1692996702.751805825]: Path constraints not satisfied for start state...
[ INFO] [1692996702.753086302]: Constraint satisfied:: Joint name: 'joint1', actual value: 0.000000, desired value: 0.000000, tolerance_above: 1.570796, tolerance_below: 1.570796
<b>[ INFO] [1692996702.753283271]: Constraint violated:: Joint name: 'joint2', actual value: -1.570465, desired value: 0.000000, tolerance_above: 1.570796, tolerance_below: 0.100000</b>
[ INFO] [1692996702.753409037]: Constraint satisfied:: Joint name: 'joint4', actual value: -0.000017, desired value: 0.000000, tolerance_above: 1.570796, tolerance_below: 1.570796
[ INFO] [1692996702.753547860]: Constraint satisfied:: Joint name: 'joint6', actual value: 0.000000, desired value: 0.000000, tolerance_above: 1.047198, tolerance_below: 1.047198
[ INFO] [1692996702.753705532]: Planning to path constraints...
[ INFO] [1692996702.760613094]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1692996702.761502160]: arm/arm: Starting planning with 1 states already in datastructure
[ INFO] [1692996702.795180442]: arm/arm: Created 5 states (2 start + 3 goal)
[ INFO] [1692996702.795437958]: Solution found in 0.034217 seconds
[ INFO] [1692996702.869114888]: SimpleSetup: Path simplification took 0.072712 seconds and changed from 4 to 2 states
[ INFO] [1692996702.875813000]: Planned to path constraints. Resuming original planning request.
[ INFO] [1692996702.883157543]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1692996702.885950925]: arm: Allocating specialized state sampler for state space
[ INFO] [1692996702.886222714]: arm/arm: Starting planning with 1 states already in datastructure
<b>[ WARN] [1692996702.970452404]: More than 80% of the sampled goal states fail to satisfy the constraints imposed on the goal sampler. Is the constrained sampler working correctly?</b>
[ INFO] [1692996703.180996463]: arm/arm: Created 4 states (2 start + 2 goal)
[ INFO] [1692996703.182618651]: Solution found in 0.298871 seconds
[ INFO] [1692996703.239968533]: SimpleSetup: Path simplification took 0.044189 seconds and changed from 3 to 2 states
[ INFO] [1692996703.253940938]: Planning adapters have added states at index positions: [ 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 ]
[ INFO] [1692996703.278146628]: First Move_group give us 66 points
[ INFO] [1692996703.283628231]: Second Move_group max_time is 10.531144 
[ INFO] [1692996703.288326886]: Now We get all tarjectory num 528
</pre>
The computer crashes after this line.

simonschmeisser, 2023-08-27:
I would recommend trying to separate your issues first. Try if the crash happens on a real computer in simulation (fake trajectory execution) with capable power supply.

The error message seems to indicate that your start state is outside the path constraint. There are path constraints and goal constraints. Maybe you mix them up?

Check if your inverse kinematics (ik) work well, maybe that's the reason for the strange movement in the first place.

Once you checked these and still struggle please provide a link to git repo with your full setup and code for the motion you're trying to plan

BrendanWooAdmetal, 2023-08-28:
Thank you for your response.
We've checked our inverse kinematics separately before running these test where the failure happened, and found no issues.

One thing we could use help with is the difference between the path and goal constraints for the MoveGroup Python interface.
We've had some difficulty finding documentation on trajectory constraints in particular.

My understanding is that path constraints constrain the start state, end, and the path in between them.
But what are trajectory constraints and goal constraints?
I can't find a reference to goal constraints in they MoveGroup code, and little documentation about what trajectory constraints are.

Thank you.

BrendanWooAdmetal, 2023-08-30:
As requested, we have investigated the issue further.
We began using the RViz Trajectory and MotionPlanning to simulate the robot's movement, and it shows no problems, but the arm still has the issue when it actually tries to move.
We also researched the constraints and are fairly certain we are not mixing them up.
Additionally, we found another way to produce the error, and the failure still happens even if no constraints are given when calling execute().

The second way the error is produced is when using compute_cartesian_path() to plan a movement, and then execute() to execute the returned trajectory.
As before, the planning went fine, and examination of the trajectory by printing it and when viewing the simulated movement in RViz showed no issues.
The planned motion was within our velocity and acceleration limits.

However, the same issue emerged when execute() was called, with the arm jerking forwards quickly (above our velocity limits) and the AGX shutting down.
The issue can appear in calls to both go() and execute(), although it fails every time in execute(), and go() only seems to have failed when given the constraints above.

As the issue does not appear in simulation but does in the actual motion, I am unsure of the cause.

Here is the code that is being used for tests:
```
def arm_moveit_test(self):
        """" Used to test MoveIt's robot arm control with a hardcoded pose """
        # c_pose = rmapi.new_posestamped(0.001, 0.0001, 0.85, 0.0001, 0, 0.001)
        # self.rm.arm_group.set_pose_target(c_pose)

        # pose_goal = geometry_msgs.msg.Pose()
        # pose_goal.orientation.w = 1.0
        # pose_goal.position.x = 0.4
        # pose_goal.position.y = 0.1
        # pose_goal.position.z = 0.4
        # self.rm.arm_group.set_pose_target(pose_goal)

        curr_pose = self.rm.get_pose()
        pose2 = curr_pose
        pose2.position.z += 0.1

        poses = [curr_pose, pose2]
        eef_step = 0.01
        jump_threshold = 0        
        
        trajectory, path_complete = self.rm.arm_group.compute_cartesian_path(poses, eef_step, jump_threshold)
        rospy.loginfo(f"[arm_moveit_test] Cartesian path calculated: {path_complete*100}% completed")
        rospy.loginfo(f"[arm_moveit_test] Cartesian path trajectory: {trajectory}")
        ret = path_complete

        #Publish to MoveIt so it can be seen in RViz
        from moveit_msgs.msg import DisplayTrajectory
        disTraj = DisplayTrajectory()
        disTraj.trajectory = [trajectory]
        self.rm.display_trajectory_publisher.publish(disTraj)
        if path_complete == 1:
            #Fails on call to execute()
            ret = self.rm.arm_group.execute(trajectory)

        if ret == 1:
            rospy.logwarn("[arm_moveit_test] Traj Completed!!!!")
        else:
            rospy.logwarn("[arm_moveit_test] Traj Failed!!!!")

        return ret
```

Output:
```
[ INFO] [1693430732.956961699]: [rm65] Sending command succced!
[ INFO] [1693430733.131023518]: Received request to compute Cartesian path
[ INFO] [1693430733.133037158]: Attempting to follow 2 waypoints for link 'Link6' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1693430733.143023397]: Computed Cartesian path with 12 points (followed 100.000000% of requested trajectory)
[INFO] [1693430733.159914]: [arm_moveit_test] Cartesian path calculated: 100.0% completed
[INFO] [1693430733.201865]: [arm_moveit_test] Cartesian path trajectory: joint_trajectory: 
  header: 
    seq: 0
    stamp: 
      secs: 0
[... 95 lines cut ...]
      time_from_start: 
        secs: 0
        nsecs: 727552474
    - 
      positions: [-0.0022925592026092657, -0.2798524425390303, 1.0975929431102869, 0.0029259322161075394, 1.5416941337213732, -0.00034946187898244447]
      velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      effort: []
      time_from_start: 
        secs: 0
        nsecs: 727552474
multi_dof_joint_trajectory: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  joint_names: []
  points: []
[ INFO] [1693430733.244284356]: Execution request received
[ INFO] [1693430733.273031817]: First Move_group give us 12 points
[ INFO] [1693430733.281581209]: Second Move_group max_time is 0.727552 
[ INFO] [1693430733.284158969]: Now We get all tarjectory num 38
```

gavanderhoorn, 2023-08-31:
> When commanding a 6-joint robot arm [..]
>
> [..] the arm still has the issue when it actually tries to move.
>
> [..] Additionally, we found another way to produce the error, and the failure still happens even if no constraints are given when calling execute().

what sort of "robot arm" is this? Could the driver be doing something strange? To what level does it (attempt to) execute the trajectory? Position only? Velocities? Accelerations?

If you plot the data for all axes for all fields in the trajectory, any discontinuities?

BrendanWooAdmetal, 2023-08-31:
Thanks for your response, a lot of helpful tips in there.

The robot is a 6-jointed arm from Realman.
The arm's datasheet says that each joint has a maximum velocity of pi rad/sec, but it has been limited to pi/4 rad/sec for our application.
The acceleration has been limited to 5 rad/s^2, which is also well below the acceleration limits stated in the datasheet.

I tried making the acceleration limits even stricter at 1 rad/sec^2 and powering the arm and AGX from wall-connected power supplies.
The AGX stayed on this time, so it was losing power due to the arm's power draw.
However, the arm still experienced the jerking, so although it was enough to cause issues with the AGX, the jerking itself doesn't seem to be the result of insufficient power to the arm.

> could you clarify how you've verified your IK plugin? Especially when using computeCartesianPath(..), "strange" things are often caused by IK solvers not deterministically returning the same solutions on repeated calls.

The inverse kinematics were tested by using the robot arm's inverse kinematics to return a solution for a given motion, and then using the returned values to move the arm according to that trajectory using arm control functions from the arm manufacturer that we know work.
So we tested the same inverse kinematics function before with these arm commands and it worked fine, but when using MoveIt to control the arm, these issues began occurring.

> [ INFO] [1693430733.273031817]: First Move_group give us 12 points
[ INFO] [1693430733.281581209]: Second Move_group max_time is 0.727552 
[ INFO] [1693430733.284158969]: Now We get all tarjectory num 38

Thanks for pointing this out.
I thought this was from MoveIt code, and since you didn't recognize it, I did some digging and found that it was part of the arm control code from the arm manufacturer.
Perhaps I had been digging into MoveIt's code when I should have been examining the code provided by the manufacturer more closely.

It seems the arm itself is reporting errors which I will need to reach out to the manufacturer about.
I will continue looking into the other suggestions as well, thank you.

gavanderhoorn, 2023-09-01:
> I thought this was from MoveIt code [..]

suggestion: update your rosconsole output format template to include the name of the node.

The default doesn't, so it's impossible to figure out *what* is printing that line.

See [wiki/rosconsole - Console Output Formatting](http://wiki.ros.org/rosconsole#Console_Output_Formatting).

gavanderhoorn, 2023-09-01:
For future reference / future readers: [link to the code of the driver mentioned](https://github.com/RealManRobot/rm_robot/blob/6c645018bd4e5392aef9407ea2a396042587f93e/rm_control/src/rm_control.cpp#L445).

BrendanWooAdmetal, 2023-09-01:
Thanks again for the help.

We've identified the issue.
The issue was on these lines of our code:

```
curr_pose = self.rm.arm_group.get_current_pose().pose
rospy.loginfo(f"moveit current pose:{curr_pose}")
rospy.loginfo(f"rm current pose:{self.rm.get_pose()}")

# This line causes the problem
pose2 = curr_pose
# Should be this instead- self.rm.arm_group.get_current_pose().pose

pose2.position.z += 0.1

poses = [curr_pose, pose2]
eef_step = 0.01
jump_threshold = 0

trajectory, path_complete = self.rm.arm_group.compute_cartesian_path(poses, eef_step, jump_threshold)

#Publish to MoveIt so it can be seen in RViz
from moveit_msgs.msg import DisplayTrajectory
disTraj = DisplayTrajectory()
disTraj.trajectory = [trajectory]
self.rm.display_trajectory_publisher.publish(disTraj)
if path_complete == 1:
    input("Ready to execute on your command: ")
    ret = self.rm.arm_group.execute(trajectory)
```

When two names in Python are assigned the same Pose() object, they point to the same object.
As a result, the two poses being given to compute_cartesian_pose() were in fact the same pose, as the two names referenced the same object.
This can be easily remedied by simply making a deep copy of the original pose object, or getting the current pose again.
Presumably, the cartesian planning can't handle trying to move in a straight line from one point to the same point.

With this in mind, the trajectories output above by compute_cartesian_path do not make sense; If the arm is going nowhere, it shouldn't be moving at all, but instead it is giving a trajectory which seems to try and move in a random direction.
MoveIt doesn't realize this trajectory is bad, and reports the trajectory was successfuly calculated.

Setting the eef_step to be larger than the distance to move does not cause this issue, it seems to be specifically when two identical poses are given as the waypoints.

I'll admit not realizing the two poses were the same was silly on my part.
However, ideally compute_cartesian_path shouldn't have tried to make a trajectory like the ones posted above when told to make a cartesian path that doesn't actually move anywhere.
I'm still not sure what caused compute_cartesian_path to do this, but perhaps a simple check if all the given poses are the same could prevent this from arising in the future?

---

When the trajectory was given to the Realman driver, it caused calculation errors that resulted in nan values being given as the angles the joints should move to, causing the jerky motions. 

The appearance of nan values can also occur in other cases.
For example, giving these values to compute_cartesian_path results in the Realman driver giving nan again.
These combinations of values in the code above caused the Realman arm to give nan values:
```pose2.position.z += 0.1 and eef_step = 0.01```
```pose2.position.z += 0.05 and eef_step = 0.005```
```pose2.position.z += 0.05 and eef_step = 0.01```

We're unsure right now if it is a problem with the trajectory given by MoveIt, or the Realman's handling of the trajectory.
We have contacted the Realman team for support, and are working on a check for nan values for the [RealMan robot API.](https://github.com/RealManRobot/rm_robot/tree/6c645018bd4e5392aef9407ea2a396042587f93e)

---

Another issue is that the current pose given as a starting point should use MoveIt's get_current_pose(), not the Realman library's get_pose().
As seem below, these have different outputs, notably with x being inverted.
This is the result of each one using a different parent frame.
My co-worker has suggested this issue could be ameliorated if PoseStamped was used instead of Pose, as PoseStamped explicitly sets the parent frame.
```
MoveIt get_current_pose-
position:
  x: 0.11199845039646755
  y: -3.4973499908381703e-06
  z: 0.450632365498854
orientation:
  x: -0.7069922696862102
  y: 4.253612103208138e-10
  z: 0.7072212737147834
  w: 2.4681598275405308e-05
  
rm get_pose-
position:
  x: -0.1120000034570694
  y: 0.0
  z: 0.44999998807907104
orientation:
  x: -0.00056
  y: 0.706915
  z: 0.000147
  w: 0.707298
  ```

BrendanWooAdmetal, 2023-09-06:
Further investigation and correspondence with the Realman arm team shows that there are error codes published by the arm during the jerky motions.
These codes (4106 and 4105 in decimal) state that the arm is exceeding its velocity and acceleration limits.
As the limits on the arm are higher than the limits we set in joint_limits.yaml for MoveIt, this confirms that the jerky motions exceed MoveIt's velocity and acceleration limits.

We will follow up with the Realman team again and continue a bit more testing, but the main issue has been diagnosed and we found the workaround documented in my previous post.
Unless we find some new information that may help others who encounter the issue, I will be marking this issue as resolved soon.
Thank you to everyone who provided support.

closed 2023-09-13 by BrendanWooAdmetal