moveit/moveit #3493: IKFast solver always returns zero for a joint value

kboonpasart, 2023-09-01:
I'm working on inverse kinematics for a ur16e robot manipulator (6DOF) attached to a mobile base. Following [this tutorial](http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html), I generate an IKFast solver from the URDF file of the whole robot (the baselink and eelink are chosen to match ones of the robot arm). The resulting .cpp solver is then made an executable for the use.

However, when given an input as a set of values representing the position and orientation of the end effector, the resulting output is not correct. From the 6 joint angles for the 6 axes, the value for the 5th angle is always zero, no matter which orientation is given as an input.

For example, an input would look something like this:

```
r00 r01 r02 t0 r10 r11 r12 t1 r20 r21 r22 t2
0 1 0 0.5 0 0 1 0.5 1 0 0 0.5
```

And a solution would look something like this:

```
[0.902574921849587, -2.626397212736598, -0.61608885723397, 0.100893416380779, 0.0, -1.570796326794897]
```

From trial and error, giving these values to the actual (virtual) robot, it seems that the other joint angle values (or at least from the 1st to the 4th one) might actually be correct. That's because, with the other joint values in place, I can just manually rotate the 5th joint and reach the expected end effector poses.

Do any of you have any idea what could cause such an error?

The generated IKFast solver:
[ikfast61_ur16e_arm_2.zip](https://github.com/ros-planning/moveit/files/12499982/ikfast61_ur16e_arm_2.zip)

rhaschke, 2023-09-01:
We cannot provide support for IKFast itself. This is an external solver, which is just wrapped by MoveIt. Please try asking for community help at https://answers.ros.org.

ZSorcerer, 2023-11-24:
I've encountered exactly the same problem as you, has your issue been resolved? [**https://github.com/rdiankov/openrave/issues/1334**](https://github.com/rdiankov/openrave/issues/1334)

rhaschke, 2023-11-24:
Are these correct solutions for your target position? If so, maybe that pose is only possible with joint_5=0?

ZSorcerer, 2023-11-24:
> Are these correct solutions for your target position? If so, maybe that pose is only possible with joint_5=0?

No，it's definitely not correct， you can view my link， I gived the right solution

rhaschke, 2023-11-24:
OK. I hope Rosen can comment on this.

kboonpasart, 2023-11-24:
> I've encountered exactly the same problem as you, has your issue been resolved? [**https://github.com/rdiankov/openrave/issues/1334**](https://github.com/rdiankov/openrave/issues/1334)
> 
> [image]

No, unfortunately not. I've moved on and used another solver from a random repo. Hope your problem will be solved soon.

closed 2023-11-24 by kboonpasart

ZSorcerer, 2023-11-25:
> No, unfortunately not. I've moved on and used another solver from a random repo. Hope your problem will be solved soon.

I have solved this problem，The reason is the UR robot model， You have to use the old version，not the latest version， the old model gives correct UR5 axis and size.  I've drawn out the model of the UR robot and noticed that the old version of the model represents the complete robot, while the new version is a disassembled model. This could lead to errors when ikfast is parsing the model.

rhaschke, 2023-11-25:
What exactly do you mean by disassembled model?

maxwelllls, 2023-12-07:
After several days of trials, I've found that different initial configurations seem to affect the solving judgment of ikfast. I'm using a robot arm similar to the UR configuration. I encountered the same problem using the erobo3_1.urdf, but after adjusting to erobo3_2.urdf, it was solvable. 
P.S.: The generation environment is the docker image from personalrobotics/ros-openrave.

pose1:fail
[image]

pose2:success
[image]

[erobo3_description.zip](https://github.com/ros-planning/moveit/files/13597043/erobo3_description.zip)

Crejan, 2024-06-10:
@ZSorcerer how did you finally solve the problem for the universal robot? Where do I find the "old model"?