moveit/moveit #3497: Non optimal motions using move_group.set_pose_target()

thve25, 2023-09-22:
### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* I tried both Source (master) and Binary (1.1.13-2focal.20230801.171807)
* 6 DOF arm

### Description

Using `move_group.set_pose_target(pose)`, Sometimes, I'm getting weird IK joint angles* and weird motions.
Using `move_group.set_joint_value_target(pose,True)`, I'm ***not*** getting weird motions.

_*= When I check the final joint angles (plan[1].joint_trajectory.points[-1].positions) of the plan=move_group.plan() in these cases, the joint angles are a valid IK solution, but far from the most optimal one. When I calculate an IK solution using roboticstoolbox, and use MotionPlanning in Rviz ( with collision aware IK) and the joint angels set to the IK solution i'm getting decent motions as intended._

However, obstacle avoidance is only supported using `set_pose_target()`, because it plans in Cartesian space.

How can I get the best of both worlds, ie. efficient motions and obstacle avoidance?

### I tried:

 - increasing planning attempts: `move_group.set_num_planning_attempts(20)`
   - didn't work
 - increase tolerance: `move_group.set_goal_tolerance(0.01)`
   - weird motions occurred less often.
  - KDLKinematicsPlugin and TRAC_IKKinematicsPlugin
  - Different planners (e.g. BiTRRT, RRT, FMT, etc)
  - Different pipelines (OMPL, OMPL-CHOMP)

###  ompl_planning.yaml:
```
planner_configs:
 BiTRRT:
    type: geometric::BiTRRT
    range: 0.0 
    temp_change_factor: 0.1  
    init_temperature: 100  
    frountier_threshold: 0.0  
    frountier_node_ratio: 0.1  
    cost_threshold: 1e300  
 etc

manipulator:
  default_planner_config: BiTRRT
  planner_configs:
    - BiTRRT
    - etc
  projection_evaluator: joints(joint_1,joint_2)
  longest_valid_segment_fraction: 0.005
```

rhaschke, 2023-09-22:
The `True` in `move_group.set_joint_value_target(pose,True)` refers to _approximate_ solutions. For the KDL solver that means, that the result of the first solution attempt is taken even if it doesn't (perfectly) reach the goal. Without that flag, the KDL solver will start a new local search from a random initial pose, causing the weird solutions you observe.
What do you mean by _obstacle avoidance is only supported using `set_pose_target()`_? Actually, collision avoidance should be active in both cases.

thve25, 2023-09-22:
```set_joint_value_target()``` only checks for collisions, and does not avoid them.

My goal is to set a new pose for the end effector, with collision avoidance.
Therefore I want to use ```set_pose_target()```

Both  TRAC and KDL result in the same behavior.

The joint limits in my URDF and joint_limits.yaml, and the resulting joint limits in the parameter server are correct and the same.

rhaschke, 2023-09-22:
You are right that `move_group.set_joint_value_target(pose, True)` doesn't avoid collisions for the goal pose.
However, if you change `True` to `False` for the second argument, collisions should be taken into account.
For actual planning of a motion from A to B, collision avoidance is active all the time. However, if your goal pose is in collision initially, a planning adapter will "wiggle" the robot into a collision-free pose first. Thus, you will actually reach a slightly different goal pose - same as with `approx=True`.

thve25, 2023-09-22:
Thanks.
I tried ```move_group.set_joint_value_target(pose, True)``` and ```move_group.set_joint_value_target(pose, False)``` again, but they do not always find a solution, while ```set_pose_target()``` does.

```set_joint_value_target()``` results sometimes in:

```
Found a contact between 'link_6' (type 'Robot link') and 'link_4' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
Collision checking is considered complete (collision was found and 0 contacts are stored)
manipulator/manipulator[BiTRRT]: Goal tree has no valid states!
No solution found after 0.030341 seconds
Invalid goal state
Unable to solve the planning problem
```

thve25, 2023-09-22:
> However, if you change `True` to `False` for the second argument,

It seems that both `True` and `False` result in a path with collision avoidance