moveit/moveit #3505: move_group not loading for dual 8 axis arm

aaryanmurgunde, 2023-10-03:
### Description
Hi Guys, I'm trying to work on settiing up a dual 8 axis abb arm with moveit. The xacro files for the description are quite large around 300mb as the arm is big in scale.  I was able to generate the package successfully and had no issues there but if I go to launch the demo.launch it is stuck on the same screen for quite some time. I'm able to move the arms with the marker but not plan for the same. I also noticed that only two cores are going 100% I'm not sure if this is relevant, but any help is appreciated. Following are attached a few images.

[Update] : I'm using kdl planner

### Your environment
* ROS Distro: Noetic
* OS Version:  Ubuntu 22.04 [Docker 20.04]

[image]
[image]
[image]
[image]

gavanderhoorn, 2023-10-03:
> The xacro files for the description are quite large around 300mb as the arm is big in scale.

you mean the *meshes* have a large file size, correct?

`.xacro` files are text files. They rarely go over a couple KBs.

I'd suggest to downsample the meshes significantly. Even such "large scale" systems can be accurately represented by meshes of only a couple MB.

aaryanmurgunde, 2023-10-20:
@gavanderhoorn Yes makes sense, I was able to get the mesh down. After alot of research I found out that I was getting stuck in a big loop for continuous checking for collisions (self and otherwise).

closed 2023-10-20 by aaryanmurgunde