moveit/moveit #3511: Is it possible to use moveit without a hardware interface or controller?

babbatem, 2023-10-11:
Hello, bit of a strange request -- is it possible to use moveit without a hardware interface or ros controllers? We are running our own controllers and trajectory execution stack and hoping to plan with moveit, then hand the trajectory off. Would this be possible, e.g. by publishing joint states from our controllers? 

Thanks for your time.

rhaschke, 2023-10-12:
As input, MoveIt just needs the URDF/SRDF model and published `/joint_states`. To pass a planned trajectory to your non-standard controller, you will need to provide a `FollowJointTrajectory` action server:
https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html

closed 2023-10-12 by rhaschke

babbatem, 2023-10-12:
Thanks for your reply.