moveit/moveit #3515: FCLShapeCache does not cache attached collision objects between different robot states on the same trajectory

rr-mark, 2023-10-26:
### Description

tl;dr:

The attached collision objects at each waypoint along a RobotTrajectory are not aliased, and so the FCLShapeCache misses the cache for each new waypoint, and a huge amount of processing time is spent in the fcl collision detection algorithm looping over every vertex and index in every collision object for every step of the trajectory.

Longer context:

I am planning trajectories using pilz_industrial_motion_planner's plan_sequence_path service. I am using relatively large attached collision meshes (~100k faces). These trajectories take a very long time to plan (~1s per waypoint), and more importantly the move_group node leaks memory continuously (~10Mb per waypoint), which is not recovered when the plan is finished.

I have been profiling the move_group, to see if what I'm wanting to do is possible, and have found that almost all (>99%) of the processing time is spent in the `FCLGeometryConstPtr::createCollisionGeometry` method. I believe (although correct me if I'm wrong) that the vast majority of this time would be saved if the FCLShapeCache was hit, but instead it is missed every time. This is happening because the attached collision objects associated with each waypoint along the RobotTrajectory are different objects (even though they represent the same objects, just moved slightly).

### Your environment
* ROS Distro: Noetic
* OS Version: 20.04
* Source or Binary build? Both; first the binary build, and then the source code when debugging.
* If binary, which release version? 1.1.12
* If source, which branch? master: b4b1a27f605031c54e80a9b3465f7a3cabe6cc88

### Steps to reproduce
I can set this up if it would help, but I don't think it's that kind of bug.

### Expected behaviour
I would assume the attached collision objects along a single trajectory would be cached, so they didn't have to be processed for every step along the trajectory.

### Actual behaviour
Planning takes a long time, and the memory usage of the move_group nodes increases quickly, and does not go down again when planning has finished.

### Backtrace or Console output

N/A

rr-mark, 2023-10-26:
I am more than happy to put some time into raising a PR for this, if my reasoning about the caching is correct. Although I am unsure whether it would be better to change the behaviour of the cache or to change how the robot trajectory is generated so that the attached collision objects are aliased.

simonschmeisser, 2023-10-26:
This is a common problem in MoveIt with big attached objects. When they are transmitted as [`moveit_msgs::RobotState`](http://docs.ros.org/en/noetic/api/moveit_msgs/html/msg/RobotState.html) they are essentially reduced to a triangle list [`shape_msgs::Mesh`](http://docs.ros.org/en/noetic/api/shape_msgs/html/msg/Mesh.html). Now the cache operates on `ShapePtr` so it has no chance to recognize those are actually the same two meshes. Safe maybe actually comparing them fully? Or hashing them?

My understanding is that you should have one `ShapePtr` per segment and that the cache should work withing each segment? Maybe you could check that with a profiler? (eg Hotspot)

https://github.com/ros-planning/moveit/blob/984a0ea2a00b4190cad9e08ea0d764c19dbf9975/moveit_planners/pilz_industrial_motion_planner/src/command_list_manager.cpp#L235

here is where they are created.

So for each segment actually a new start state in form of a `moveit_msgs::RobotState` is created to pass it to the planning pipeline.

Preserving them here is hard  but maybe the mesh could be hashed and stored in another cache? This could be done here:

https://github.com/ros-planning/moveit/blob/984a0ea2a00b4190cad9e08ea0d764c19dbf9975/moveit_core/robot_state/src/conversions.cpp#L287

This could possibly be thread local so it does not add any locking complexity.

Now to your memory leak:

There are two effects that look like memory leaks but are actually "caches". FCLShapeCache is thread local so if you plan from many threads (or use a multi threaded planner like OMPL) then the mesh will be cached N times. The other effect is that the MoveIt OMPL plugin has a list of "currently used" "planner instances" and each keeps one PlanningScene around for post planning analysis (never actually done ...). So if you have many planning requests in parallel this might also keep some memory.

But there might be other issues, maybe you can pin point it using https://github.com/KDE/heaptrack

If you think we are talking about completely different things, please add a few links to code

rr-mark, 2023-10-27:
Perfect, thank you. I will have a look at creating a shape cache today. Hopefully that will also fix the memory issues I'm seeing (I agree with you; it is likely that this is a massive cache rather than a true memory leak), and if it doesn't I'll look into KDE and hopefully come back with a separate issue with a bit more information.

rr-mark, 2023-10-27:
Okay, looks like adding a mesh cache resolves the memory problems and makes planning a lot faster. I'll turn my proof-of-concept into a proper cache, then submit a PR.

simonschmeisser, 2023-10-27:
Lovely, looking forward to it!

rr-mark, 2023-10-27:
Okay, mostly working. I'll finish it off next week.

https://github.com/rr-mark/moveit