moveit/moveit #3521: Allow flexible trajectory planning that automatically reduce velocities/accelerations to fit within joint limits

antoinedang, 2023-10-31:
**Is your feature request related to a problem? Please describe.**
It should be possible to plan a trajectory where achieving exact velocities is not important, but the velocity should be as close as possible to a specific target velocity. For example, when programming pick-and-place trajectories, it is important to go as fast as possible between some waypoints. But currently, finding that maximum velocity at which the joint velocity/acceleration limits are respected has to be done through trial-and-error. I've already implemented a solution to this problem in a private fork but I see it has not been addressed here.

**Describe the solution you'd like**
An argument to the motion plan request or some way of indicating to the trajectory generation part of the code that the target velocity is "flexible" or something. Then, when computing the trajectory, the solution should be as close as possible to the target but not raise any errors regarding position/velocity/acceleration limits. 

There's two ways of adapting the trajectory to the target velocity (that I can think of). The more complex and computationally expensive one that results in a "tighter fit" to the target velocity would be to scale the velocities of all joints by the same multiplier at each time step (one velocity multiplier per timestep) in a way that respects all constraints and is as close as possible to the target. This has a disadvantage that the cartesian velocity of the end effector over time will change. But the average error will be smaller. The second way is easier to implement and faster to compute and would be to just have one velocity multiplier for all joints over all timesteps. This makes it easier because calculating the multiplier required is closed form. I'm not sure if there's always a closed form for the first version but it would definitely be more computation.

**Describe alternatives you've considered**
I'm sure there's some things I'm overlooking here. I had already implemented these (both versions, although my implementation for the first option definitely has room for improvement) and it worked well, so I figured it might be useful to other people.

AndyZe, 2023-10-31:
The standard Time Optimal Trajectory Generation plugin already does what you request. It moves through the given points as quickly as possible, given the vel & accel limits of the robot.  https://github.com/ros-planning/moveit/blob/master/moveit_core/trajectory_processing/include/moveit/trajectory_processing/time_optimal_trajectory_generation.h

Or am I missing something?

antoinedang, 2023-10-31:
> The standard Time Optimal Trajectory Generation plugin already does what you request. It moves through the given points as quickly as possible, given the vel & accel limits of the robot. [...]

I see, thanks for pointing that out. How would that fit into the pilz_industrial_motion_planner? I should have specified, my implementation was specifically for that planner.

AndyZe, 2023-10-31:
Pilz is separate from the other pipelines so I see why you were confused. You'll probably need to get your waypoints from Pilz then manually call `trajectory_processing::computeTimeStamps()`. Note that `computeTimeStamps()` will pass through the waypoints you provide to it, but it might not stick to a predictable path between the waypoints.   :/    I don't know of a good way to get around this limitation.

antoinedang, 2023-11-01:
> Pilz is separate from the other pipelines so I see why you were confused. You'll probably need to get your waypoints from Pilz then manually call `trajectory_processing::computeTimeStamps()`. Note [...]

Okay that makes sense, I was worried I had done all that work for nothing 😅. Would it be useful then to be able to do straight or circular lines on point-to-point paths for the pilz motion planner? I think it should be possible as long as the relative velocities between all joints are kept the same

AndyZe, 2023-11-01:
`but it might not stick to a predictable path between the waypoints`

I think you'll find that in practice, this isn't such a big issue. If it is, then you could provide more dense waypoints. I just mentioned it so you're aware

antoinedang, 2023-11-01:
> `but it might not stick to a predictable path between the waypoints`
> 
> I think you'll find that in practice, this isn't such a big issue. If it is, then you could provide more dense waypoints. I just mentioned it so you're aware

Ok I see, thanks for clarifying then! I'll close the issue.

closed 2023-11-01 by antoinedang

rhaschke, 2023-11-02:
> Pilz is separate from the other pipelines so I see why you were confused. You'll probably need to get your waypoints from Pilz then manually call `trajectory_processing::computeTimeStamps()`.

You simply need to add the planning adapter for time parameterization to the Pilz pipeline config:
```diff
- <arg name="planning_adapters" default=""/>
+ <arg name="planning_adapters" default="default_planner_request_adapters/AddTimeParameterization"/>
```