moveit/moveit #3526: Maximum sample density while generating ACM is too low

salihmarangoz, 2023-11-14:
### Description

Maximum sample density while generating ACM is too low for multi-arm configurations. 

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Binary: 1.1.13-2focal.20231013.200622

### Steps to reproduce
Create a multi-arm configuration using moveit_setup_wizard and use the maximum sample density in the "Self-Collisions" section.

### Expected behaviour
The setup wizard should generate a reliable ACM.

### Actual behaviour
There are "disable_collisions" entries in SRDF for some links that can actually easily collide with each other. For example in a two-arm example, arm-1's last links can collide with arm-2's last links. 

### Possible Solution
Replacing the label GUI item showing the number of samples with a textbox to remove the hardcoded upper limit would be a nice solution. Also, I was able to overcome this problem by increasing the hardcoded upper limit in the code. However, this increased the computation time significantly. So, in the case of the upper limit being set by the user, it may be also good to have an interrupt button.

rhaschke, 2023-11-14:
It's not a bug, it's a feature ;-)
As you noticed, increasing the sample size requires much more computation time. To not block MSA, the number of samples is limited by design. If you have a multi-arm config, you should build an ACM for each robot individually and then compose a large one using some insight to your setup.
Of course, you are welcome to contribute a PR implementing the suggested interrupt button.

salihmarangoz, 2023-11-14:
> It's not a bug, it's a feature ;-) As you noticed, increasing the sample size requires much more computation time. To not block MSA, the number of samples is limited by design. If you have a [...]

Thank you for your answer. I think random sampling is not complete. Additionally, optimizing the distance for the remaining link pairs using a global solver may be a near-complete solution. Just an idea...

For this issue, I will be creating a PR with an upper limit in the textbox and an interrupt button, is that ok?

rhaschke, 2023-11-14:
> For this issue, I will be creating a PR with an upper limit in the textbox and an interrupt button, is that ok?

Your contribution is very welcome.