moveit/moveit #3527: Allowed Collision Matrix not updating using apply_planning_scene service.

BrendanWooAdmetal, 2023-11-14:
### Description

I am attempting to modify the Allowed Collision Matrix (ACM) to prevent collision checking with an object added to the scene.
I have followed the steps for updating the ACM recommended here: [Updating allowed collision matrix used for motion plans](https://github.com/ros-planning/moveit/issues/2818)

I retrieve the ACM using the `get_planning_scene` service, modify the ACM, and send it to the `apply_planning_scene` service.
However, checking `get_planning_scene` after shows the update has not been applied.
Notably I am able to successfully add new objects to the scene using `apply_planning_scene`, but the ACM is not changed.
The same is true if I use the `planning_scene` topic.

### Your environment
* ROS Distro: Noetic
* OS Version:  Ubuntu 20.04
* Binary
* 1.1.13-2focal.20231014.010446

### Steps to reproduce
The following code is a minimal test intended to replicate the issue.
Run it alongside MoveIt.

```
import rospy
import moveit_commander
from moveit_msgs.srv import GetPlanningScene, ApplyPlanningSceneRequest
from moveit_msgs.msg import PlanningScene, AllowedCollisionEntry

def allow_collisions_with_object(obj_name, scene):
    """ Updates the MoveIt PlanningScene using the AllowedCollisionMatrix to ignore collisions for an object """
    # Set up service to get the current planning scene
    service_timeout = 5.0
    _get_planning_scene = rospy.ServiceProxy(
        "get_planning_scene", GetPlanningScene
    )
    _get_planning_scene.wait_for_service(service_timeout)

    request = GetPlanningScene()
    request.components = 0    # Get just the Allowed Collision Matrix
    planning_scene = _get_planning_scene.call(request)
    print(f"\n\n\n--- allowed_collision_matrix before update:{planning_scene.scene.allowed_collision_matrix} ---\n\n\n")

    # Set this object to ignore collisions with all objects. The entry values are not updated
    planning_scene.scene.allowed_collision_matrix.entry_names.append(obj_name)
    for entry in planning_scene.scene.allowed_collision_matrix.entry_values:
        entry.enabled.append(False)
    enabled = [False for i in range(len(planning_scene.scene.allowed_collision_matrix.entry_names))]
    entry = AllowedCollisionEntry(enabled=enabled)  # Test kwarg in constructor
    planning_scene.scene.allowed_collision_matrix.entry_values.append(entry)

    # Set the default entries. They are also not updated
    planning_scene.scene.allowed_collision_matrix.default_entry_names = [obj_name]
    planning_scene.scene.allowed_collision_matrix.default_entry_values = [False]
    planning_scene.scene.is_diff = True # Mark this as a diff message to force an update of the allowed collision matrix
    planning_scene.scene.robot_state.is_diff = True

    planning_scene_diff_req = ApplyPlanningSceneRequest()
    planning_scene_diff_req.scene = planning_scene.scene

    # Updating the Allowed Collision Matrix through the apply_planning_scene service shows no effect.
    # However, adding objects to the planning scene works fine.
    # scene._apply_planning_scene_diff.call(planning_scene_diff_req)
    scene.apply_planning_scene(planning_scene.scene)

    # Attempting to use the planning_scene topic for asynchronous updates also does not work
    # planning_scene_pub = rospy.Publisher("planning_scene", PlanningScene, queue_size=5)
    # planning_scene_pub.publish(planning_scene.scene)

    print(f"\n--- Sent message:{planning_scene.scene.allowed_collision_matrix} ---\n")

    # The planning scene retrieved after the update should have taken place shows the Allowed Collision Matrix is the same as before
    request = GetPlanningScene()
    request.components = 0    # Get just the Allowed Collision Matrix
    planning_scene = _get_planning_scene.call(request)
    print(f"\n--- allowed_collision_matrix after update:{planning_scene.scene.allowed_collision_matrix} ---\n")

if __name__ == "__main__":
    # Moveit scene variable
    scene = moveit_commander.PlanningSceneInterface()
    allow_collisions_with_object("temp_obj_name", scene)
    
```

### Expected behaviour
After the update, the ACM is retrieved. It should now show the changes in the ACM requested in the message.

### Actual behaviour
After the update, the ACM is the same as it was before, with none of our changes to the default entries or entries displayed.

### Backtrace or Console output
```
--- allowed_collision_matrix before update:entry_names: 
  - Gripper_cam_link
  - Gripper_link
  - Link1
  - Link2
  - Link3
  - Link4
  - Link5
  - Link6
  - Mount1
  - base_link
  - robot_body_link
entry_values: 
  - 
    enabled: [False, True, False, False, True, True, True, True, True, False, False]
  - 
    enabled: [True, False, False, False, True, True, True, True, True, False, False]
  - 
    enabled: [False, False, False, True, True, True, False, False, False, True, True]
  - 
    enabled: [False, False, True, False, True, True, False, False, False, True, True]
  - 
    enabled: [True, True, True, True, False, True, True, True, True, True, False]
  - 
    enabled: [True, True, True, True, True, False, True, True, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, False, True, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, True, False, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, True, True, False, False, False]
  - 
    enabled: [False, False, True, True, True, False, False, False, False, False, True]
  - 
    enabled: [False, False, True, True, False, False, False, False, False, True, False]
default_entry_names: []
default_entry_values: [] ---

--- Sent message:entry_names: 
  - Gripper_cam_link
  - Gripper_link
  - Link1
  - Link2
  - Link3
  - Link4
  - Link5
  - Link6
  - Mount1
  - base_link
  - robot_body_link
  - temp_obj_name
entry_values: 
  - 
    enabled: [False, True, False, False, True, True, True, True, True, False, False, False]
  - 
    enabled: [True, False, False, False, True, True, True, True, True, False, False, False]
  - 
    enabled: [False, False, False, True, True, True, False, False, False, True, True, False]
  - 
    enabled: [False, False, True, False, True, True, False, False, False, True, True, False]
  - 
    enabled: [True, True, True, True, False, True, True, True, True, True, False, False]
  - 
    enabled: [True, True, True, True, True, False, True, True, True, False, False, False]
  - 
    enabled: [True, True, False, False, True, True, False, True, True, False, False, False]
  - 
    enabled: [True, True, False, False, True, True, True, False, True, False, False, False]
  - 
    enabled: [True, True, False, False, True, True, True, True, False, False, False, False]
  - 
    enabled: [False, False, True, True, True, False, False, False, False, False, True, False]
  - 
    enabled: [False, False, True, True, False, False, False, False, False, True, False, False]
  - 
    enabled: [False, False, False, False, False, False, False, False, False, False, False, False]
default_entry_names: 
  - temp_obj_name
default_entry_values: [False] ---

--- allowed_collision_matrix after update:entry_names: 
  - Gripper_cam_link
  - Gripper_link
  - Link1
  - Link2
  - Link3
  - Link4
  - Link5
  - Link6
  - Mount1
  - base_link
  - robot_body_link
entry_values: 
  - 
    enabled: [False, True, False, False, True, True, True, True, True, False, False]
  - 
    enabled: [True, False, False, False, True, True, True, True, True, False, False]
  - 
    enabled: [False, False, False, True, True, True, False, False, False, True, True]
  - 
    enabled: [False, False, True, False, True, True, False, False, False, True, True]
  - 
    enabled: [True, True, True, True, False, True, True, True, True, True, False]
  - 
    enabled: [True, True, True, True, True, False, True, True, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, False, True, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, True, False, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, True, True, False, False, False]
  - 
    enabled: [False, False, True, True, True, False, False, False, False, False, True]
  - 
    enabled: [False, False, True, True, False, False, False, False, False, True, False]
default_entry_names: []
default_entry_values: [] ---
```

rhaschke, 2023-11-15:
This works for me as expected. However, note that the object `temp_obj_name`, you want to disable collisions with, needs to be _known_ to the planning scene! So, first add the object and then disable collisions. Not doing so, I get the same result as you got.

BrendanWooAdmetal, 2023-11-22:
Unfortunately, the issue is present even when the object is present in the planning scene. 
The original test I created did have an object present in the scene, but I observed that the ACM did not update even when it was present, and I did not include the adding of the object in the original version of the minimal testing script.

Here is an updated version of the minimal testing script that adds the object to the scene as suggested.

As you can see in the output, the collision object is known to the scene. The code checks that it was added successfully, and the object can be found in the PlanningScene -> world -> collision_objects, before attempting to update the ACM.
However, even with the object present the ACM is not updated.

Can you please share the code that worked for you as expected for comparison?

**Updated testing script:**
```
import rospy
import moveit_commander
from moveit_msgs.srv import GetPlanningScene, ApplyPlanningSceneRequest
from moveit_msgs.msg import PlanningScene, AllowedCollisionEntry
import hardware_interfaces.robot_arm.realman_ROS_api as rmapi

def add_object_to_scene(obj_name, scene, height, radius, x, y, z):
    # Create cylinder
    cylinder_pose = rmapi.new_posestamped(
        # Detection location is front of bottle. Displace cylinder so its center is in the center of the bottle
        x,
        y, 
        z,
        0, 0, 0,
        "robot_body_link")
    
    scene.add_cylinder(obj_name, cylinder_pose, height, radius)
    
    # Check and make sure the collision object was added to the scene
    return check_collision_obj_exist(obj_name, scene, True, False)

def check_collision_obj_exist(obj_name, scene, obj_is_known, obj_is_attached, timeout=3):
        start = rospy.get_time()
        seconds = rospy.get_time()
        while (seconds - start < timeout) and not rospy.is_shutdown():
            # Test if the box is in the scene.
            # Note that attaching the box will remove it from known_objects
            is_known = obj_name in scene.get_known_object_names()
            
            # Test if the box is in attached objects
            attached_objects = scene.get_attached_objects([obj_name])
            is_attached = len(attached_objects.keys()) > 0

            # Test if we are in the expected state
            if (obj_is_known == is_known) and (obj_is_attached == is_attached):
                rospy.loginfo(f"[Check collision box] attached matching {obj_is_attached == is_attached}; known matching {obj_is_known == is_known}")
                return True

            # Sleep so that we give other threads time on the processor
            rospy.sleep(0.1)
            seconds = rospy.get_time()

        # If we exited the while loop without returning then we timed out
        return False

def allow_collisions_with_object(obj_name, scene):
    """ Updates the MoveIt PlanningScene using the AllowedCollisionMatrix to ignore collisions for an object """
    # Set up service to get the current planning scene
    service_timeout = 5.0
    _get_planning_scene = rospy.ServiceProxy(
        "get_planning_scene", GetPlanningScene
    )
    _get_planning_scene.wait_for_service(service_timeout)

    request = GetPlanningScene()
    request.components = 0    # Get just the Allowed Collision Matrix
    planning_scene = _get_planning_scene.call(request)
    print(f"\n--- allowed_collision_matrix before update:{planning_scene.scene.allowed_collision_matrix} ---\n")
[... 187 lines cut ...]
  - 
    enabled: [False, True, False, False, True, True, True, True, True, False, False]
  - 
    enabled: [True, False, False, False, True, True, True, True, True, False, False]
  - 
    enabled: [False, False, False, True, True, True, False, False, False, True, True]
  - 
    enabled: [False, False, True, False, True, True, False, False, False, True, True]
  - 
    enabled: [True, True, True, True, False, True, True, True, True, True, False]
  - 
    enabled: [True, True, True, True, True, False, True, True, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, False, True, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, True, False, True, False, False]
  - 
    enabled: [True, True, False, False, True, True, True, True, False, False, False]
  - 
    enabled: [False, False, True, True, True, False, False, False, False, False, True]
  - 
    enabled: [False, False, True, True, False, False, False, False, False, True, False]
default_entry_names: []
default_entry_values: [] ---
```

rhaschke, 2023-11-24:
I can confirm the issue now. I don't know why/how it worked for me the last time?!? I will have a look asap.

rhaschke, 2023-11-25:
False alarm: Modifying the ACM works as expected, but you did not attempt to _allow_ collisions but just _forbid_ them.
Because the ACM only stores deviations from the default (which is considering collisions of course), your changes were not visible.
<details>
 <summary>Here is the code I used</summary>
<pre>
import rospy
from moveit_commander import PlanningSceneInterface
from moveit_msgs.srv import GetPlanningScene, ApplyPlanningSceneRequest
from moveit_msgs.msg import PlanningScene, AllowedCollisionEntry
from geometry_msgs.msg import PoseStamped

def add_object_to_scene(obj_name, psi: PlanningSceneInterface):
    pose = PoseStamped()
    pose.header.frame_id = "panda_link0"
    pose.pose.position.x = 0.5
    pose.pose.position.z = 0.25
    pose.pose.orientation.w = 1.0
    psi.add_box(obj_name, pose, [0.5, 0.5, 0.5])

def get_acm():
    # Set up service to get the current planning scene
    service_timeout = 5.0
    _get_planning_scene = rospy.ServiceProxy("get_planning_scene", GetPlanningScene)
    _get_planning_scene.wait_for_service(service_timeout)

    request = GetPlanningScene()
    request.components = 0  # Get just the Allowed Collision Matrix
    scene = _get_planning_scene.call(request).scene
    print(f"\n--- ACM:\n{scene.allowed_collision_matrix}\n")
    return scene.allowed_collision_matrix

def set_acm_for_object(acm, obj, other=None, enabled=False):
    """Updates the MoveIt PlanningScene using the AllowedCollisionMatrix to ignore collisions for an object"""
    if other is None:
        set_acm_default_for_object(acm, obj, enabled)
        return

    other_idx = acm.entry_names.index(other)
    if obj not in acm.entry_names:
        acm.entry_names.append(obj)
        for entry in acm.entry_values:
            entry.enabled.append(enabled)
        acm.entry_values.append(
            AllowedCollisionEntry(
                enabled=[enabled for i in range(len(acm.entry_names))]
            )
        )
        acm.entry_values[-1].enabled[other_idx] = enabled
    else:
        obj_idx = acm.entry_names.index(obj)
        acm.entry_values[obj_idx].enabled[other_idx] = enabled
        acm.entry_values[other_idx].enabled[obj_idx] = enabled

def set_acm_default_for_object(acm, obj, enabled=False):
    if obj not in acm.default_entry_names:
        acm.default_entry_names.append(obj)
        acm.default_entry_values.append(enabled)
    else:
        idx = acm.default_entry_names.index(obj)
        acm.default_entry_values[idx] = enabled

def apply_acm(psi, acm):
    req = ApplyPlanningSceneRequest()
    req.scene.allowed_collision_matrix = acm
    req.scene.is_diff = True
    req.scene.robot_state.is_diff = True
    psi._apply_planning_scene_diff.call(req)

if __name__ == "__main__":
    rospy.init_node("acm_test_node", anonymous=True)

    # MoveIt PSI
    psi = PlanningSceneInterface()
    add_object_to_scene("obj", psi)
    input("Added object, continue?")

    acm = get_acm()
    set_acm_for_object(acm, "obj", enabled=True)
    apply_acm(psi, acm)

    get_acm()
</pre>
</details>

I will augment the existing python API for the PSI and add some unit tests...

BrendanWooAdmetal, 2023-12-01:
I have retested, and this was indeed the issue.
I had assumed any update to the ACM would be reflected.
Thank you very much for your help.

closed 2023-12-01 by BrendanWooAdmetal

rhaschke, 2024-01-25:
Please open a new issue/feature request against [MoveIt2](https://github.com/ros-planning/moveit2) and refer to the existing solution in MoveIt 1. Ideally, you would reference the corresponding PR/commit that should be ported.

PR #3534 (2024-02-06): Augment Python PlanningSceneInterface API
This PR addresses several shortcomings of the current python API for PSI. See individual commit messages for details.
Closes #3527