moveit/moveit #3533: Multiple planning plugins available. You should specify the '~planning_plugin' parameter. using ur3e

SebiGoebel, 2023-11-25:
I am having a similar issue. I am tyring to run benchmarks with a ur3e and also getting this message:
```
[ INFO] [1651585062.424733103]: Multiple planning plugins available. You should specify the '~planning_plugin' parameter. Using 'chomp_interface/CHOMPPlanner' for now.
[ INFO] [1651585062.436282821]: Using planning interface 'CHOMP'
```
However when i look at the `planning_plugins` and the `default_planning_pipeline` I get the following:
```
$ rosparam list | grep planning_plugin
/move_group/planning_plugin
$ rosparam get /move_group/default_planning_pipeline
''

``` 
I already tried renaming the launch-file from `XXX_planning_pipeline.launch.xml` to `XXX_planning_pipeline.launch`
How can I fix this issue? How can I set the `planning_plugin` ?

rhaschke, 2023-11-26:
Please link the ur3e moveit_config you are using. Which roslaunch command of the tutorial is failing for you?
Running [panda_moveit_config's](https://github.com/ros-planning/panda_moveit_config.git) [demo.launch ](https://github.com/ros-planning/panda_moveit_config/blob/noetic-devel/launch/demo.launch), I have the following plugin parameters:
```sh
$ rosparam list | grep plugin
/move_group/planning_pipelines/chomp/planning_plugin
/move_group/planning_pipelines/ompl/planning_plugin
/move_group/planning_pipelines/pilz_industrial_motion_planner/planning_plugin
```
and move_group's `default_planning_pipeline` is `ompl`:
```SH
$ rosparam get /move_group/default_planning_pipeline
ompl
```

corcasta, 2024-03-24:
I'm facing the exact same problem using ros-noetic following the benchmark tutorial step 10  produces such broken behavior. Actually I face 2 problems: first the one described in this issue and the second once it finish benchmarking it does not save the results and the node dies.

[image]