moveit/moveit #3535: move_group crash

dbdxnuliba, 2023-11-27:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: |Noetic]
* OS Version: e.g. Ubuntu 2004
* Source or Binary build? BInary
* If binary, which release version? 1.1.13
* If source, which branch?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
[ INFO] [1701055950.510498840]: arm/arm[ABITstar]: Finished with a solution of cost 2.7070 (4 vertices) found from 111300 samples by processing 36152 edges (153 collision checked) to create 105 vertices and perform 9 rewirings. The final graph has 54 vertices.
[move_group-3] process has died [pid 52879, exit code -11, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/robot/.ros/log/d341e5a0-8cd3-11ee-9113-67298751ea94/move_group-3.log].
log file: /home/robot/.ros/log/d341e5a0-8cd3-11ee-9113-67298751ea94/move_group-3*.log

rhaschke, 2023-11-27:
This issue report is not yet helpful. Please describe how to reproduce the crash!

dbdxnuliba, 2023-11-27:
when I use moveit   to plan and execute sometime move_group will crash ,but some it will not crash

rhaschke, 2023-11-27:
If we cannot reproduce the issue, we cannot work on it. Please provide example code (ideally on the basis of panda_moveit_config) that crashes reliably (even only after several planning attempts).

dbdxnuliba, 2023-11-29:
and again it crashed，the info
[ INFO] [1701250771.074355678]: arm/arm[ABITstar]: Finished with a solution of cost 6.5472 (5 vertices) found from 37600 samples by processing 2204 edges (24 collision checked) to create 25 vertices and perform 0 rewirings. The final graph has 9 vertices.
[move_group-5] process has died [pid 96963, exit code -11, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/robot/.ros/log/1a464112-8e9b-11ee-b22e-8b918045474c/move_group-5.log].
log file: /home/robot/.ros/log/1a464112-8e9b-11ee-b22e-8b918045474c/move_group-5*.log

dbdxnuliba, 2023-12-06:
@alaflaquiere ,please have a look at the issue ,please

dbdxnuliba, 2024-01-03:
> I'll have a look as soon as I find the time. Unfortunately, if there's no backtrace, I'm afraid the investigation will be difficult.
> 
> Just as a sanity check, is the OMPL version used post-[562f3ac](https://github.com/ompl/ompl/commit/562f3ac562f2c3bdcaaab7cd81da4fe00982b598) commit?

@rhaschke , is the ompl version used post-[562f3ac](https://github.com/ompl/ompl/commit/562f3ac562f2c3bdcaaab7cd81da4fe00982b598) commit?

Hugal31, 2024-01-31:
> I use rviz to MotionPlanning plugin to plan and execute the arm,sometimes again it crashed，
> 
> the info
> 
> ```
> [ INFO] [1701250771.063615455]: arm/arm[ABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 100 samples.
> [ INFO] [1701250771.065020831]: arm/arm[ABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 100 samples.
> [ INFO] [1701250771.066118752]: arm/arm[ABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 99 samples.
> [ INFO] [1701250771.070224801]: arm/arm[ABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 100 samples.
> [ INFO] [1701250771.070479578]: arm/arm[ABITstar]: Finished with a solution of cost 6.5732 (4 vertices) found from 36800 samples by processing 1827 edges (17 collision checked) to create 18 vertices and perform 0 rewirings. The final graph has 5 vertices.
> [ INFO] [1701250771.071309724]: arm/arm[ABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 100 samples.
> [ INFO] [1701250771.073293984]: arm/arm[ABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 99 samples.
> [ INFO] [1701250771.073907860]: arm/arm[ABITstar]: Finished with a solution of cost 6.8740 (4 vertices) found from 30200 samples by processing 7880 edges (99 collision checked) to create 90 vertices and perform 7 rewirings. The final graph has 70 vertices.
> [ INFO] [1701250771.074355678]: arm/arm[ABITstar]: Finished with a solution of cost 6.5472 (5 vertices) found from 37600 samples by processing 2204 edges (24 collision checked) to create 25 vertices and perform 0 rewirings. The final graph has 9 vertices.
> [move_group-5] process has died [pid 96963, exit code -11, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/robot/.ros/log/1a464112-8e9b-11ee-b22e-8b918045474c/move_group-5.log].
> log file: /home/robot/.ros/log/1a464112-8e9b-11ee-b22e-8b918045474c/move_group-5*.log
> ```

This looks a lot like [OMPL#949](https://github.com/ompl/ompl/issues/949).