moveit/moveit #3553: Acceleration/Velocity limits using PILZ planner

shobhitagg, 2024-01-15:
### Description
I am trying to use PILZ planner for cartesian trajectories via `/plan_kinematic_path service`. For some pose goals, it works well, but for others, I get an error for exceeding joint acceleration. If I am using a scaling factor<1, how does it even plan for acceleration greater than what's specified in the joint_limits.yaml file. Is the service using some other reference for planning? or not using any at all?

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04

### Steps to reproduce
Below is a code snippet that I am using to call `plan_kinematic_path` service and execute trajectory.

```c++
	moveit_msgs::MotionPlanRequest mp_req;
	moveit_msgs::MotionPlanResponse mp_res;

	mp_req.pipeline_id = "pilz_industrial_motion_planner";
	mp_req.planner_id = "LIN";
	mp_req.group_name = "manipulator";
	mp_req.num_planning_attempts = 5;
	mp_req.max_acceleration_scaling_factor = 0.3;

	float pos_tolerance = 0.001;
	float orient_tolerance = 0.01;
	// move_group_.constructMotionPlanRequest(mp_req);
	// convert geometry_msgs::Pose to geometry_msgs::PoseStamped
	geometry_msgs::PoseStamped target_pose_stamped;
	target_pose_stamped.header.frame_id = move_group_.getPlanningFrame();
	target_pose_stamped.pose = req.pose;
	moveit_msgs::Constraints pose_goal = kinematic_constraints::constructGoalConstraints(move_group_.getEndEffectorLink(), target_pose_stamped, pos_tolerance, orient_tolerance);
	mp_req.goal_constraints.push_back(pose_goal);

	ros::ServiceClient plan_kinematics_path_client = nh_.serviceClient<moveit_msgs::GetMotionPlan>("plan_kinematic_path");
	plan_kinematics_path_client.waitForExistence();
	moveit_msgs::GetMotionPlan srv;
	srv.request.motion_plan_request = mp_req;

	moveit_msgs::MoveItErrorCodes ret_val;
	ret_val.val = moveit_msgs::MoveItErrorCodes::TIMED_OUT;

	while (ret_val.val != moveit_msgs::MoveItErrorCodes::SUCCESS)
	{
		if (plan_kinematics_path_client.call(srv))
		{
			mp_res = srv.response.motion_plan_response;
			if (mp_res.error_code.val != moveit_msgs::MoveItErrorCodes::SUCCESS)
			{
				std::cerr << "Planning failed due to ERROR CODE = " << mp_res.error_code.val << std::endl;
			}
			else
			{
				std::cout << "Planning successful\n";
				moveit::planning_interface::MoveGroupInterface::Plan plan;
				plan.trajectory_ = mp_res.trajectory;
				ret_val = move_group_.execute(plan);
			}
		}
		else
		{
			ROS_ERROR("Failed to call service plan_kinematics_path");
			return false;
		}
	}
```

### Expected behaviour
Receive a trajectory with SUCCESS code from the service

### Actual behaviour
Planning error due to joint acceleration limit exceeded.

### Backtrace or Console output
```shell
[ INFO] [1705352556.851481396] [/move_group]: Using planning pipeline 'pilz_industrial_motion_planner'
[ INFO] [1705352556.851516577] [/move_group]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[ INFO] [1705352556.851538535] [/move_group]: Generating LIN trajectory...
[ERROR] [1705352556.852993342] [/move_group]: Joint acceleration limit of joint_1_s violated. Set the acceleration scaling factor lower! Actual joint acceleration is -312.634, while the limit is 100.
[ERROR] [1705352556.853009923] [/move_group]: Inverse kinematics solution at 1s violates the joint velocity/acceleration/deceleration limits.
[ERROR] [1705352556.853034762] [/move_group]: Failed to generate valid joint trajectory from the Cartesian path
```

rr-mark, 2024-01-17:
I have just run into this problem too.

Looking through the source code, my understanding of what is going on is:

 1. The velocity profile is generated using `TrajectoryGenerator::cartesianTrapVelocityProfile`, which considers only the limits scaling factors in cartesian space, and not the joint limits.

 2. Then `computePoseIK` is used to convert the cartesian poses path into a joint values.

 3. Then `verifySampleJointLimits` is compares the generated joint values against the joint limits, and throws an error if the joint limits are exceeded.

Is this behaviour as intended? I would naively have expected Pilz to generate a trajectory which respects both cartesian and joint limits, rather than throwing an error if joint limits are exceeded? I'm happy to put some time into improving this behaviour if this is not working as intended.

rr-mark, 2024-01-17:
This issue is made worse by the time sampling in `pilz_industrial_motion_planner::generateJointTrajectory`.

My understanding is:
- The `time_samples` vector is the times at which the trajectory is sampled (between 0 and trajectory.Duration).
- `sampling_time` specifies the time between samples.
- Rather than evenly spacing the samples, `time_samples` is `{0, sampling_time, 2*sampling_time, ...,  n*sampling_time, Duration}`.
- If `Duration` is only slightly larger than `n*sampling_time`, tiny IK differences between the penultimate and final points are massively amplified, leading to large breaches of the joint velocity limits.

In the code I am running, I am seeing `sampling_time=0.1` and `trajectory.Duration=0.504`, so the final step is only `0.004` long, and the limits are basically always violated because of small variations in the IK.

rhaschke, 2024-01-25:
> I would naively have expected Pilz to generate a trajectory which respects both cartesian and joint limits, rather than throwing an error if joint limits are exceeded? I'm happy to put some time into improving this behaviour if this is not working as intended.

I agree that planning a trajectory that respects both, Cartesian and joint limits would be highly desirable. Obviously, the Pilz guys didn't implement it that way. Your contribution to improve is highly welcome.

My suggestion is to scale down the timing of a trajectory (or a segment only?) if joint limits are exceeded. If you see a better approach to directly calculate trajectories that respect both kind of limits, this is highly welcome as well.

v4hn, 2024-01-26:
Imo, the key issue that provoked your question is that the PILZ planner [abuses `max_*_scaling_factor`](https://github.com/ros-planning/moveit/blob/9440572fe08618f643d3c93105f193ee58cbb7d5/moveit_planners/pilz_industrial_motion_planner/src/trajectory_generator.cpp#L274-L275) (1.) to apply in Cartesian space whereas [they are documented to apply only to the limits in joint space](https://github.com/ros-planning/moveit_msgs/blob/9756797dcbc293c7cae66541a7bc0c7acec85892/msg/MotionPlanRequest.msg#L45-L52) and (2.) to specify required Cartesian speeds when they are expected to constitute **upper limits**.

Both decisions where clearly taken because they wanted (or where asked to) provide their interface through [`MotionPlanRequest.msg`](https://github.com/ros-planning/moveit_msgs/blob/master/msg/MotionPlanRequest.msg) (which only really makes sense for the PTP module in PILZ).
For the `CIRC` planner they used the fuzzy semantics of the [`path_constraints` field](https://github.com/ros-planning/moveit/blob/9440572fe08618f643d3c93105f193ee58cbb7d5/moveit_planners/pilz_industrial_motion_planner/src/trajectory_generator_circ.cpp#L67-L75) with very custom meaning, which might have been the slightly better approach for `LIN` too.

> Is this behaviour as intended? I would naively have expected Pilz to generate a trajectory which respects both cartesian and joint limits, rather than throwing an error if joint limits are exceeded? I'm happy to put some time into improving this behaviour if this is not working as intended.

I remember the discussion in the maintainer meeting back then on exactly that point and it was an intentional design from their side. It's a way of getting around the joint-space-jump detection for waypoint-based trajectory planning (as we apply it in `computeCartesianPath`).
If you also consider the option to slow down your Cartesian trajectory to make it adhere to the joint limits, you also have to make sure that you did not accidentally pick consecutive waypoints from different IK manifolds and the robot goes off-path between the points.

There are other ways to handle this, e.g. through, subsampling violating waypoint segments repeatedly with a minimal resolution threshold before failure, optimizing waypoints with fixed distances in joint space (and variable duration), requiring a (very small) maximum angle between the Jacobian-projected normalized joint space delta and the required Cartesian direction, ... In my opinion all of them go beyond the current PILZ planner and have their own issues in the small print.

captain-yoshi, 2024-03-28:
Probably not related to this case, but one can have this issue if one of the joint position is off limit. E.g. Trac-IK will give another solution branch, which will give high acceleration. For this case, the error is a valid.

Danilrivero, 2024-06-27:
Commenting for traceability and check on the issue. Same problems arise in MoveIt2 here https://github.com/moveit/moveit2/issues/2741 and https://github.com/moveit/moveit2/issues/2741. If a fix in MoveIt1 is made making a backport to MoveIt2 would be great to solve those.

I could solve the increasing time problem by applying the fix proposed by https://github.com/moveit/moveit/issues/3583#issuecomment-2035540755. I only encounter the acceleration limits problem on the real robot.

Checking as well to ask if any outcome or decision came from the PR at https://github.com/moveit/moveit/pull/3554