moveit/moveit #3555: Cannot compile MoveIt Controller Handle Allocator plugin

01binary, 2024-01-19:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source build of MoveIt
* Master branch
* Following directions in https://moveit.ros.org/install/source/

### Steps to reproduce

Following directions for writing a Controller Handle Allocator plug-in outlined in this help topic:
https://moveit.ros.org/documentation/plugins/#controllerhandleallocator

Here is the relevant section of the topic for convenience:

> By implementing the [ControllerHandle.h](https://github.com/ros-planning/moveit/blob/4ac0c7432d335f57aab6836cbcaaac3fccf4b6f9/moveit_plugins/moveit_ros_control_interface/include/moveit_ros_control_interface/ControllerHandle.h) interface it is possible to offer allocations of handlers for action based controllers. The controller handler are based on the class [MoveItControllerHandle](https://github.com/ros-planning/moveit/blob/master/moveit_core/controller_manager/include/moveit/controller_manager/controller_manager.h). These handlers communicate with the controller in order to e.g. send trajectories.

> The interface is defined in [ControllerHandle.h](https://github.com/ros-planning/moveit/blob/4ac0c7432d335f57aab6836cbcaaac3fccf4b6f9/moveit_plugins/moveit_ros_control_interface/include/moveit_ros_control_interface/ControllerHandle.h).

> An exemplary implementation of this interface is the [joint_trajectory_controller_plugin.cpp](https://github.com/ros-planning/moveit/blob/master/moveit_plugins/moveit_ros_control_interface/src/joint_trajectory_controller_plugin.cpp). This is also currently the only implementation available.

I add the following to my `package.xml`:

- `<depend>moveit_core</depend>
- `<depend>moveit_ros_control_interface</depend>`

And the following in `CMakeLists.txt` under `find_package` and `catkin_package` macros:

- `moveit_core`
- `moveit_ros_control_interface`

Then I try to include the following headers:

- `#include <moveit/controller_manager/controller_manager.h>`
- `#include <moveit/moveit_plugins/moveit_ros_control_interface/controller_handle_allocator.h>`
- `#include <moveit/moveit_plugins/moveit_ros_control_interface/moveit_controller_handle.h>`
- `#include <moveit/moveit_plugins/moveit_ros_control_interface/follow_joint_trajectory_controller_handle.h>`

These headers can also be seen here:

https://github.com/ros-planning/moveit/tree/master/moveit_plugins

### Expected behaviour

The headers are found by CMake because the `${catkin_INCLUDE_DIRS}` CMake variable will contain the location of MoveIt headers.

### Actual behaviour

The headers are not found, and they don't appear to exist anywhere on the system other than in the cloned MoveIt repository in `~/catkin_ws/src` (following MoveIt Source Install documentation).

### Backtrace or Console output

```
fatal error: moveit/moveit_plugins/moveit_ros_control_interface/controller_handle_allocator.h: No such file or directory
```

01binary, 2024-01-23:
After including the `ros_control_interface` as a dependency in `package.xml` and `CMakeLists`, I decided to dump all header paths known by CMake by using this trick:

[How to Print Variable in CMake](https://stackoverflow.com/questions/68139352/how-to-print-variables-in-cmake)

This revealed that bringing in `ros_control_interface` gave me access to just one MoveIt header I really needed:

```
#include <moveit_ros_control_interface/ControllerHandle.h>
```

Basically I just looked for any header paths known to CMake that contained the string "moveit_ros_control_interface", and then followed a pattern used by other headers which were included successfully to find out the `#include` did not start with `moveit/`, etc.

This was of course not documented (but don't worry, I will update the docs when I have more time).

Upon including this header, I gained access to the following namespaces and classes, which is all that was needed to implement a controller handle, and a controller handle allocator:

- `controller_interface::Controller`
- `moveit_controller_manager::MoveItControllerHandle`
- `moveit_ros_control_interface::ControllerHandleAllocator`

With this, MoveIt finally recognized my custom trajectory controller as a controller it can use. You can see my implementation here:

- [trajectoryController.h](https://github.com/01binary/str1ker/blob/master/src/include/trajectoryController.h)
- [trajectoryController.cpp](https://github.com/01binary/str1ker/blob/master/src/trajectoryController.cpp)

Leaving this issue open as a reminder for me to update MoveIt 1 docs.

rhaschke, 2024-01-23:
Not sure, which problem you report here.
The file `controller_handle_allocator.h` doesn't exist (in ROS1).
The file `moveit_ros_control_interface/ControllerHandle.h` does exist.
Why do you think, you need the former?

01binary, 2024-01-23:
I was trying to guess the header name because `#include <moveit_ros_control_interface/ControllerHandle.h>` was not found.

rhaschke, 2024-01-23:
> because `#include <moveit_ros_control_interface/ControllerHandle.h>` was not found

For me, the file exists.

01binary, 2024-01-23:
It could be that the lack of documentation left me confused, let me look into this on the weekend and see if a brand new catkin package can build when including that file.

At the very least, you can see that the MoveIt 1 help page for plugins has a lot of "TODO" placeholders, and this issue can be re-defined to update the placeholder/blurb about how to create MoveIt integration classes to enable a custom trajectory controller to be used with MoveIt.

If you don't like this issue existing due to inaccurate initial post description feel free to close, I can submit a PR to update MoveIt 1 later and don't necessarily need an issue to connect to the PR.

rhaschke, 2024-01-25:
I have changed the issue title. Hope that fits better now. If not, please tell me.
If you run successfully through the process, please improve the docs by providing a PR against the tutorials repo.

01binary, 2024-01-28:
All right, how does this look? It turns out that documenting how to integrate ROS controllers with MoveIt in sufficient depth will probably require a sub-topic (I imagine we'll need sub-topics for how to implement motion planning and IK plugins as well at some point).

https://github.com/01binary/trajectory_controller_example

rhaschke, 2024-01-28:
Thanks for your very detailed description of what you think is used by MoveIt to control a robot. Reading this lengthy description, I've got the impression that you didn't fully understand the inner workings yet.
I just re-checked the existing documentation and found this [low-level controller tutorial](https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html).

MoveIt essentially provides three _controller managers_:
- [`MoveItSimpleControllerManager`](https://github.com//blob/master/moveit_plugins/moveit_simple_controller_manager/src/moveit_simple_controller_manager.cpp) allows to interface any custom controller that implements either the `FollowJointTrajectory` or the `GripperCommand` action interface.
- [`MoveItFakeControllerManager`](https://github.com//blob/master/moveit_plugins/moveit_fake_controller_manager/src/moveit_fake_controller_manager.cpp) doesn't interface a real robot, but just copies the desired to the actual joint values. 
- [`MoveItControllerManager`](https://github.com//blob/master/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp) simply uses all matching controllers from a `ros_control` controller manager instance. This is probably the most commonly used controller manager.

Now the question is: Why do you want to add another controller manager to this list?
What do you actually want to achieve? Surely, the documentation on how to implement a new controller manager is very sparse, but these three classes provide good examples. However, I don't see the need for another controller manager. Probably you just want to implement your own `ros_control` controller?

01binary, 2024-01-28:
> I just re-checked the existing documentation and found this [low-level controller tutorial](https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html).

The [low-level controller tutorial](https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html) page you are referring to is the last one I read before I gave up on the documentation and had to delve into the code to understand how it works.

I don't doubt that it makes perfect sense to you as the co-author of the library. However, there is a sizeable disconnect between the knowledge required to write MoveIt plugins, and the knowledge that can be built up from scratch exclusively by reading documentation. It is this gap I am trying to address.

My issues started when I wrote a ROS Joint Trajectory Controller that exposed an Action Server on the right namespace, but MoveIt was refusing to load it.

When I debugged through MoveIt Simple Controller Manager, I found that it doesn't matter if you write the controller correctly or not, it won't work without a Controller Handle Allocator that's registered with the same name.

It's not possible to re-use the stock controller handle allocator because it's bound to the string `xxx_controllers/JointTrajectoryController` (where xxx is position, velocity, or effort). My controller was not called that.

> Now the question is: Why do you want to add another controller manager to this list?

Here's the outline of the topics that I felt would need to be covered if the "past me" were to have a chance at integrating a custom ROS controller with MoveIt for the first time by reading documentation:

```
- Add package dependencies
- Implement a ROS Trajectory Controller (Boilerplate)
- Export the ROS Trajectory Controller
- Implement a MoveIt Controller Handle
- Implement a MoveIt Controller Handle Allocator
- Export the MoveIt Controller Handle Allocator
- Configure the new ROS Controller
- Test the new ROS Controller and its MoveIt integration
```

The list of topics in the outline does not include a topic on writing another MoveIt controller manager. I suspect that you may have found the references I made to "MoveIt Controller Manager" confusing because there are three controller managers as you outlined (Simple, Fake, and... "regular"?). If that's the issue I can replace all references to "MoveIt Controller Manager" with "MoveIt Simple Controller Manager".

> [MoveItControllerManager](https://github.com//blob/master/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp) simply uses all matching controllers from a ros_control controller manager instance. This is probably the most commonly used controller manager.

If you were to create a new MoveIt Config package using the Setup Assistant, isn't the Simpler Controller Manager the default? I recently created [basic_velocity_control_moveit_config](https://github.com/01binary/basic_velocity_control_moveit_config/blob/main/config/simple_moveit_controllers.yaml) because RRBot was not working for me, and the controllers are loaded from this configuration file:

https://github.com/01binary/basic_velocity_control_moveit_config/blob/main/config/simple_moveit_controllers.yaml

Since the configuration file is called "Simple MoveIt Controllers", would I be right to assume that Simple MoveIt Controller Manager is the default for new MoveIt projects generated by MoveIt Setup Assistant?

If this is true, then "the default" would be equivalent to "the most common" (if all new MoveIt users get Simple Controller Manager by default, and new users are far more numerous than existing users, then Simple MoveIt Controller Manager is the most common?)

rhaschke, 2024-01-29:
I'm sorry that you had trouble getting your custom joint trajectory controller running with MoveIt. However, I still don't really understand your actual problem.

> It's not possible to re-use the stock controller handle allocator because it's bound to the string `xxx_controllers/JointTrajectoryController`

I'm not sure what exactly you are referring to here as "stock controller handle". The class [`ControllerHandleAllocator`](https://github.com/ros-planning/moveit/blob/9440572fe08618f643d3c93105f193ee58cbb7d5/moveit_plugins/moveit_ros_control_interface/include/moveit_ros_control_interface/ControllerHandle.h) is just an abstract base class. One concrete implementation provided by MoveIt is the [`JointTrajectoryControllerAllocator`](https://github.com/ros-planning/moveit/blob/9440572fe08618f643d3c93105f193ee58cbb7d5/moveit_plugins/moveit_ros_control_interface/src/joint_trajectory_controller_plugin.cpp), which is linked to various ROS controllers (`xxx_controllers/JointTrajectoryController`) [here](https://github.com/ros-planning/moveit/blob/9440572fe08618f643d3c93105f193ee58cbb7d5/moveit_plugins/moveit_ros_control_interface/moveit_ros_control_interface_plugins.xml).
In principle, you could link this class to any other name with a custom plugin config XML.

Thus, to summarize, you can easily register any custom trajectory controller implementing the `FollowJointTrajectory` action server API with MoveIt simply by registering it in a plugin xml.

Yes, you could consider the SimpleControllerManager the default. It allows both FollowJointTrajectory actions as provided by appropriate ROS controllers as well as GripperCommand actions.

> > MoveItControllerManager reads what controller(s) to use from a controllers.yaml file.
> 
> This file does not exist in a package auto-generated by MoveIt Setup Assistant. I assume this is referring to `simple_moveit_controllers.yaml` which was renamed at some point.

Maybe, that's the only thing that needs to be clarified in the docs? As I mentioned, there are multiple controller managers. Each of them uses a slightly different yaml file to configure the corresponding ROS parameters. Probably, the docs used the general file name `controllers.yaml` to cover all of them.

If you see improvement to the docs given all this information, please don't hesitate to provide an improving PR for the tutorials.

01binary, 2024-02-02:
> However, I still don't really understand your actual problem.

MoveIt documentation does not clearly explain how controller integration works, and therefore understanding this requires building MoveIt from source and debugging.

The most visible symptom of this issue is a large number of questions about this error:
```
Unable to identify any set of controllers that can actuate the specified joints
```

Some examples:
* This person had to debug MoveIt in order to discover that MoveIt Simple Controller Manager setup by default cannot work with just any ROS controller, only specific ones that have handles:
  * https://answers.ros.org/question/338984/ros-moveit-error-unable-to-identify-any-set-of-controllers-that-can-actuate-the-specified-joints/
* This person found that using a custom ROS controller requires switching to MoveIt ROS Control Manager from MoveIt Simple Controller Manager configured by default (so MoveIt 2 documentation has a similar issue):
  * https://github.com/ros-planning/moveit2/issues/1394
* Still-open issue about the same documentation area:
  * https://github.com/ros-planning/moveit/issues/112
* General confusion about MoveIt integration with controllers
  * https://answers.ros.org/question/327031/setting-up-controllers-with-moveit-no-controllers-found/
  * The responder essentially states that this documentation assumes prior knowledge of the origins of both ROS Control and MoveIt, which could only come from working with PR2 and other robots this code came from (`I guess the assumption is that you already know a thing or two about setting things like this up`)

I think the issue with these two topics ([Low Level Controllers](https://ros-planning.github.io/moveit_tutorials/doc/controller_configuration/controller_configuration_tutorial.html) and [Plugins](https://moveit.ros.org/documentation/plugins/#controllerhandleallocator)) is that different types of configuration are conflated, and the information about controller manager types must be pieced together from fragments, where there is no clear "entry point" (i.e. "I want to get a custom ROS controller working with MoveIt, how do I do that?")

I will look for the right edits to clarify the following:

- There are three types of MoveIt Controller Managers that can be configured for a MoveIt project 
- MoveIt Setup Assistant configures new MoveIt projects for Simple Controller Manager out of the box
- Simple Controller Manager supports only Joint Trajectory Controller and Gripper Action Controller from ROS Controls
- If you want to use your own ROS controller, you must switch to `ros_control` type of MoveIt Controller Manager, from the default `simple` type. If you don't do this, then you have to link an existing MoveIt controller handle for your custom controller or write your own, otherwise it won't be found
- I haven't seen any mention of how to re-configure a MoveIt package created by MoveIt Setup Assistant to switch MoveIt Controller Manager type
- If you want to use a totally custom controller not based on ROS Control, you could write your own MoveIt Controller Manager or use MoveIt Simple Controller Manager and write your own handle/handle allocator
- Controller configuration file names appear to be incorrect: `ros_controllers.yaml` file appears to be conflated with `simple_moveit_controllers.yaml` file, and referred to as `controllers.yaml` which doesn't exist

01binary, 2024-02-12:
Posted the third draft of my proposed improvements in the appropriate `moveit_tutorials`.