moveit/moveit #3569: Sequence planning in PILZ has error growth in attached collision object poses.

rr-mark, 2024-02-13:
### Description

As far as I understand:

Background:
 - The sequence planning capability (plan sequence service, plan sequence action etc.) plans a series of connected motions.
 - The end state from each motion is used to seed the start state of the next motion.
 - This update should really only update the joint state, but in practice the whole robot state is regenerated each time, including a conversion from `robot_state::RobotState` to `moveit_msgs::RobotState`.

The problem:
 - The poses of the attached collision objects in the start state (which should be constant) change between each waypoint in the sequence. This change is initially minuscule, but the rate of change grows and eventually (waypoint ~30 in one path we've seen, later in other paths) this error becomes very large (~several degrees).

My suspicions:
 - The attached collision object poses are being converted back and forth between different representations. This conversion involves FK and so is slightly lossy, and these errors compound exponentially each time the poses are re-calculated. I haven't been able to track down where this might be happening.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source build - 1.1.13-c09acf72487071593c9a60edae0357e7ceb87210

### Steps to reproduce
Call pilz' `plan_sequence_path` service with a toolpath containing many waypoints, using an attached collision object as the planning frame.

The motion plan will drift away from the toolpath, with increasing drift as the waypoints increase.

This can be observed more directly by putting a print statement in `TrajectoryGenerator::generate` in `trajectory_generator.cpp`, to look at the poses in `req.start_state.attached_collision_objects[i].pose` for some `i`.

### Expected behaviour
The poses of attached collision objects should remain rigid, and not vary along a sequence plan.

### Actual behaviour
The poses of attached collision objects gain numerical error with each step along a sequence plan. This error compounds exponentially.

### Backtrace or Console output
N/A

rr-mark, 2024-02-13:
I also suspect this problem might have been introduced / exposed in https://github.com/ros-planning/moveit/pull/3519

rr-mark, 2024-02-14:
So `CommandListManager::solveSequenceItems` needs to update `req.start_state` using the output of `getPreviousEndState(motion_plan_responses)`.

At the moment, it calls `robotStateToRobotStateMsg` to generate a whole new `req.start_state` from the previous end state.

I wonder if it would be better to use the old `req.start_state`, but updated with the changed joints from the new state, rather than overwriting it entirely?

Some kind of
```
void updateRobotState(moveit_msgs::RobotState& start_state, const robot_state::RobotState& previous_end_state);
```
which would probably also need to take some kind of joint model group to know what to update?

rhaschke, 2024-04-29:
Fixed via #3590

closed 2024-04-29 by rhaschke