moveit/moveit #3579: Starting MongoDB constraint database node leads to class_loader::LibraryUnloadException

stevenoh2003, 2024-03-18:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04

### Steps to reproduce
Steps can be reproduced by simplying running the `default_warehouse_db.launch` file inside your moveit_config folder generated by moveit.

### Expected behaviour
It should just launch a warehouse database server using MongoDB without outputting any errors.

### Actual behaviour
It prints out error

```terminate called after throwing an instance of 'class_loader::LibraryUnloadException'
  what():  Attempt to unload library that class_loader is unaware of.```

rhaschke, 2024-03-18:
I can run `roslaunch moveit_resources_panda_moveit_config default_warehouse_db.launch` w/o issues.
Please provide more details.

stevenoh2003, 2024-03-18:
> I can run `roslaunch moveit_resources_panda_moveit_config default_warehouse_db.launch` w/o issues. Please provide more details.

Sorry, I just updated my post. You also have to set the "reset" argument to be "true". (it is false by default)

rhaschke, 2024-03-18:
Yes, now I can reproduce the error message.

rhaschke, 2024-03-19:
I shortly looked into this. You can safely ignore this error message. It occurs during the shutdown of the database initialization program while releasing a static resource, which should have been released earlier.
Nevertheless, I filed a PR to fix that issue: #3581

stevenoh2003, 2024-03-20:
> I shortly looked into this. You can safely ignore this error message. It occurs during the shutdown of the database initialization program while releasing a static resource, which should have been released earlier.
> 
> Nevertheless, I filed a PR to fix that issue: #3581

Thank you for the PR!

closed 2024-04-16 by rhaschke