moveit/moveit #3583: Pilz trajectory: Trajectory message contains waypoints that are not strictly increasing in time

captain-yoshi, 2024-03-28:
### Description
This does not hapen often, but sometimes the last two trajectory `time_from_start` are identical with Pilz LIN.
```bash
Trajectory message contains waypoints that are not strictly increasing in time
```
Trajectory points.
```bash
[ WARN] [1711638790.262639288]: time from start = 0.000000000
[ WARN] [1711638790.262673428]: time from start = 0.076080427
[ WARN] [1711638790.262707381]: time from start = 0.207295897
[ WARN] [1711638790.262737042]: time from start = 0.331884750
[ WARN] [1711638790.262772077]: time from start = 0.372734245
[ WARN] [1711638790.262804160]: time from start = 0.372734245
```

I suspect that this has to do with the uneven sample time from #3553. Will dig deeper because can reproduce deterministically on my setup.

liujun0808, 2024-04-02:
hello！I also encountered this problem, when i controlled the physical robot arm, did you find a solution?

captain-yoshi, 2024-04-03:
Cannot reproduce anymore.

You can bypass it by adding the check below which adds the epsilon value to the end waypoint `time_from_start` if the last 2 waypoints are equal.

```cpp
  // File: moveit/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp
  // Methods: pilz_industrial_motion_planner::generateJointTrajectory
  if (joint_trajectory.points.size() >= 2)
  {
    if ((joint_trajectory.points.end() - 1)->time_from_start == (joint_trajectory.points.end() - 2)->time_from_start)
    {
      ROS_WARN("last 2 waypoints time from start are equal");

      auto& point = joint_trajectory.points[joint_trajectory.points.size() - 1];
      point.time_from_start = point.time_from_start + ros::Duration(epsilon);
    }
  }
  
  error_code.val = moveit_msgs::MoveItErrorCodes::SUCCESS;
  double duration_ms = (ros::Time::now() - generation_begin).toSec() * 1000;
  ROS_DEBUG_STREAM("Generate trajectory (N-Points: " << joint_trajectory.points.size() << ") took " << duration_ms
                                                     << " ms | " << duration_ms / joint_trajectory.points.size()
                                                     << " ms per Point");

  return true;
}
```

liujun0808, 2024-04-05:
> Cannot reproduce anymore.
> 
> You can bypass it by adding the check below which adds the epsilon value to the end waypoint `time_from_start` if the last 2 waypoints are equal.
> 
> ```c++
>   // File: moveit/moveit_planners/pilz_industrial_motion_planner/src/trajectory_functions.cpp
>   // Methods: pilz_industrial_motion_planner::generateJointTrajectory
>   if (joint_trajectory.points.size() >= 2)
>   {
>     if ((joint_trajectory.points.end() - 1)->time_from_start == (joint_trajectory.points.end() - 2)->time_from_start)
>     {
>       ROS_WARN("last 2 waypoints time from start are equal");
> 
>       auto& point = joint_trajectory.points[joint_trajectory.points.size() - 1];
>       point.time_from_start = point.time_from_start + ros::Duration(epsilon);
>     }
>   }
>   
>   error_code.val = moveit_msgs::MoveItErrorCodes::SUCCESS;
>   double duration_ms = (ros::Time::now() - generation_begin).toSec() * 1000;
>   ROS_DEBUG_STREAM("Generate trajectory (N-Points: " << joint_trajectory.points.size() << ") took " << duration_ms
>                                                      << " ms | " << duration_ms / joint_trajectory.points.size()
>                                                      << " ms per Point");
> 
>   return true;
> }
> ```

Thanks for your reply. I use a loop to check whether the penultimate element and the penultimate element in the points list are equal. If they are equal, I will delete the last element. If they are not equal, I will jump out of the loop. It feels similar to the method you mentioned. It can run smoothly now and no errors have been reported yet.