moveit/moveit #3584: Collision Checking Request Not Respecting Group Name with Bullet

raghavsood1996, 2024-03-28:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20
* Binary Build
* 1.1.13

### Steps to reproduce
* Launch a robot that has multiple move groups defined.
* Launch in a sate that some links in one of the group are colliding
* Now run a collision checking query for a different group that is not in collision using Bullet

`
  collision_detection::CollisionRequest req;
  collision_detection::CollisionResult res;
  req.group_name = group_name_;
  verbose = true;

  if (verbose)
  {
    req.contacts = true;
  }
  local_planning_scene_ptr_->setActiveCollisionDetector(collision_detection::CollisionDetectorAllocatorBullet::create(), true);
  local_planning_scene_ptr_->checkCollision(req, res, robot_state);
  if (verbose && res.collision)
  {
    ROS_WARN_STREAM("Contact points are");
    for (auto& pr : res.contacts)
    {
      ROS_WARN_STREAM(pr.first.first << " " << pr.first.second << " with collision distance of " << pr.second.front().depth);
    }
  }
  return !res.collision;
`

### Expected behaviour
The collision checking should return false

### Actual behaviour
Collision checking returns true and print lins in collision that re not part of this group for which we are checking collisions

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

simonschmeisser, 2024-05-13:
Bullet collision checking plugin is quite incomplete unfortunately. There is a PR open to improve exactly this, please feel invited to finish it :smile:

simonschmeisser, 2024-05-13:
#2502

simonschmeisser, 2024-05-26:
We still do releases for noetic from time to time, no worries