moveit/moveit #3586: How to use python to control the franka gripper

upc-ghy, 2024-04-09:
Hello,

I just started learning about Franka related knowledge, can anyone give me some guidance on using Python to control the franka gripper?

ChatGPT told me to pip install frank_gripper, but it prompted that I cannot import it. I want to useimport franka_gripperdirectly in python to solve the problem, but I was told that I don't have this package.

I have already installed libfranka, franka_ros, and moveit, and can control Franka's robotic arm by import moveit_commander in Python.

My device is:
PC
Ubuntu 20.04
Ros noetic

rhaschke, 2024-04-09:
This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here. 
If you can operate the Franka arm through MoveIt, you can do so with the gripper as well. Just switch to the `panda_hand` group instead of `panda_arm`.

closed 2024-04-09 by rhaschke

upc-ghy, 2024-04-10:
@rhaschke 
Thank you very much for your guidance and valuable time. Wishing you all the best in your work!