moveit/moveit #3588: orientation path constraint planning bug

vegfood, 2024-04-17:
### Description

Even an orientation constraint is set, it appears that the desired_rotation_matrix_inv_,  which is defined in kinematic_constraint.h, is always a unit matrix while the OrientationConstraint::decide  function is running. If I understand correctly, its value should be the inverse of the desired rotation matrix. I also notice that desired rotation matrix is exactly the value of my orientation constraint, which is correct. I tested this using the move_group python interface.

* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source code
* master

rhaschke, 2024-04-23:
`desired_rotation_matrix_inv_` is only used if the reference frame is a fixed frame:
https://github.com/ros-planning/moveit/blob/0f90db67a45a0a69d871fc46ba0efba3e51a32b7/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp#L579-L592

Which _concrete_ buggy _behavior_ do you observe?

rhaschke, 2024-04-23:
I just confirmed that planning with an orientation constraint works as expected. Closing. Please feel free to reopen if you still see issue. Please describe your problem in more detail then.

closed 2024-04-23 by rhaschke

vegfood, 2024-04-27:
Here is my robot urdf, I do this in gazebo simulation,  I set the reference frame as the 'base_link'. It appears that it treats the 'base_link' as a mobile frame. I also tested setting the 'world' as the reference frame, and this time the 'world' is treated as a fixed frame. So in moveit, how to determine whether a frame is a fixed frame or not?
[image]

rhaschke, 2024-04-28:
Fixed frames are all frames, which you explicitly declared in MoveIt (plus the robot's root frame).
So, if you didn't declare any frames, you just have the root frame as the base.
You didn't yet describe your genuine problem...

vegfood, 2024-04-29:
Sorry for the confusion. I realized that the reason the expected inverse matrix of the rotation matrix doesn't change is because MoveIt treats my 'base_link' as a moving frame of reference. In my URDF, 'base_link' is connected to the world frame via two fixed joints, so I naturally assumed that 'base_link' would also be considered a fixed frame, but it clearly isn't. So, what should I do to make MoveIt treat my 'base_link' as a fixed frame?

rhaschke, 2024-04-29:
Why do you want MoveIt to consider your joint as a fixed frame? As I noticed above, both moving and fixed frames should work just fine for orientation constraints. But there is some extra optimization for fixed frames - using the inverse matrix. Moving frames don't use that inverse matrix.
I again ask you to provide a concrete error description, if you observe any error. The inverse matrix not being used or initialized is _not_ an error.

vegfood, 2024-04-29:
I initially thought this was a bug in Moveit's handling of orientation constraints. Now I found that it was just because the reference frame 'base_link' I set was treated as a mobile frame by moveit, so the inverse matrix of the expected rotation matrix was always the unit matrix.  As shown in my URDF above, my 'base_link' frame and root frame 'world' are connected through two fixed joints, so in fact it is also a fixed reference frame, but obviously moveit doesn't think so. So, how should I modify my urdf or is there any other way to make moveit realize that this is a fixed frame? Sorry to bother you again.

rhaschke, 2024-04-29:
I still don't get why you want MoveIt think of `base_link` as a "fixed" frame (there is no way to do that). 
Handling orientation constraints should work with `base_link` as a moving frame as well.

vegfood, 2024-04-29:
I'm very confused. Since base_link is connected to the root frame through fixed joints, it is essentially a fixed frame. Why doesn't moveit treat it as a fixed frame? Or does it mean that only the root frame will be regarded as fixed in Moveit?

rhaschke, 2024-04-29:
_Fixed_ frames in the context of constraint handling have a very special meaning in MoveIt: Only manually registered frames as well as the root frame are considered _fixed_. 
Just consider your frame as "moving" in this context. MoveIt doesn't analyze your URDF structure to determine fixed frames. It just considers all robot frames (except the root) as _moving_.

vegfood, 2024-04-29:
Thanks, your explanation is very clear. If all robot frames are regarded as moving, what other frames can be manually registered as fixed? If possible, could you please provide an example of manually registered fixed frame?

rhaschke, 2024-04-29:
I have not seen code using the PlanningScene's transforms object in practice. But essentially, you would need to do this:
```cpp
// get access to the scene's transforms
auto tfs = planning_scene->getTransformsNonConst();
// register new "fixed" transforms - fixed w.r.t. the planning/root frame of the robot
tfs.setTransform(Eigen::Isometry3d::Identity, "my_frame");
// there are multiple other variants of setTransform()
```