moveit/moveit #3596: Add progress to MoveGroupSequence Feedback

rr-tom-noble, 2024-04-29:
**Is your feature request related to a problem? Please describe.**
When planning a sequence of poses, it would be useful to receive information about the number of points planned in the feedback. This allows for measuring the progress of the plan and estimations of the time left to complete.

**Describe the solution you'd like**
I can think of a few options that could be added to the Feedback portion of the action:
  - The number of planned points and total number of points
  - Just the number of planned points (expect user to fetch total points from e.g. the request)
  - The percentage (or fractional equivalent) of planned points 

**Describe alternatives you've considered**
Previously, we moved away from using the GetMotionSequence to performing the bulk of the logic ourselves, however, this proved very slow as it required making a call to MoveGroupInterface::plan() i.e. calling the MoveGroup action for each point.

**Additional context**
Happy to have a go at implementing this myself

rhaschke, 2024-04-29:
I suggest going for option 1: the number of planned and total sub tasks.

rr-tom-noble, 2024-04-29:
I've just raised https://github.com/ros-planning/moveit_msgs/pull/176 on the moveit_msgs side

rr-tom-noble, 2024-04-29:
@rhaschke I can only seem to find reference to the sequence action within the Pilz code. Do you know if this is the only implementation of the action?

rr-tom-noble, 2024-04-29:
Looks like the action server currently blocks on CommandListManager::solve(). I can think of a few ways to update for continuous feedback:

- Add an optional callback to CommandListManager::solve() which is called after every item (or every N items).
- Add a new method CommandListManager::solveNItems() which returns after N items (and retains any state needed to call again).
- Add a new method CommandListManager::solveAsync() which returns a std::future and some extra methods for querying state.