moveit/moveit #3600: crash when run moveit_rviz.launch

ncnynl, 2024-05-13:
### Description

Will crash when starting
  
  <include file="$(find open_manipulator_moveit_config)/launch/moveit_rviz.launch">
    <arg name="config" value="true"/>
    <arg name="debug" value="$(arg debug)"/>
  </include>

[image]

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04

simonschmeisser, 2024-05-13:
This is not a crash but rather it does not even start because some libraries don't match. If you compiled from source, please recompile your entire workspace. If you installed from .deb please check that everything is up to date.

This is actually a question in the template, please fill it out entirely next time. Also pasting errors as text makes them searchable

ncnynl, 2024-05-13:
@simonschmeisser  

Thank you, it has been solved now

I deleted all the moveit versions installed by apt and reinstalled them from source. The version installed from the source code can run normally.

closed 2024-05-13 by rhaschke