moveit/moveit #3609: LoadScene operation causes existing objects to continue to be displayed without disappearing.

Takayuki9999, 2024-06-19:
### Description
When I perform the LoadScene operation in the StoredScenes tab of the Motion Planning Panel for Rviz, the existing objects do not disappear and remain.   
The existing objects and the newly LoadScene'd objects are displayed on Rviz at the same time. I saw that this happens with Moveit (1.1.14).  
I checked the messages on the `/move_group/monitored_planning_scene` topic, but the existing objects are not published. It seems that the existing objects remain displayed on the local Rviz.  
When I press the `RESET` button in the bottom left corner, the existing objects on the screen disappear and only the newly LoadScene'd objects are redisplayed.  
When I was using Moveit (1.1.9), all existing objects disappeared with just the LoadScene operation.  
Has the functionality changed?

Additionally, I noticed that if I use a sqlite DB file created in 1.1.9 as is in 1.1.14, I get an MD5 error during the SaveQuery operation and can't save it. This seems to be related to the change in moveit_msg/MotionPlanRequest.msg in [#130](https://github.com/moveit/moveit_msgs/pull/130).

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Source build
* branch: https://github.com/moveit/moveit/tree/master
* version: 1.1.14

### Steps to reproduce
**Preparation**  
We are using `moveit/warehouse_ros_sqlite` as our DB.  
Configure `panda_moveit_config` to use `warehouse_ros_sqlite`.
1. Install [moveit/warehouse_ros_sqlite](https://github.com/moveit/warehouse_ros_sqlite). 

2. Modify `warehouse.lauch` and `warehouse_setteings.lauch` in `~/ws_moveit/src/pand_moveit_config/launch` to use sqlite.

    **warehouse.lauch**  
    ```diff
      
    -  <node name="$(anon mongo_wrapper_ros)" cwd="ROS_HOME" type="mongo_wrapper_ros.py" pkg="warehouse_ros_mongo">
    -    <param name="overwrite" value="false"/>
    -    <param name="database_path" value="$(arg moveit_warehouse_database_path)" />
    -  </node>
    ```
    **warehouse_setteings.lauch**
    ```diff
    
    -  <arg name="moveit_warehouse_host" default="localhost" />
    +  <arg name="moveit_warehouse_host" default="/tmp/warehouse.sqlite" />

    
    <param name="warehouse_port" value="$(arg moveit_warehouse_port)"/>
    <param name="warehouse_host" value="$(arg moveit_warehouse_host)"/>
    -  <param name="warehouse_exec" value="mongod" />
    -  <param name="warehouse_plugin" value="warehouse_ros_mongo::MongoDatabaseConnection" />
    +  <param name="warehouse_plugin" value="warehouse_ros_sqlite::DatabaseConnection" />
    ```

**How to reproduce**  
1. Launch panda_moveit_config with db options.
    ```
    roslaunch panda_moveit_config demo.launch db:=true
    ```
2. Connect warehouse from the Context tab.
3. Add and publish a `Box` object from the SceneObjects tab.(Position X to 0.5)
4. Use the SaveScene operation from the StoredScenes tab to save the Scene.
5. Clear plannnig scene and publish operations from the SceneObjects tab.
6. Add and publish a `Sphere` object from the SceneObjects tab.(Position Y to 0.5)
7. Use the LoadScene operation in the StoredScenes tab to load the scene of the `Box` object saved in step 1.
8. Both the `Box` and `Sphere` objects are displayed at the same time.
9. Press the RESET button and only the `Sphere` will disappear.

### Expected behaviour
All existing objects disappear just by performing the LoadScene operation.

### Actual behaviour
When I perform a LoadScene operation, the existing objects remain.
They disappear when I perform a RESET operation.

### Backtrace or Console output
When I try to operate on SaveQuery I get the following MD5 error:
```
[ERROR] [1718781457.852316713]: The md5 sum for the ROS messages saved in the database differs from that of the compiled message. Can only query metadata.
```

rhaschke, 2024-06-19:
Bisecting for the offending commit, resulted in #3538, @v4hn.

Takayuki9999, 2024-06-20:
Thank you for your quick response.

Yes, those objects were cleared in the past.  
I'll wait for the merge of #3610 and try it.

I understand about the MD5 sum error. Thank you for your suggestion.

Takayuki9999, 2024-06-21:
We have confirmed that the behavior of this issue is resolved by the merged #3610 .
Thank you for fixing this issue.

closed 2024-06-21 by Takayuki9999