moveit/moveit #3612: When moving the Nullspace-Slider, sometimes the robot's joint values ​​cannot be returned to their original values.

Takayuki9999, 2024-06-25:
**Is your feature request related to a problem? Please describe.**  
Sometimes after manipulating the Nullspace-Slider, even if I change the joint values ​​of the robot query back to their original values, the joint values ​​are automatically changed, which can confuse me.
I'm not very familiar with this code, but from looking into it I found that it works like this:
* If the dimension of the null-space basis is reduced while sliding a Nullspace-Slider, the Nullspace-Slider be hide() while still having a value s sometimes.
* In this state, if I change the joint values ​​of the robot query back to their original values, the Nullspace-Slider will be show() again, but the retained values ​​will be applied and the joint values ​​will change.
* This immediately reduces the dimension of the null-space basis again and hide() the Nullspace-Slider while retaining its value.
* As a result the joint values ​​cannot be restored to their original values.
* It seems that there are  robots that encounter this issue and others that do not.
 
A video of this happening is added below in "Additional context".

**Describe the solution you'd like**  
Could you please set the value of Nullspace-slider to 0 with setValue(0) at the same time as you hide() it?  
I'm not familiar with Qt, so I don't know if this solution is correct.  
In a quick test, it seemed to work fine.

**Describe alternatives you've considered**

**Additional context**  
We'll use `moveit_resources_prbt_moveit_config` as an example.  
*Note:*   
This issue could not occur in `panda_moveit_config` and `moveit_resources_fanuc_moveit_config` because they rarely hide() during Nullspace-Slider operations.

```
roslaunch moveit_resources_prbt_moveit_config demo.launch
```

https://github.com/moveit/moveit/assets/51587552/d1829272-0162-4d07-a960-ae885cefff4d

rhaschke, 2024-06-25:
Thanks for pointing out this issue. Could you please verify the proposed solution in #3613?

Takayuki9999, 2024-06-26:
I have confirmed that the solution in #3613  resolves the issue.
This behavior disappeared for `moveit_resources_prbt_moveit_config`  and MyRobot (which was causing this issue often).
Thank you for your always quick response.

PR #3613 (2024-06-26): nullspace exploration: reset sliders to zero before hiding
Fixes #3612 implementing the proposed solution from there
closed 2024-06-26 by rhaschke