moveit/moveit #3617: Questionable State Update When Constraint Check Fails

jackie-wayland, 2024-06-28:
https://github.com/moveit/moveit/blob/e8eb9c11053cdd8f1acf9d224dc56f7a8d691677/moveit_core/constraint_samplers/src/default_constraint_samplers.cpp#L670

I've come across a potential issue in the `callIK()` where it appears that the state is being set to  `jmg_` and updated even when certain constraints are not met (`validate(state)` return false). 

Updating the state without ensuring that all constraints are satisfied may lead to inconsistent or unreliable behavior in the process.

Is this intended behavior? If so, could you provide some insight into why the state update is performed even when the constraints fail?

Looking forward to your feedback and suggestions on this matter.

Best regards

rhaschke, 2024-06-28:
The information that the state is invalid is propagated up the call-chain. If so, the state isn't used for planning.
I don't see an issue there. Did you face a particular problem?

jackie-wayland, 2024-06-29:
The information about the failed state validation, while being propagated to the upper level, is still being used as the initial state for the next inverse kinematics attempt. This can potentially lead to incorrect inverse kinematics solutions.

rhaschke, 2024-06-29:
> The failed state is used as the initial state for the next inverse kinematics attempt.

Why do you think so? An invalid state will be rejected by the planner.

jackie-wayland, 2024-06-30:
if `use_as_seed` is set `true`,  the next sample point will use this `jmg_` state as the seed.
https://github.com/moveit/moveit/blob/e8eb9c11053cdd8f1acf9d224dc56f7a8d691677/moveit_core/constraint_samplers/src/default_constraint_samplers.cpp#L648

Although the planner does not directly utilize failed validation states, they may still have an effect on subsequent IK solution calculations. Is this expected behavior?

rhaschke, 2024-06-30:
I think your assumption is that a subsequent call to callIK() will use the same state again. But that's not true. This state will be rejected by the planner and not used any further.