moveit/moveit #3629: moveit_servo for fiducial markers

ManasviniSrini, 2024-08-09:
Hi,
I'm trying to use moveit_servo to track fiducial markers and would really appreciate some help. I've been at my wits end and any leads would give me the opportunity to explore. 

Thanks

rhaschke, 2024-08-11:
Please precisely describe your issue. What did you achieved? What does fail?
Did you have a look at existing issues, e.g. https://github.com/moveit/moveit/issues/2617?

ManasviniSrini, 2024-08-11:
Hi, thank you so much for getting back to me. I appreciate your help. I'm going to share a detailed overview of my current setup and what's functioning properly so far. Please bear with me as I provide the full context.

**Enviroment:**
ROS Distro: Noetic
OS Version: Ubuntu 20.04
**Robot:**
UR5 - real robot

**Steps to reproduce**:
`roslaunch ur_robot_driver ur5_bringup.launch`
(The robot works perfectly with moveit)

**Controller list:**

To check the status of the controllers, I ran the following command:
`rosservice call /controller_manager/list_controllers`

Here is the output I received:
```bash
controller: 
  - 
    name: "joint_state_controller"
    state: "running"
    type: "joint_state_controller/JointStateController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::JointStateInterface"
        resources: []
  - 
    name: "pos_joint_traj_controller"
    state: "initialized"
    type: "position_controllers/JointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  - 
    name: "scaled_pos_joint_traj_controller"
    state: "stopped"
    type: "position_controllers/ScaledJointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "scaled_controllers::ScaledPositionJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  - 
    name: "joint_group_vel_controller"
    state: "running"
    type: "velocity_controllers/JointGroupVelocityController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::VelocityJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
[... 337 lines cut ...]
    ros::spinOnce();
    rate.sleep();
  }

  // Set the ArUco marker pose as the target pose
  tracker.resetTargetPose();
  latest_marker_pose.header.stamp = ros::Time::now();
  target_pose_pub.publish(latest_marker_pose);

  // Run the pose tracking in a new thread
  std::thread move_to_pose_thread(
      [&tracker, &lin_tol, &rot_tol] { tracker.moveToPose(lin_tol, rot_tol, 0.1 /* target pose timeout */); });

  // Wait for the robot to reach the target pose
  move_to_pose_thread.join();

  // Make sure the tracker is stopped and clean up
  tracker.stopMotion();

  return EXIT_SUCCESS;
}
```
I'm aware that this issue might be somewhat out of the usual scope for this forum, but any leads or guidance would be helpful as I'm trying to resolve this problem. If I'm missing any details or if there's anything more you'd need to know to provide assistance, I'd be happy to provide more information.

Thanks a lot again.

AndyZe, 2024-08-11:
I'd say that either the sign on your gains is incorrect, for example `/servo_server/x_proportional_gain` should be -1.5, or one of the frames is incorrect.

ManasviniSrini, 2024-08-11:
So I double checked on the frames and @AndyZe was right. 
The manipulator moves in the direction of the marker now. I also played with the gains and the sign seems alright for where the marker is in my environment. However, it doesn't go all the way and the arm only moves by a little bit. 
Do you have any suggestions to improve what's happening ? I also get the really close to singularity error when the arm is not close to a singularity. 

```yaml
[ INFO] [1723409195.202405413]: Servo status: Close to a singularity, decelerating
[ INFO] [1723409195.210512900]: Servo status: Very close to a singularity, emergency stop
[ INFO] [1723409195.362420551]: Servo status: Close to a singularity, decelerating
[ INFO] [1723409195.370440395]: Servo status: Very close to a singularity, emergency stop
[ INFO] [1723409195.594421121]: Servo status: Close to a singularity, decelerating
[ INFO] [1723409195.602422599]: Servo status: Very close to a singularity, emergency stop
[ INFO] [1723409196.106529147]: Servo status: Close to a singularity, decelerating
[ INFO] [1723409196.114424860]: Servo status: Very close to a singularity, emergency stop
[ INFO] [1723409196.138419679]: Servo status: Close to a singularity, decelerating
[ INFO] [1723409196.146358151]: Servo status: Very close to a singularity, emergency stop
[ INFO] [1723409196.242428384]: Servo status: Close to a singularity, decelerating
```

Would really appreciate any suggestions.

Thanks a lot

AndyZe, 2024-08-11:
Try increasing these 2 parameters:

```
lower_singularity_threshold:  17  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30
```