moveit/moveit #3635: UR5e + MoveIt trajectory execution timeout

j3knk, 2024-08-28:
I am opening this issue since I found no better way to ask for advice regarding code and the thoughts behind it:

I am running a UR5e robot with a customized robot model file and MoveIt. Due to safety reasons, the speed of the robot is set to half. This seems to cause wrong predictions for the execution duration of the robot, and the move execution gets cancelled about midway with a timeout error. 
Now, I am trying to track down whether the issue is driver/Kinematics Plugin/MoveIt/config related.
The code line responsible for triggering the timeout is https://github.com/moveit/moveit/blob/9caa14eabc69c6ba2328eacfb5f6253a8acff085/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L1270 and the `expected_trajectory_execution` is mainly influenced by  https://github.com/moveit/moveit/blob/9caa14eabc69c6ba2328eacfb5f6253a8acff085/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L1203-L1208

What I do not understand is how the `time_from_start` attribute is used to deliver information about the expected execution duration. To my understaning, the attribute is set on every call to https://github.com/moveit/moveit/blob/9caa14eabc69c6ba2328eacfb5f6253a8acff085/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L266-L267

At this point I got lost in the source code. When is this system invoked by which entities?

rhaschke, 2024-08-28:
> Due to safety reasons, the speed of the robot is set to half.

Do you set the scaling factor in the UR5 controller or in MoveIt?
In the former case, MoveIt cannot know about the slow down and will complain because execution lags behind (that's what you observed). You should change the `max_velocity_scaling_factor`:
https://answers.ros.org/question/251282/how-to-change-velocity-acceleration-in-motionplanrequest-topic/

j3knk, 2024-08-30:
Thx, that will most likely fix the timeout. Will test it asap.

Still curious after the code inspection on what parameters the move time estimation is calculated. Would appreciate some pointers.

rhaschke, 2024-08-30:
> Still curious after the code inspection at which point the move time estimation is calculated.

MoveIt simply compares the expected trajectory timing (as given in trajectory's member `time_from_start`) with the actual timing.

j3knk, 2024-08-30:
That part is clear, the unclear part is what instance provides the `time_from_start` value.

rhaschke, 2024-09-02:
The `time_from_start` members of a trajectory are provided by the time parameterization plugin of your planning pipeline.

j3knk, 2024-09-14:
Thanks a lot for your help. The `max_velocity_scaling_factor` seems to work fine, at the cost of some hickups in the speed consistency. Maybe retrieving the UR controller scaling parameter and pass it somehow to the planning pipeline would be more elegant, but I suppose it will cost me way more time to implement than its worth. 👍

closed 2024-09-14 by j3knk