moveit/moveit #3638: Moveit Servo bad `joint_trajectory` causes the robot to suddendly jump to another pose

torydebra, 2024-09-04:
### Description

I have a configuration in which I am using, alternatively, `moveit_cpp` and `moveit_servo`. I make sure I never run them togheter, but instead run one or the other. 
I noticed that, sometimes, the robot jumps because `moveit_servo` "thinks" the robot is in another pose instead of the actual one. Specifically, this happens when:
- moving the robot commanding poses with `moveit_cpp`, 
- then with switching to `moveit_servo`   

`moveit_servo` generates a trajectory, not from *where the robot actually is*, but instead from where the `moveit_servo` was stopped before running the `moveit_cpp`, or from the initial position if moveit_servo was never started. 

I printed some robot states inside `moveit_servo` and noticed that the planning_scene_monitor (which is the same for `moveit_cpp` and `moveit_servo`) works correctly, and that inside the `moveit_servo` (actually inside the `servo_calcs`) the robot state is updated correctly at each loop.

So the issue seems to be here: 
https://github.com/moveit/moveit/blob/361de4b9295cd867b4ff446f26c8349d8a53cb94/moveit_ros/moveit_servo/src/servo_calcs.cpp#L351-L359

Indeed, I solved overwriting `joint_trajectory` not with the `last_sent_command_` (which evidently is indeed the last command sent with `moveit_servo` before switching to `moveit_cpp`) but with the actual robot state:
```c++
else
  {
    // Joint trajectory is not populated with anything, so set it to the last positions and 0 velocity
    *joint_trajectory = *last_sent_command_;
    //joint_trajectory->header.stamp = ros::Time(0);
    //joint_trajectory->header.frame_id = parameters_.planning_frame;
    //joint_trajectory->joint_names = internal_joint_state_.name;
    trajectory_msgs::JointTrajectoryPoint point;
    point.time_from_start = ros::Duration(parameters_.publish_period);

    if (parameters_.publish_joint_positions)
      current_state_->copyJointGroupPositions(joint_model_group_, point.positions);
    if (parameters_.publish_joint_velocities)
    {
      std::vector<double> velocity(num_joints_);
      point.velocities = velocity;
    }
    if (parameters_.publish_joint_accelerations)
    {
      // I do not know of a robot that takes acceleration commands.
      // However, some controllers check that this data is non-empty.
      // Send all zeros, for now.
      point.accelerations.resize(num_joints_);
    }
    joint_trajectory->points.clear();
    joint_trajectory->points.push_back(point);

    for (auto& point : joint_trajectory->points)
    {
      point.velocities.assign(point.velocities.size(), 0);
    }

  }
```

which seems to have solved the issue.
I am not sure if this may cause other problems. Do you see any problems in updating the `joint_trajectory` in this `else` with the last actual robot state (as done above), instead of the last commanded trajectory?

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Binary build

### Steps to reproduce
- Start `moveit_servo`, move the robot with it in a certain pose A.
- Stop `moveit_servo`, move the robot with some other mean (e.g. `moveit_cpp`), in a different certain pose B.
- Restart `moveit_servo` and move the robot with it. Sometimes it happens that the trajejectory generated by `moveit_servo` begins from A instead of from B, causing a sudden jump from A to B in a few control loops.

AndyZe, 2024-09-05:
This definitely has been an annoying bug that was fixed in the MoveIt2 version. It would be nice if you can make a PR  :+1:

torydebra, 2024-09-05:
ok done in #3642

closed 2024-09-11 by torydebra