moveit/moveit #3648: segmentation fault (SIGSEGV)

Shobinvinod, 2024-09-25:
I am experiencing a segmentation fault (SIGSEGV) during the execution of a planned motion using the pilz_industrial_motion_planner. Below are the logs and details of the error:

Log :
############################
[INFO] [1727262234.816205, 21.164000]: Moveit Controller state : PLANNING
[INFO] [1727262234.821536, 21.169000]: Moveit Controller log : This goal has been accepted by the simple action server
[ INFO] [1727262234.836346254, 21.181000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1727262234.836606850, 21.181000000]: Planning attempt 1 of at most 1
[ INFO] [1727262234.836832316, 21.181000000]: Using planning pipeline 'pilz_industrial_motion_planner'
[INFO] [1727262234.837476, 21.182000]: Moveit Controller state : PLANNING
[ INFO] [1727262234.837945126, 21.182000000]: Generating LIN trajectory...
[INFO] [1727262234.839870, 21.184000]: Moveit Controller log : This goal has been accepted by the simple action server
--Type <RET> for more, q to quit, c to continue without paging--

Thread 15 "move_group" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe549b700 (LWP 15447)]
0x00007ffff00453e5 in pilz_industrial_motion_planner::TrajectoryGeneratorLIN::extractMotionPlanInfo(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, pilz_industrial_motion_planner::TrajectoryGenerator::MotionPlanInfo&) const ()
   from /opt/ros/noetic/lib/libplanning_context_loader_lin.so

#######################################

Environment Details
MoveIt Version:  1.1.14-1focal.20240830.082438
ROS Distro: Noetic
OS: Ubuntu 20.04.6
Pilz Industrial Motion Planner: 1.1.14

Additional Notes
The segmentation fault occurs in the function TrajectoryGeneratorLIN::extractMotionPlanInfo.
It seems related to accessing or handling the motion plan information.
Is there any known issue or workaround related to this function or the libplanning_context_loader_lin.so?

rhaschke, 2024-09-25:
Generally, that function is known to work. Please provide your source code to reproduce the segfault - ideally on some robot supported by moveit_resources, e.g. the Panda arm.

rhaschke, 2024-10-07:
Closing due to inactivity. Please feel free to reopen the issue and provide more details as requested.

closed 2024-10-07 by rhaschke