moveit/moveit #3650: When I am trying to build my catkin_ws I am getting following error error: static assertion failed: Signal and slot arguments are not compatible.   314 |         Q_STATIC_ASSERT_X((FunctorArgumentCount >= 0),

Akumar201, 2024-09-26:
```
In file included from /usr/include/aarch64-linux-gnu/qt5/QtCore/qnamespace.h:43,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qobjectdefs.h:48,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qobject.h:46,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/QObject:1,
                 from /opt/ros/noetic/include/rviz/properties/property.h:34,
                 from /opt/ros/noetic/include/rviz/properties/status_property.h:32,
                 from /opt/ros/noetic/include/rviz/display.h:38,
                 from /home/orangepi/targetarm/tular_servo_ws/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/include/moveit/rviz_plugin_render_tools/trajectory_visualization.h:41,
                 from /home/orangepi/targetarm/tular_servo_ws/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp:40:
/usr/include/aarch64-linux-gnu/qt5/QtCore/qobject.h: In instantiation of ‘static typename std::enable_if<(QtPrivate::FunctionPointer<Func2>::ArgumentCount == -1), QMetaObject::Connection>::type QObject::connect(const typename QtPrivate::FunctionPointer<Func>::Object*, Func1, const QObject*, Func2, Qt::ConnectionType) [with Func1 = void (rviz::Property::*)(); Func2 = std::nullptr_t; typename std::enable_if<(QtPrivate::FunctionPointer<Func2>::ArgumentCount == -1), QMetaObject::Connection>::type = QMetaObject::Connection; typename QtPrivate::FunctionPointer<Func>::Object = rviz::Property]’:
/opt/ros/noetic/include/rviz/properties/property.h:187:94:   required from ‘typename std::enable_if<(! QtPrivate::FunctionPointer<Func>::IsPointerToMemberFunction), QMetaObject::Connection>::type rviz::Property::connect(const QObject*, Func&&, Qt::ConnectionType) [with Func = std::nullptr_t; typename std::enable_if<(! QtPrivate::FunctionPointer<Func>::IsPointerToMemberFunction), QMetaObject::Connection>::type = QMetaObject::Connection]’
/opt/ros/noetic/include/rviz/properties/bool_property.h:56:5:   required from ‘rviz::BoolProperty::BoolProperty(const QString&, bool, const QString&, rviz::Property*, Func&&, const R*) [with Func = std::nullptr_t; R = moveit_rviz_plugin::TrajectoryVisualization]’
/home/orangepi/targetarm/tular_servo_ws/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp:108:72:   required from here
/usr/include/aarch64-linux-gnu/qt5/QtCore/qobject.h:314:9: error: static assertion failed: Signal and slot arguments are not compatible.
  314 |         Q_STATIC_ASSERT_X((FunctorArgumentCount >= 0),
      |         ^~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/qt5/QtCore/qobjectdefs.h:50,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qobject.h:46,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/QObject:1,
                 from /opt/ros/noetic/include/rviz/properties/property.h:34,
                 from /opt/ros/noetic/include/rviz/properties/status_property.h:32,
                 from /opt/ros/noetic/include/rviz/display.h:38,
                 from /home/orangepi/targetarm/tular_servo_ws/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/include/moveit/rviz_plugin_render_tools/trajectory_visualization.h:41,
                 from /home/orangepi/targetarm/tular_servo_ws/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp:40:
/usr/include/aarch64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h: In instantiation of ‘struct QtPrivate::FunctorReturnType<std::nullptr_t, QtPrivate::List<> >’:
/usr/include/aarch64-linux-gnu/qt5/QtCore/qobject.h:317:158:   required from ‘static typename std::enable_if<(QtPrivate::FunctionPointer<Func2>::ArgumentCount == -1), QMetaObject::Connection>::type QObject::connect(const typename QtPrivate::FunctionPointer<Func>::Object*, Func1, const QObject*, Func2, Qt::ConnectionType) [with Func1 = void (rviz::Property::*)(); Func2 = std::nullptr_t; typename std::enable_if<(QtPrivate::FunctionPointer<Func2>::ArgumentCount == -1), QMetaObject::Connection>::type = QMetaObject::Connection; typename QtPrivate::FunctionPointer<Func>::Object = rviz::Property]’
/opt/ros/noetic/include/rviz/properties/property.h:187:94:   required from ‘typename std::enable_if<(! QtPrivate::FunctionPointer<Func>::IsPointerToMemberFunction), QMetaObject::Connection>::type rviz::Property::connect(const QObject*, Func&&, Qt::ConnectionType) [with Func = std::nullptr_t; typename std::enable_if<(! QtPrivate::FunctionPointer<Func>::IsPointerToMemberFunction), QMetaObject::Connection>::type = QMetaObject::Connection]’
/opt/ros/noetic/include/rviz/properties/bool_property.h:56:5:   required from ‘rviz::BoolProperty::BoolProperty(const QString&, bool, const QString&, rviz::Property*, Func&&, const R*) [with Func = std::nullptr_t; R = moveit_rviz_plugin::TrajectoryVisualization]’
/home/orangepi/targetarm/tular_servo_ws/src/moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp:108:72:   required from here
/usr/include/aarch64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h:367:52: error: request for member ‘operator()’ in ‘QtPrivate::FunctorReturnType<std::nullptr_t, QtPrivate::List<> >::dummy<std::nullptr_t>()’, which is of non-class type ‘std::nullptr_t’
  367 |         typedef decltype(dummy<Functor>().operator()((dummy<ArgList>())...)) Value;
      |                          ~~~~~~~~~~~~~~~~~~~~~~~~~~^
make[2]: *** [moveit/moveit_ros/visualization/rviz_plugin_render_tools/CMakeFiles/moveit_rviz_plugin_render_tools.dir/build.make:141: moveit/moveit_ros/visualization/rviz_plugin_render_tools/CMakeFiles/moveit_rviz_plugin_render_tools.dir/src/trajectory_visualization.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:29457: moveit/moveit_ros/visualization/rviz_plugin_render_tools/CMakeFiles/moveit_rviz_plugin_render_tools.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```

rhaschke, 2024-09-26:
Looks like outdated code. Which versions of MoveIt and rviz are you using?

Akumar201, 2024-09-28:
This is the moveit version 

```
apt show ros-noetic-moveit
Package: ros-noetic-moveit
Version: 1.1.14-1focal.20240830.093105
Priority: optional
Section: misc
Maintainer: Dave Coleman <dave@picknik.ai>
Installed-Size: 22.5 kB
Depends: ros-noetic-moveit-commander, ros-noetic-moveit-core, ros-noetic-moveit-planners, ros-noetic-moveit-plugins, ros-noetic-moveit-ros, ros-noetic-moveit-setup-assistant
Homepage: http://moveit.ros.org
Download-Size: 9,840 B
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: Meta package that contains all essential package of MoveIt.
 Until Summer 2016 MoveIt had been developed over multiple repositories, where developers' usability and maintenance effort was non-trivial. See the detailed discussion for the merge of several repositories.

```

rhaschke, 2024-09-30:
You are not using MoveIt from apt binaries, but building from source. Which version are you using in your catkin workspace?

rhaschke, 2024-10-11:
Closing for inactivity. Feel free to reopen when providing the requested information.

closed 2024-10-11 by rhaschke