moveit/moveit #3651: moveit 1.1.15 Cartesian path - PLANNING BROKEN

ron007d, 2024-10-05:
### Moveit 1.1.15 has issue with its cartesian path planning. 

the planning is not working properly. if given more way points then its going to the point then coming back to the first one.
for example suppose there is `way_points[200]` then the robot moves to `way_points[1]` then again to `way_points[0]`
### Your environment
* ROS Distro: Noetic
* OS Version:  Ubuntu 20.04
* Binary
* Version 1.1.15 Focal

### Steps to reproduce
Go to this repo https://github.com/ron007d/ros_pick_and_place_simulation and build the source

then run
```bash
roslaunch simulation_moveit_config demo.launch 
```
then run this ipynb file 
https://drive.google.com/file/d/12d0-KrohQ47bmtfS_hv8DPSLspo4oM-a/view?usp=sharing

### Expected behavior
The expected behavior is the robot will follow this trajectory
[image]

### in moveit 1.1.14 its working :white_check_mark:  :point_down: :point_down: 
**VIDEO BELLOW ** 
[image]

> AS you can see in this video / gif the movement is happening smoothly

### Actual behaviour
### Moveit 1.1.15 has this movement :point_down: :point_down: 
[image]

rhaschke, 2024-10-06:
Thanks for reporting the issue. I have filed a fix: #3652

rhaschke, 2024-10-06:
I was surprised to see a JumpThreshold argument in your notebook file. This was replaced by a CartesianPrecision arg.

ron007d, 2024-10-06:
Yeah that file was used for testing 1.1.14 and 1.1.15

So, the file was last changed on testing 1.1.14 thus the jump threshold arg. 😅

closed 2024-10-06 by rhaschke

6chtom, 2024-10-09:
I have the same issue.

ron007d, 2024-10-22:
@rhaschke  checked the new 1.1.16 but the issue with the given as `IPYNB` is still there

USING BINARY 1.1.16-1 focal

for here is output plan of the joint trajectory
in this trajectory plan I can see that there are multiple NON moving points.

https://github.com/user-attachments/assets/3a2818d7-e3f5-4d78-87c9-e2a8ea5db406

## JOINT TRAJECTORY OUTPUT
```yaml
joint_trajectory: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: "world"
  joint_names: 
    - robot1_shoulder_pan_joint
    - robot1_shoulder_lift_joint
    - robot1_elbow_joint
    - robot1_wrist_1_joint
    - robot1_wrist_2_joint
    - robot1_wrist_3_joint
  points: 
    - 
      positions: [1.5708296407416484, -1.5707126266806066, 1.5707738719101205, -1.5708064734530196, -1.5707346707180163, 1.5707386779251997]
      velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      accelerations: [-0.843514250628877, 0.0, 0.0, 0.0, 0.0, 0.0]
      effort: []
      time_from_start: 
        secs: 0
        nsecs:         0
    - 
      positions: [1.5270869961824003, -1.621971638146858, 1.6206290502394185, -1.5693999544969492, -1.5707369194657417, 1.5269960333413102]
      velocities: [-0.24670362460887874, -0.26859669405003395, 0.25097630024949635, 0.01763551205597896, -1.271767619272153e-05, -0.24670362474423752]
      accelerations: [-0.9751162770161464, -0.9623978811151136, 0.8457705694457806, 0.1166869251477958, -5.043707904466559e-05, -0.9751162775336549]
      effort: []
      time_from_start: 
        secs: 0
        nsecs: 322048517
    - 
      positions: [1.4796078389318756, -1.6721657093732996, 1.6667226860576398, -1.5652966116703704, -1.5707393695942953, 1.4795168760649995]
      velocities: [-0.414086335958215, -0.41281705978532374, 0.36725263811913156, 0.0455875644822596, -2.370217636220976e-05, -0.41408633592634697]
      accelerations: [-0.9326135617092359, -0.5741907965605143, 0.3318545625720662, 0.24235680074057736, -8.664168231104054e-05, -0.9326135579779613]
      effort: []
      time_from_start: 
        secs: 0
        nsecs: 454827267
    - 
      positions: [1.4280432704123733, -1.7212116719709596, 1.7091671353776046, -1.5586924260036827, -1.5707425419382317, 1.4279523075737606]
      velocities: [-0.521797324929616, -0.46801400703001883, 0.39249029287136106, 0.0755507982050186, -3.205576831501198e-05, -0.5217973246485539]
      accelerations: [-0.9881420515444092, -0.393819596451446, 0.09901936713230226, 0.2948522386204753, -5.9902044581706616e-05, -0.9881420510981935]
      effort: []
      time_from_start: 
        secs: 0
        nsecs: 564401173
    - 
      positions: [1.3720570999338355, -1.768933702929369, 1.748017378073112, -1.549817728118931, -1.5707459772924028, 1.3719661371249436]
      velocities: [-0.6165604782436758, -0.4951458613037089, 0.3907026369174651, 0.104474591364283, -3.8248757258581264e-05, -0.6165604778559254]
      accelerations: [-0.9183198897301798, -0.14176316515656467, -0.1458696730212108, 0.28766631372021473, -6.512216555974138e-05, -0.9183198879683767]
      effort: []
      time_from_start: 
        secs: 0
        nsecs: 662107771
    - 
      positions: [1.311312810363469, -1.815115951061163, 1.7833332971266997, -1.5389483665301542, -1.5707497811942677, 1.3112218475979478]
      velocities: [-0.7037723095112234, -0.5027790117271016, 0.37258822374623213, 0.13022582093484117, -4.415134036102168e-05, -0.70377230899873]
      accelerations: [-0.9700018243051978, -0.020199752910465336, -0.2487660759644282, 0.2690120060056787, -6.25211616496811e-05, -0.9700018233902273]
[... 796 lines cut ...]
    - 
      positions: [1.2455414059254688, -1.8594390612468477, 1.8151348170684052, -1.5264235106132378, -1.5707539139856546, 1.245450443208669]
      velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      effort: []
      time_from_start: 
        secs: 0
        nsecs: 842125261
    - 
      positions: [1.2455414059254688, -1.8594390612468477, 1.8151348170684052, -1.5264235106132378, -1.5707539139856546, 1.245450443208669]
      velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
      effort: []
      time_from_start: 
        secs: 0
        nsecs: 842125261
multi_dof_joint_trajectory: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  joint_names: []
  points: []```

rhaschke, 2024-10-22:
I cannot reproduce the issue anymore with 1.16. Here is what I get:

https://github.com/user-attachments/assets/f6fd02a9-9e43-4284-accd-55dc84179a61

rhaschke, 2024-10-23:
So it does work for you as well now?

ron007d, 2024-10-23:
yeah.. its working as well :+1:  :white_check_mark:

PR #3652 (2024-10-06): Fix Cartesian interpolation
Fixes #3651