moveit/moveit #3653: Moveit package error when relaunched

JohnL-Fxa, 2024-10-09:
### Description

I am controlling a ur3e with two grippers by launching the bringup file, moveit package and my GUI python scripts to do pick and placement tasks, and it works fine with no issues. 
For a new project I needed to use only one gripper, so I did another urdf where the ur3e has only one gripper and created a new moveit package with the new urdf for a new project, the problem is that the moveit package behaves fine the first time its launched only. and then everytime I relaunch I have to regenerate the moveit package from the moveit setup assistant without really changing anything just regenerating and It works but I need to do the same thing everytime I relaunch.

### Environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Source or Binary build? binary
* If binary, which release version? noetic

### Steps to reproduce
When launching the new moveit package with one gripper

### Expected behaviour
It should be launched properly and I see the full urdf on rviz

### Actual behaviour
The package behaves properly only the first time I launch it, and when I close and relaunch everything, the second time I get this error in the moveit terminal
NODES
  /
    move_group (moveit_ros_move_group/move_group)

ROS_MASTER_URI=http://localhost:11311

process[move_group-1]: started with pid [5660]
[ERROR] [1728458468.160071488]: Semantic description is not specified for the same robot as the URDF
[ERROR] [1728458468.161426752]: Virtual joint does not attach to a link on the robot (link 'world' is not known)
[ERROR] [1728458468.161588224]: Joint 'coupling_tool0_joint' declared as part of group 'gripper' is not known to the URDF
[ERROR] [1728458468.161621792]: Joint 'robotiq_hande_base_joint' declared as part of group 'gripper' is not known to the URDF
[ERROR] [1728458468.161658176]: Joint 'hande_left_finger_joint' declared as part of group 'gripper' is not known to the URDF
[ERROR] [1728458468.161686720]: Joint 'hande_right_finger_joint' declared as part of group 'gripper' is not known to the URDF
[ERROR] [1728458468.161709664]: Joint 'epick_tip_joint' declared as part of group 'gripper' is not known to the URDF
[ERROR] [1728458468.161736384]: Link 'coupling' declared as part of a chain in group 'gripper' is not known to the URDF
[ WARN] [1728458468.161777216]: Group 'gripper' is empty.
[ WARN] [1728458468.162009472]: Link 'coupling' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162047936]: Link 'coupling' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162074176]: Link 'coupling' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162095808]: Link 'coupling' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162115392]: Link 'coupling' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162135456]: Link 'coupling' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162154432]: Link 'coupling' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162173568]: Link 'ee_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162192960]: Link 'ee_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162211680]: Link 'ee_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162230208]: Link 'ee_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162250496]: Link 'ee_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162267488]: Link 'ee_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162287648]: Link 'hand_e_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162304256]: Link 'hand_e_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162320992]: Link 'hand_e_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162352992]: Link 'hand_e_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162375072]: Link 'hand_e_link' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162396480]: Link 'hande_left_finger' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162419424]: Link 'hande_left_finger' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162440352]: Link 'hande_left_finger' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162462400]: Link 'hande_left_finger' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162483040]: Link 'hande_right_finger' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162505696]: Link 'hande_right_finger' is not known to URDF. Cannot disable/enable collisons.
[ WARN] [1728458468.162521632]: Link 'hande_right_finger' is not known to URDF. Cannot disable/enable collisons.
[ERROR] [1728458468.162566016]: Joint 'hande_left_finger_joint' marked as passive is not known to the URDF. Ignoring.
[ INFO] [1728458468.162729664]: Loading robot model 'ur3e_robot'...
[ INFO] [1728458468.162829824]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1728458468.277921024]: Group 'gripper' must have at least one valid joint
[ WARN] [1728458468.277996864]: Failed to add group 'gripper'

and then if I open rviz and add motion planning the gripper is not there, and to to fix it I open the moveit package in the moveit setup assistant and regenerate it again without changing anything then it works fine but if I relaunch I get the same error again.

rhaschke, 2024-10-09:
Looks like you screwed your URDF and/or SRDF definitions: there are many errors regarding the URDF/SRDF descriptions.
Verify that you are using the correct files. Theoretically, there is no difference between first and subsequent launches.
Ensure that you are always starting from a clean ROS environment: Don't have a `roscore` (and its ROS parameters) loitering around from a previous call.

rhaschke, 2024-10-10:
Please provide a zip file of your workspace sources, so that we can reproduce the issue locally.
Also provide the detailed commands you use to launch your application.
As I said, I cannot imagine any reason that the launching works only for the first time. There is no state remembered.

JohnL-Fxa, 2024-10-14:
I found the solution the problem is for some reason in the move_group.launch the  "load_robot_description" is set to false and when changing that and including the proper path to the planning_context.launch everything behaves as expected. The previous packages package that I have created it used to be true by default.

  <arg name="load_robot_description" default="false" />
  
  <include file="$(dirname)/planning_context.launch">
    <arg name="load_robot_description" value="$(arg load_robot_description)" />
  </include>
_________________________________________________________________
  <arg name="load_robot_description" default="true" />
  
  <include file="$(find ur3e_one_gripper_moveit_config)/launch/planning_context.launch">
    <arg name="load_robot_description" value="$(arg load_robot_description)" />
  </include>

rhaschke, 2024-10-14:
The `load_robot_description` argument is only `true` for the top-level .launch file by intention:
https://github.com/moveit/moveit/blob/a1b1bfb88326a8520b895b794e05f06e9fa545fc/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/demo.launch#L15

This is because, for a real robot, the bringup.launch of the robot will actually load the robot_description.

closed 2024-10-14 by rhaschke