moveit/moveit #3658: Error in Cartesian Trajectory Computation in Recent MoveIt! Versions (1.1.15 and 1.1.16)

canongun, 2024-10-17:
### Description

Hi,
After updating from version 1.1.14 to 1.1.15, I can no longer use the Cartesian path planner in Python as before. Below is the snippet of how my script was set up:
```
(plan, fraction) = self.move_group.compute_cartesian_path(
                                        waypoints,  # waypoints to follow
                                        0.01,       # eef_step
                                        0.0)        # jump_threshold
```

The issue arises when invoking the `compute_cartesian_path` function, with the error message indicating that `jump_threshold` is no longer recognized as a valid argument. After comparing multiple versions of MoveIt!, I found that starting from version 1.1.14, the jump_threshold parameter was removed. Both version 1.1.15 (stable) and 1.1.16 (latest) no longer support this argument.
However, even after omitting the `jump_threshold` parameter, I still encountered errors because this parameter is essential for configuring the space between consecutive points.

Here is the function signature in the newer versions:
```
def compute_cartesian_path(
        self,
        waypoints,
        eef_step,
        avoid_collisions=True,
        path_constraints=None,
    ):
        """Compute a sequence of waypoints that make the end-effector move in straight line segments that follow the poses specified as waypoints.
        Configurations are computed for every eef_step meters.
        The jump_threshold specifies the maximum distance in configuration space between consecutive points in the resultingpath.
        Kinematic constraints for the path given by path_constraints will be met for every point along the trajectory.
        If the Kinematic constraints are not met, a partial solution will be returned.
        The return value is a tuple: the actual RobotTrajectory and the fraction of how much of the path was followed.
        """
        if path_constraints:
            if type(path_constraints) is Constraints:
                constraints_str = conversions.msg_to_string(path_constraints)
            else:
                raise MoveItCommanderException(
                    "Unable to set path constraints, unknown constraint type "
                    + str(type(path_constraints))
                )
            (ser_path, fraction) = self._g.compute_cartesian_path(
                [conversions.pose_to_list(p) for p in waypoints],
                eef_step,
                avoid_collisions,
                constraints_str,
            )
        else:
            (ser_path, fraction) = self._g.compute_cartesian_path(
                [conversions.pose_to_list(p) for p in waypoints],
                eef_step,
                avoid_collisions,
            )

        path = RobotTrajectory()
        path.deserialize(ser_path)
        return (path, fraction)
```
Despite the `jump_threshold` being removed from the function signature, the function’s documentation still references it.

I was able to resolve the issue by building MoveIt! from source and downgrading to version 1.1.14. However, I would prefer to stick with the latest versions whenever possible.

Could you please provide guidance on whether I am missing something or how to address this issue? Thank you in advance for your help.

---
### Environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04.6 LTS
* Binary build
* Version 1.1.15

rhaschke, 2024-10-17:
The `jump_threshold` parameter was removed in #3618. The old implementation was brittle and frequently caused issues.
Please remove the parameter wherever used.

> However, even after omitting the `jump_threshold` parameter, I still encountered errors because this parameter is essential for configuring the space between consecutive points.

I'm curious to learn more about the errors you encounter with the new implementation.

> Despite the `jump_threshold` being removed from the function signature, the function’s documentation still references it.

That should be fixed. Thanks for the hint.

canongun, 2024-10-17:
Hello Mr. Haschke,

Thank you for your response. Following your advice, I deleted the source-built MoveIt! and installed the binary version again, this time using version 1.1.16.

Now, my Cartesian planner function looks like this:
```
(plan, fraction) = self.move_group.compute_cartesian_path(
                                        waypoints,  # waypoints to follow
                                        0.01,       # eef_step
                                        avoid_collisions=False,
                                        path_constraints=None)
```
This worked as expected. I apologize for the confusion earlier—I had not tested the exact function from the repository. I also reviewed the [documentation](https://docs.ros.org/en/jade/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#ad7f6d93d73bf43268ba983afb0dc4f23) and the latest [repository commit](https://github.com/moveit/moveit/commit/88b386581c5f25cc5733585bd39dfd2ea690329b#diff-9a9441d3080aaa78482b41ede6d9e263d91f5494ddb0d816cb5a28e1a4b542bdL656), where I noticed that "jump_threshold" is still being referenced. I initially thought this was a mismatch in the function signature.

It would be so nice, if you could update the documentation as well.
Best regards.

rhaschke, 2024-10-17:
> It would be so nice, if you could update the documentation as well.

Done.

closed 2024-10-17 by rhaschke