moveit/moveit #3661: Facing issues with setting up Moveit2 with custom robot, OMPL error.

atharv180904, 2024-10-29:
Hello Community,

I am encountering an issue with motion planning in the MoveIt2 framework. I have created a MoveIt configuration package for my custom robot using the MoveIt Setup Assistant, and I wrote a script similar to the `hello_moveit` example on the website.

Here's my code snippet:

```cpp
#include <memory>
#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>

int main(int argc, char * argv[])
{
  // Initialize ROS and create the Node
  rclcpp::init(argc, argv);
  auto const node = std::make_shared<rclcpp::Node>(
    "hello_moveit",
    rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true)
  );

  // Create a ROS logger
  auto const logger = rclcpp::get_logger("hello_moveit");

  // Create the MoveIt MoveGroup Interface
  using moveit::planning_interface::MoveGroupInterface;
  auto move_group_interface = MoveGroupInterface(node, "m2_arm");

  move_group_interface.setPlanningTime(15);
  move_group_interface.setNumPlanningAttempts(10);

  // Set a target Pose
  auto const target_pose = []{
    geometry_msgs::msg::Pose msg;
    msg.orientation.w = 1.0;
    msg.position.x = 0.2;
    msg.position.y = 0.2;
    msg.position.z = 0.2;
    return msg;
  }();
  move_group_interface.setPoseTarget(target_pose);

  // Create a plan to that target pose
  auto const [success, plan] = [&move_group_interface]{
    moveit::planning_interface::MoveGroupInterface::Plan msg;
    auto const ok = static_cast<bool>(move_group_interface.plan(msg));
    return std::make_pair(ok, msg);
  }();

  // Execute the plan
  if(success) {
    move_group_interface.execute(plan);
  } else {
    RCLCPP_ERROR(logger, "Planning failed!");
  }

  // Shutdown ROS
  rclcpp::shutdown();
  return 0;
}
```

I execute the script with the following command:

```bash
ros2 run plan_using_moveit plan_using_moveit
```

### Terminal Output

Here's the terminal output when I run the script:

```plaintext
[INFO] [1730117591.767709189] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 1.00116 seconds
[INFO] [1730117591.767769536] [moveit_robot_model.robot_model]: Loading robot model 'physics'...
[WARN] [1730117592.751783980] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[INFO] [1730117592.763678763] [move_group_interface]: Ready to take commands for planning group m2_arm.
[INFO] [1730117592.763775224] [move_group_interface]: MoveGroup action client/server ready
[INFO] [1730117592.764384284] [move_group_interface]: Planning request accepted
[INFO] [1730117607.804645767] [move_group_interface]: Planning request aborted
[ERROR] [1730117607.804714445] [move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
[ERROR] [1730117607.804744300] [hello_moveit]: Planning failed!
```

In the terminal where I run my `demo.launch.py` script, I see the following messages:

```plaintext
[move_group-3] [INFO] [1730117592.764222994] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-3] [INFO] [1730117592.764295673] [moveit_move_group_default_capabilities.move_action_capability]: Executing...
[move_group-3] [INFO] [1730117592.771602180] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-3] [INFO] [1730117592.772053840] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'm2_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-3] [ERROR] [1730117607.778441015] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - m2_arm/m2_arm: Unable to sample any valid states for goal tree
[move_group-3] [WARN] [1730117607.779802213] [ompl]: ./src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 15.007609 seconds
[move_group-3] [INFO] [1730117607.804293642] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-3] [INFO] [1730117607.804330103] [moveit_move_group_default_capabilities.move_action_capability]: Catastrophic failure
```

### Troubleshooting Attempts

I've tried the following solutions without success:

1. Increased precision of the orientation coordinates.
2. Loaded kinematics parameters via a launch file.
3. Set the `use_sim_time` parameter to `true`.

None of these resolved the issue. The error message suggests that the planner (`RRTConnect`) is unable to sample valid states for the goal tree.

My MoveIt configuration and script files are available on GitHub: [https://github.com/atharv180904/temp](https://github.com/atharv180904/temp).

### Help Needed

I'm not sure what else to try. Any suggestions on how to resolve this issue would be greatly appreciated!

rhaschke, 2024-10-29:
The problem is that you didn't define kinematic parameters: 
`[WARN] [1730117592.751783980] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!`

As you provide a Cartesian target, there needs be an IK solver available. Ensure that you correctly load your kinematics.yaml in the move_group node (started with demo.launch.py).

atharv180904, 2024-10-29:
Hi @rhaschke ,

First, I want to express my gratitude for the assistance I’ve received so far.

I recently resolved the `kinematics.yaml` error by passing it as a parameter using a launch file. Here’s the code for my launch file:

```python
from launch import LaunchDescription
from launch_ros.actions import Node
from moveit_configs_utils import MoveItConfigsBuilder

def generate_launch_description():
    # Initialize MoveIt configuration for your robot
    moveit_config = MoveItConfigsBuilder("roarm").to_dict()  # Replace "roarm" with your actual robot name

    # Define the node with parameters
    plan_using_moveit_node = Node(
        package="plan_using_moveit",  # Replace with your package name
        executable="plan_using_moveit",  # Replace with your node/executable name
        output="screen",
        parameters=[
            moveit_config["robot_description_kinematics"],
        ],
    )

    return LaunchDescription([
        plan_using_moveit_node
    ])
```

When I run this launch file, the terminal output indicates that the `kinematics.yaml` error is resolved:

```
[INFO] [launch]: All log files can be found below /home/atharv/.ros/log/2024-10-29-16-06-44-164590-atharv-ROG-Strix-G512LI-G512LI-68195
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [plan_using_moveit-1]: process started with pid [68196]
[plan_using_moveit-1] [INFO] [1730198205.833903855] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 1.52173 seconds
[plan_using_moveit-1] [INFO] [1730198205.833962540] [moveit_robot_model.robot_model]: Loading robot model 'physics'...
[plan_using_moveit-1] [INFO] [1730198206.807022303] [moveit_ros.robot_model_loader]: Kinematics Plugin Successfully Defined
[plan_using_moveit-1] [INFO] [1730198206.826839533] [move_group_interface]: Ready to take commands for planning group m2_arm.
[plan_using_moveit-1] [INFO] [1730198206.826964273] [move_group_interface]: MoveGroup action client/server ready
[plan_using_moveit-1] [INFO] [1730198206.827669258] [move_group_interface]: Planning request accepted
[plan_using_moveit-1] [INFO] [1730198221.837458996] [move_group_interface]: Planning request aborted
[plan_using_moveit-1] [ERROR] [1730198221.837855580] [move_group_interface]: MoveGroupInterface::plan() failed or timeout reached
[plan_using_moveit-1] [ERROR] [1730198221.837885532] [hello_moveit]: Planning failed!
[INFO] [plan_using_moveit-1]: process has finished cleanly [pid 68196]
```

However, I am encountering a new issue related to planning. The terminal output from my `demo.launch.py` file shows the following:

```
[move_group-3] [INFO] [1730198206.827680046] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-3] [INFO] [1730198206.829487784] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-3] [INFO] [1730198206.830448533] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'm2_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-3] [ERROR] [1730198221.830994111] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - m2_arm/m2_arm: Unable to sample any valid states for goal tree
[move_group-3] [ERROR] [1730198221.832325395] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - m2_arm/m2_arm: Unable to sample any valid states for goal tree
[move_group-3] [ERROR] [1730198221.836729002] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - m2_arm/m2_arm: Unable to sample any valid states for goal tree
[move_group-3] [ERROR] [1730198221.836754295] [ompl]: ./src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:252 - m2_arm/m2_arm: Unable to sample any valid states for goal tree
[move_group-3] [WARN] [1730198221.836824588] [ompl]: ./src/ompl/tools/multiplan/src/ParallelPlan.cpp:138 - ParallelPlan::solve(): Unable to find solution by any of the threads in 15.006123 seconds
[move_group-3] [INFO] [1730198221.836991920] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[move_group-3] [INFO] [1730198221.837036636] [moveit_move_group_default_capabilities.move_action_capability]: Catastrophic failure
```

The key issues I’m facing now are:
1. The planning request is aborted with a timeout, indicating that the planner cannot find a valid state.
2. The `m2_arm` configuration is unable to sample any valid states for the goal tree, resulting in a catastrophic failure.

I would greatly appreciate any guidance on how to resolve these planning issues.

Thank you for your help!

Best regards,  
Atharv Kulkarni

rhaschke, 2024-10-29:
Both issues are identical: the goal sampler cannot find an IK solution within the given timeout.
Validate that your arm can actually reach the desired Cartesian pose. Does your arm have at least 6 joints? 
Can you move the arm to the desired pose using the interactive marker in rviz?

atharv180904, 2024-10-29:
1) Yes, the position values are in the robot's reach 
2) My arm has 4 joints in total, including the gripper. (4DOF)
3) Using the RViz Interactive Marker, I am unable to move the robots end effector by moving the ball, but I am able to move the particular joints in the planning group, and my robot moves, upon hitting Plan & Execute. Once I click the Approx IK Solution tab in RViz, I am able to move the marker, but the movement is not smooth.  

Here is a video to show that: https://drive.google.com/drive/folders/197d2M5gdXvbHi9s1semeUCCovV8rj4bi?usp=drive_link

rhaschke, 2024-10-29:
Planning triggered from rviz always uses joint-space poses. Hence, you don't observe IK issues there during planning.
As your arm only has 3 DOF, you cannot reach arbitrary Cartesian poses (3 DOF position + 3 DOF orientation).
At best, you can perform [position-only IK](https://moveit.picknik.ai/humble/doc/examples/kinematics_configuration/kinematics_configuration_tutorial.html#position-only-ik).

The flag "approximate IK solution" is only available through MGI's function `setJointValueTarget`, i.e. replace `move_group_interface.setPoseTarget(target_pose);`
in your code with:
`move_group_interface.setJointValueTarget(target_pose, "", "", true);`