moveit/moveit #3662: How can I set the move_group base frame myself?

nblkStudy, 2024-10-30:
### Description

Now I have added tf_prefix to robot_state_publisher, so in order for move_group to successfully plan, I need to add this prefix to the move_group node as well or modify its base frame.

<node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" respawn="true" output="screen">
      
      <param name="tf_prefix" value="$(arg robot_namespace)" />
</node>

<node name="move_group" launch-prefix="$(arg launch_prefix)" pkg="moveit_ros_move_group" type="move_group" respawn="false" output="screen" args="$(arg command_args)">
How can I set the base frame myself?

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

nblkStudy, 2024-10-30:
<node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" respawn="true" output="screen">
      
      <param name="tf_prefix" value="$(arg robot_namespace)" />
</node>

<node name="move_group" launch-prefix="$(arg launch_prefix)" pkg="moveit_ros_move_group" type="move_group" respawn="false" output="screen" args="$(arg command_args)">

rhaschke, 2024-10-30:
MoveIt does not rely on TF frames. It directly uses the link names of the URDF. The planning frame is the root frame of the robot.

nblkStudy, 2024-10-30:
> MoveIt does not rely on TF frames. It directly uses the link names of the URDF. The planning frame is the root frame of the robot.
But it returned this to me:
[ WARN] [1730296618.201497943]: Unable to transform object from frame 'robot5/ur_arm_wrist_1_link' to planning frame 'base_link' ("base_link" passed to lookupTransform argument target_frame does not exist.)
Since I want to run multiple robots, I added the tf_prefix to their robot_state_publisher and hope that MoveIt can control them separately.

rhaschke, 2024-10-30:
You can ignore this warning. It just indicates that MoveIt cannot use those TF frames for planning (because they are not linked to the robot tree).