moveit/moveit #3669: DO NOT USE MOVEIT VERSION 1.1.16

Kuo-Feng, 2024-11-21:
### Description

The MoveIt planner does not consider collisions, and the planned path shows severe joint jumps!

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Source, 1.1.16-1focal.20241008.004520

### Steps to reproduce
1. Add some boxes or objects to the environment.
2. Observe the following issues:
- In the first video, you can see that the planned path contains severe joint jumps, yet the planner still returns a successful plan.
- In the second video, the situation is even more severe.
Links to the videos:
https://github.com/user-attachments/assets/63a5117c-a965-4b4c-8f18-b2f26177ef93
https://github.com/user-attachments/assets/12199652-55e3-4b37-ac16-b5c8400acc09

### Expected behaviour
The planner should generate a collision-free and smooth path without joint jumps.

### Actual behaviour
The planner produces a path with severe joint jumps and does not account for collisions:
https://github.com/user-attachments/assets/12199652-55e3-4b37-ac16-b5c8400acc09

rhaschke, 2024-11-21:
Thanks for providing this bug report. It would be helpful if you could reproduce those issues in a simplified setting, e.g. with the `moveit_resources_panda_moveit_config` package. Which kind of goal and path constraints did you use? Is there an older MoveIt release that doesn't exhibit the observed issue? Which one? Can you pinpoint the offending commit by back git bisection?
Provide the log output!

Kuo-Feng, 2024-11-25:
No problem, maintaining a good open-source project is everyone's responsibility.

I will attempt to reproduce the issue using the moveit_resources_panda_moveit_config package, but I am currently quite busy with work in the short term. For now, I’ll address your questions. First, I can confirm that the issue does not occur frequently, but as seen in the video, its severity when it happens cannot be ignored. I can confirm that version 1.1.16 has the issue, while version 1.1.14 (commit 9d48a84803d0994c0d4a3baf4f81aab4d13636d3) does not. I have not tested version 1.1.15.

Additionally, here is the process and planning I used:
1. Add obstacles.
2. Set the final goal position and use computeCartesianPath for planning. (moveGroupInterface.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectoryMsg);)
3. Repeat step 2 to plan again using computeCartesianPath for the final goal position. (moveGroupInterface.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectoryMsg);)
4. Combine the two paths from the above planning steps.

I suspect that the issue might be caused by the removal of the jump_threshold. Below is the method I used to combine the two trajectories:
```c++
bool Movement::combinePlans()
{
    if (movePlans.size() < 1)
    {
        return false;
    }
    else if (movePlans.size() == 1)
    {
        return true;
    }

    moveit_msgs::RobotTrajectory trajectory = movePlans[0].trajectory_;

    for (size_t i = 1; i < movePlans.size(); i++)
    {
        trajectory.joint_trajectory.points.insert(trajectory.joint_trajectory.points.end(),
                                    movePlans[i].trajectory_.joint_trajectory.points.begin() + 1,
                                    movePlans[i].trajectory_.joint_trajectory.points.end());
    }

    robot_trajectory::RobotTrajectory rt(moveit.mgInterface().getRobotModel(), "arm");
    rt.setRobotTrajectoryMsg(*moveit.mgInterface().getCurrentState(), trajectory);

    // Create an instance of the IterativeParabolicTimeParameterization class
    trajectory_processing::IterativeParabolicTimeParameterization iptp;
    if (iptp.computeTimeStamps(rt, 1.0, 1.0))
    {
        rt.getRobotTrajectoryMsg(trajectory);

        moveit::planning_interface::MoveGroupInterface::Plan plan;
        plan.trajectory_ = trajectory;

        movePlans.clear();
        movePlans.emplace_back(plan);
        return true;
    }

    return false;
}
```

P.S. Apologies for using the "DO NOT USE" title. My main intention was to highlight the severity of this issue and help others avoid encountering the same serious phenomenon.

rhaschke, 2024-12-01:
As pointed out in #3618, the new implementation of `computeCartesianPath` ignores obstacles if the MaxEEFStep is too large (and jumps "over" the obstacle. However, your videos don't look like this.
Instead of manually planning and combining individual Cartesian trajectories, why don't you use the corresponding function:
https://github.com/moveit/moveit/blob/675495426dcceb18453ab0a689f5fd8319248998/moveit_core/robot_state/include/moveit/robot_state/cartesian_interpolator.h#L164-L176

I'm still very curious to see this issue reproduced in a standard setup. If this is really an issue in the current release, it should get fixed as soon as possible. Can you please provide a minimal example?

rhaschke, 2024-12-09:
@Kuo-Feng: friendly ping

Kuo-Feng, 2024-12-09:
> [@Kuo-Feng](https://github.com/Kuo-Feng): friendly ping

Apologies for the past few weeks of intense work, which have kept me busy. I will upload a runnable program that can reproduce the error here within this week.

Kuo-Feng, 2024-12-10:
@rhaschke  : update
Since yesterday, I have been testing the previous configurations and executable files (which previously encountered errors), but I have not been able to reproduce the issue. In fact, the problem we face does not occur frequently. For example, one machine ran for a month without any errors before the issue occurred suddenly one day, while another machine experiences the problem about once a week. It seems I was relatively "lucky" during my last debugging session—I happened to pinpoint the exact posture and planning that caused the error. However, this time, even with the same configuration, such as identical object positions and the same executable file, the issue cannot be reproduced. Currently, the conditions for reproducing the error remain unclear. Tomorrow, I will introduce some random variables into the testing environment to ensure the planned paths are not identical, and then check if I can once again identify a path with excessively large angular changes.

rhaschke, 2024-12-10:
The issue being hard to reproduce is (partially) good news.  On the other hand, being it non-deterministic is surprising to me. The Cartesian path generation should be rather deterministic. Which IK solver do you use? How many DoFs does your arm have?

Kuo-Feng, 2024-12-11:
Yes, the fact that it is hard to reproduce might be somewhat good news, but it is still dangerous when it happens, especially during obstacle avoidance. The difficulty in reproducing the issue is also why I feel puzzled when it occurs. I know it might be related to the version of MoveIt, but since the issue occurs so infrequently, I initially thought it was due to errors in my target points or waypoint settings. However, the program has been running for a year without such errors, which makes me suspect it is related to MoveIt. During the last investigation, I discovered the issue as previously recorded videos. Moreover, the computeCartesianPath function I called in version 1.1.16 is marked as deprecated. Here is the code I used:

```cpp
fraction = moveGroupInterface.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectoryMsg);
```

The **kinematic solver** information I am using is as follows:
```yaml
arm:
  kinematics_solver: kdl_kinematics_plugin/KDLKinematicsPlugin
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005
```
The robotic arm has **6 degrees of freedom**.

rhaschke, 2024-12-11:
Of course, in the long run, you should migrate from the deprecated to the new function - just drop the `jump_threshold` argument.
However, for now, the deprecated function just calls the new one. I can't explain the behavior yet - for this reason, I'm very curious to see a reproducible example ;-)
However, actually MoveIt 1.1.16 should have exactly fixed issues like the ones you have observed!