moveit/moveit #3674: MoveIt mesh_filter only adds MESH collision objects, ignores shapes from primitives

micat001, 2024-12-11:
**Is your feature request related to a problem? Please describe.**
Adding self-filtering for your robot using `mesh_filter` only removes links that use a mesh object for collision.
See https://github.com/moveit/moveit/blob/675495426dcceb18453ab0a689f5fd8319248998/moveit_ros/perception/mesh_filter/src/depth_self_filter_nodelet.cpp#L192

**Describe the solution you'd like**
`mesh_filter` should convert shape primitives to MESH and remove them from the depth image. The tutorial should also be updated to run (the `ur5_description` launch file is out-of-date).

**Describe alternatives you've considered**
Other 3rd party libraries exist (such as [realtime_urdf_filter](https://github.com/blodow/realtime_urdf_filter) or [robot_body_filter](https://github.com/peci1/robot_body_filter)) but are slow and/or require additional dependencies. 

**Additional context**
Here's our robot (Bravo 7 from [Reach Robotics](https://quote.reachrobotics.com/product/bravo-manipulators/bravo-7/) that has only the gripper modeled using a STL file - all other collisions are primitive shapes:

[image]

And here's our robot when I remodel all the collision to use STL files:
[image]

micat001, 2024-12-11:
Draft PR that addresses the issue is here: https://github.com/moveit/moveit/pull/3675

rhaschke, 2025-01-07:
Closed via #3675.

closed 2025-01-07 by rhaschke