moveit/moveit #3679: ABORTED: TIMED_OUT when i run hardware

Mohammad-Khaliafah, 2024-12-26:
Hello everyone,

I'm encountering an issue with my robotic system and would appreciate your insights.

Problem Description:

When launching the hardware, I receive the following error messages:

[ERROR] [1735031784.794243270]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 4.192990 seconds). Stopping trajectory.
[ INFO] [1735031784.794457591]: Cancelling execution for /frrobot/fr5v6g_arm_controller
[ INFO] [1735031784.794697687]: Completed trajectory execution with status TIMED_OUT ...
[ INFO] [1735031784.795971630]: Controller '/frrobot/fr5v6g_arm_controller' successfully finished
[ INFO] [1735031784.827509122]: ABORTED: TIMED_OUT

[frrobot/frrobot_hardware_interface-5] process has died [pid 117910, exit code -6, cmd /home/mk/frcobot_ck/devel/lib/ros_control_boilerplate/frrobot_hw_main __name:=frrobot_hardware_interface __log:=/home/mk/.ros/log/efc449ae-c1d7-11ef-aab0-a10d75fade99/frrobot-frrobot_hardware_interface-5.log].
log file: /home/mk/.ros/log/efc449ae-c1d7-11ef-aab0-a10d75fade99/frrobot-frrobot_hardware_interface-5*.log
System Configuration:

Control Loop Settings:

loop_hz: 125
cycle_time_error_threshold: 0.01
Hardware Interface:

Joints: j1 to j6, gripper_robotiq_85_left_knuckle_joint
sim_control_mode: 0 (position control)
Joint State Controller:

publish_rate: 50
Arm Controller (/frrobot/fr5v6g_arm_controller):

Type: position_controllers/JointTrajectoryController
Joints: j1 to j6
Constraints:
goal_time: 3.0
Individual joint trajectory: 0.60
Individual joint goal: 0.15
Individual Joint Position Controllers:

Type: position_controllers/JointPositionController
PID: {p: 100.0, i: 0.01, d: 10.0}
Gripper Controller (/frrobot/gripper_controller):

Type: position_controllers/JointTrajectoryController
Joint: gripper_robotiq_85_left_knuckle_joint
Constraints:
goal_time: 3.0
Trajectory: 0.60
Goal: 0.15
Observations:

The system reports that the controller is taking too long to execute the trajectory, exceeding the expected duration and resulting in a timeout.
The hardware interface process terminates with exit code -6 (SIGABRT), indicating an abort signal, possibly due to an internal error or assertion failure.
Steps Taken:

Verified the controller configurations and constraints.
Ensured that the hardware interface is correctly set up.
Monitored system resources to rule out performance bottlenecks.
Request for Assistance:

ROS Distribution: [ Noetic]
Operating System: [Ubuntu 20.04]
Robot Model: FR5V6

Has anyone encountered similar issues with trajectory execution timeouts and process terminations in ROS control systems? I would appreciate any guidance on potential causes or solutions to this problem.

Thank you in advance for your help.

rhaschke, 2024-12-28:
Controller timeouts are typically resulting from controllers not matching the max velocity/acceleration limits used during planning. If your controller aborts, you should first fix that issue. Obviously, this can cause timeouts at the higher levels as well.

v4hn, 2025-01-06:
```
[frrobot/frrobot_hardware_interface-5] process has died [pid 117910, exit code -6, cmd /home/mk/frcobot_ck/devel/lib/ros_control_boilerplate/frrobot_hw_main __name:=frrobot_hardware_interface __log:=/home/mk/.ros/log/efc449ae-c1d7-11ef-aab0-a10d75fade99/frrobot-frrobot_hardware_interface-5.log].
```
> The hardware interface process terminates with exit code -6 (SIGABRT), indicating an abort signal, possibly due to an internal error or assertion failure.

These clearly indicate that the error is not in MoveIt, as Robert suggested.
MoveIt even seems to successfully generate a trajectory and send it to the controller.
But the controller crashes when trying to execute that trajectory.
First step:
check the `goal` topic of your `FollowJointTrajectory` action server when the crash happens.
The message sent there is the only information MoveIt communicates to your controller.
You can probably even record it in a rosbag and replay it to test your failure.

Simplification: Try to create a simple (1-2 waypoints) trajectory yourself that your robot can perform and send it to the controller. If that also triggers the error you have a good lead on how to fix it.

> The system reports that the controller is taking too long to execute the trajectory, exceeding the expected duration and resulting in a timeout.

If your controller crashes, of course MoveIt will wait for it to report success until it believes (because too much time passed) that something bad happened because it didn't hear back.

closed 2025-01-06 by v4hn