moveit/moveit #3681: DXF File to XYZ Coordinate

Mohammad-Khaliafah, 2025-01-04:
Hello everyone.
How do you convert a dxf or SVG file to XYZ coordinate and give them to Moveit to make a continuous motion, Is there any pkg or tutorial to do it?

v4hn, 2025-01-06:
This is straight-forward using any of the CartesianPath interfaces [[0](https://github.com/moveit/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/cartesian_interpolator.h),[1](https://github.com/moveit/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/cartesian_interpolator.h), [2](https://github.com/moveit/moveit/blob/ba704d59921578b20fea676dfaa2a7d8bb475004/moveit_ros/move_group/src/default_capabilities/cartesian_path_service_capability.cpp#L74), [4](https://github.com/moveit/moveit/blob/master/moveit_setup_assistant/templates/moveit_config_pkg_template/launch/pilz_industrial_motion_planner_planning_pipeline.launch.xml)], but there is no tutorial for it.

It would be interesting to add one to the tutorials, e.g., with a demo panda robot drawing a simple image, if you are willing to contribute it.

Otherwise closing as not related to the framework itself.

closed 2025-01-06 by v4hn

Mohammad-Khaliafah, 2025-01-07:
Thank you for your response. I appreciate the clarification and the suggestion. I would be happy to contribute a tutorial demonstrating how to draw a simple image after completing my project.

> This is straight-forward using any of the CartesianPath interfaces [...]

Thank you for your response. I appreciate the clarification and the suggestion. After completing my project, I would be happy to contribute a tutorial demonstrating how to draw a simple image.