moveit/moveit #3686: How to Control FR5V6 Robot Arm with ROS Noetic, MoveIt, and TCP/IP?

Mohammad-Khaliafah, 2025-01-13:
I am using the [[frcobot_ros](https://github.com/FAIR-INNOVATION/frcobot_ros)](https://github.com/FAIR-INNOVATION/frcobot_ros) package to control the robot. However, when I run the **fr5v6 demo hardware**, I encounter an error. As a result, I am considering controlling the robot from scratch using **TCP/IP** as described below.

I have the **FR5V6 robot arm** and its **URDF file**, and I want to control it using **ROS Noetic**, **MoveIt**, and the **TCP/IP protocol** to communicate with its controller. The protocol specifies commands to control the robot and feedback for its state (e.g., joint states, errors).

### Objectives:
1. **Create a TCP communication interface in ROS** to send commands and receive feedback.
2. **Parse state feedback from the robot**, such as joint states and errors, and publish it on appropriate ROS topics.
3. **Integrate the TCP interface with MoveIt** to enable trajectory execution and planning.
4. Ensure **real-time communication** and debugging capabilities.

### Assumptions:
- I have already installed **ROS Noetic**, **MoveIt**, and the robot's **URDF** file.
- I am familiar with basic ROS concepts but need guidance on creating and integrating the communication layer.

### Requirements:
- Provide all necessary ROS packages, libraries, and dependencies required for the implementation.
- Include **example code snippets** (Python or C++) to:
  - Implement the TCP communication node.
  - Publish robot state feedback to ROS topics.
  - Execute trajectories planned in MoveIt using the TCP interface.
- Suggest debugging tools or techniques to ensure real-time performance.

### Additional Information:
- **Robot controller feedback details:** [[Google Drive Link](https://drive.google.com/drive/folders/1fc-LgAn3hU9eat1aJMMga4VfzB4la1fi?usp=sharing)](https://drive.google.com/drive/folders/1fc-LgAn3hU9eat1aJMMga4VfzB4la1fi?usp=sharing)

Any help, examples, or guidance would be greatly appreciated!

rhaschke, 2025-01-13:
This is nothing related to MoveIt. Instead of developing a controller from scratch, I suggest to fix the bugs in the existing hardware interface!
Regarding the PATH_TOLERANCE_VIOLATED error, there are [numerous articles](https://www.google.com/search?q=Moveit+PATH_TOLERANCE_VIOLATED&oq=Moveit+PATH_TOLERANCE_VIOLATED&gs_lcrp=EgZjaHJvbWUyBggAEEUYOTIJCAEQIRgKGKAB0gEIMjMwOWowajeoAgCwAgA) on the web explaining how to fix that.

closed 2025-01-13 by rhaschke

Mohammad-Khaliafah, 2025-01-13:
> This is nothing related to MoveIt. Instead of developing a controller from scratch, I suggest to fix the bugs in the existing hardware interface! Regarding the PATH_TOLERANCE_VIOLATED error, there [...]

Could this issue be related to PID tuning? The robot moves in a zigzag manner in real motion, without any smooth acceleration or deceleration. If this is the case, how can I tune the PID controllers to resolve this? You can find all the controller details in the package description. The code used for Gazebo simulation corresponds to the real controller in the fr5v6_moveit_config package, and the controller for hardware is located in frrobot_control

rhaschke, 2025-01-13:
You better ask for controller tuning help at the manufacturer's site.
Again there are numerous resources [available on the internet](https://www.google.com/search?q=ros+pid+controller+tuning&oq=ros+pid+controller+tuning&gs_lcrp=EgZjaHJvbWUyBggAEEUYOTIHCAEQIRigATIHCAIQIRigATIHCAMQIRigAdIBCDUxMzRqMGo3qAIAsAIA) as well.