moveit/moveit #3691: regression due to updated locking of PSM's scene updates

rhaschke, 2025-01-25:
Looks like, #3682 breaks rviz' PSM: I observe flaky but frequent crashes doing:
`roslaunch panda_moveit_config demo.launch`
Then, in the rviz MotionPlanning plugin, create a collision object and move the robot. 
This crashes when doing collision checks against the new object:

<details>
 <summary>backtrace</summary>
<pre>
#10   Source "/vol/sandbox/ros/src/moveit/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_display.cpp", line 728, in changedQueryGoalState [0x774a82b93dd7]
        725:     return;
        726:   setStatusTextColor(query_goal_color_property_->getColor());
        727:   addStatusText("Changed goal state");
      > 728:   drawQueryGoalState();
        729:   addBackgroundJob([this] { publishInteractiveMarkers(true); }, "publishInteractiveMarkers");
        730: }
#9  | Source "/vol/sandbox/ros/src/moveit/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_display.cpp", line 748, in getCollidingLinks
    |   746:       // update link colors
    |   747:       std::vector<std::string> collision_links;
    | > 748:       getPlanningSceneRO()->getCollidingLinks(collision_links, *state);
    |   749:       status_links_goal_.clear();
    |   750:       for (const std::string& collision_link : collision_links)
      Source "/vol/sandbox/ros/src/moveit/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h", line 616, in drawQueryGoalState [0x774a82b935c9]
        613:   /** \brief Get the names of the links that are involved in collisions for the state \e robot_state */
        614:   void getCollidingLinks(std::vector<std::string>& links, const moveit::core::RobotState& robot_state) const
        615:   {
      > 616:     getCollidingLinks(links, robot_state, getAllowedCollisionMatrix());
        617:   }
        618: 
        619:   /** \brief  Get the names of the links that are involved in collisions for the state \e robot_state given the
#8    Source "/vol/sandbox/ros/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp", line 564, in getCollidingLinks [0x774a8c0a46e3]
        561:                                       const collision_detection::AllowedCollisionMatrix& acm) const
        562: {
        563:   collision_detection::CollisionResult::ContactMap contacts;
      > 564:   getCollidingPairs(contacts, robot_state, acm);
        565:   links.clear();
        566:   for (collision_detection::CollisionResult::ContactMap::const_iterator it = contacts.begin(); it != contacts.end();
        567:        ++it)
#7    Source "/vol/sandbox/ros/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp", line 548, in getCollidingPairs [0x774a8c0a0b5f]
        545:   req.max_contacts_per_pair = 1;
        546:   req.group_name = group_name;
        547:   collision_detection::CollisionResult res;
      > 548:   checkCollision(req, res, robot_state, acm);
        549:   res.contacts.swap(contacts);
        550: }
#6    Source "/vol/sandbox/ros/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp", line 498, in checkCollision [0x774a8c0a072e]
        495:                                    const collision_detection::AllowedCollisionMatrix& acm) const
        496: {
        497:   // check collision with the world using the padded version
      > 498:   getCollisionEnv()->checkRobotCollision(req, res, robot_state, acm);
        499: 
        500:   // do self-collision checking with the unpadded version of the robot
        501:   if (!res.collision || (req.contacts && res.contacts.size() < req.max_contacts))
#5    Source "/vol/sandbox/ros/src/moveit/moveit_core/collision_detection_fcl/src/collision_env_fcl.cpp", line 322, in checkRobotCollisionHelper [0x774a82a4e2e1]
        319:   CollisionData cd(&req, &res, acm);
        320:   cd.enableGroup(getRobotModel());
        321:   for (std::size_t i = 0; !cd.done_ && i < fcl_obj.collision_objects_.size(); ++i)
      > 322:     manager_->collide(fcl_obj.collision_objects_[i].get(), &cd, &collisionCallback);
        323: 
        324:   if (req.distance)
        325:   {
#4    Source "/vol/sandbox/ros/src/moveit/moveit_core/collision_detection_fcl/src/collision_common.cpp", line 92, in collisionCallback [0x774a82a397c1]
         89:   if (cdata->acm_)
         90:   {
         91:     AllowedCollision::Type type;
      >  92:     bool found = cdata->acm_->getAllowedCollision(cd1->getID(), cd2->getID(), type);
         93:     if (found)
         94:     {
         95:       // if we have an entry in the collision matrix, we read it</pre>
</details>

The `std::string` reference returned by `cd[12]->getID()` is invalid (0x18).
However, I don't yet understand why and I won't have time to dive into that deeper soon. 
@v4hn: do you have a chance to look into this next week?

git bisecting the PR revealed that already the very first commit https://github.com/moveit/moveit/pull/3682/commits/dea261bed005c0f5e5a6ba1ebbee1048bd00e023 is causing the issues.
Reverting the whole PR, but keeping #3676, didn't yet exhibit the crash.

Ideally, one should continue with `asan`. However rviz didn't even start with asan for me.

rhaschke, 2025-01-26:
Rebuilding also rviz with asan, I've got the following result:
<details>
 <summary>asan analysis</summary>
<pre>
==271565==ERROR: AddressSanitizer: heap-use-after-free on address 0x6130000b2f58 at pc 0x77828f5212bc bp 0x7ffe265042b0 sp 0x7ffe265042a0
READ of size 8 at 0x6130000b2f58 thread T0
    #0 0x77828f5212bb in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::size() const /usr/include/c++/11/bits/basic_string.h:921
    #1 0x77828f5212bb in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const /usr/include/c++/11/bits/basic_string.h:2876
    #2 0x77828f5212bb in bool std::operator< <char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/basic_string.h:6343
    #3 0x77828f5212bb in std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const /usr/include/c++/11/bits/stl_function.h:400
    #4 0x77828f5212bb in std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, collision_detection::AllowedCollision::Type, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, collision_detection::AllowedCollision::Type> > > >, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, collision_detection::AllowedCollision::Type, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, collision_detection::AllowedCollision::Type> > > > >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, collision_detection::AllowedCollision::Type, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, collision_detection::AllowedCollision::Type> > > > > >::_M_lower_bound(std::_Rb_tree_node<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, collision_detection::AllowedCollision::Type, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, collision_detection::AllowedCollision::Type> > > > > const*, std::_Rb_tree_node_base const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const /usr/include/c++/11/bits/stl_tree.h:1921
    #5 0x77828f5212bb in std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, collision_detection::AllowedCollision::Type, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, collision_detection::AllowedCollision::Type> > > >, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, collision_detection::AllowedCollision::Type, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, collision_detection::AllowedCollision::Type> > > > >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, collision_detection::AllowedCollision::Type, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, collision_detection::AllowedCollision: [... cut ...]
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==271565==ABORTING
</pre>
</details>

Looks like the world object acquired in `MotionPlanningFrame::updateCollisionObjectPose`:
https://github.com/moveit/moveit/blob/b7ac2f52562ef6a4dfe42d5a591500e6d19830b8/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_objects.cpp#L359

is eventually destroyed, when the function is left. But why?

rhaschke, 2025-01-29:
@v4hn: How should we proceed with this issue? If nobody of us can debug this on a short time scale (I can't), we should revert the commit for now.

v4hn, 2025-01-29:
> However, this cannot be the reason for the failure as that worked before for years.

I believe that *is* the issue at hand, though maybe exposed through a different one with the changed locking.

> The question is: why does the CollisionEnv of the PS still has an ObjectPtr to the already deleted instance?

Because `CollisionEnvFCL::notifyObjectChange(const ObjectConstPtr& obj, World::Action action)` assumes the world object pointer remains valid on `action==MOVE`.
[In this case `updateFCLObject` is avoided and only the fcl object transforms are updated](https://github.com/moveit/moveit/blob/d6eebbdfb582f64a0d4265d41ef6b06328d04834/moveit_core/collision_detection_fcl/src/collision_env_fcl.cpp#L437-L474).
The payload pointer remains untouched.

Your analysis above nicely documents a design flaw in `World`.
`getObject` returns `shared_ptr`s for anyone to use, `setObjectPose` and other methods use `ensureUnique` to ensure full ownership over the object (from other scenes), and the FCL env assumes that only the transforms changed on `MOVE`.

potential ways to address this:
- drop the `ensureUnique` and try to catch all side-effects in the code base
  - do not share object ownership at all, but return a reference in `getObject` if it should be owned by the world.
- assert uniqueness on call
- change the fcl code to update the whole object
  - maybe just the user pointers?
  - the bullet env does not have this extra conditional
- at least release the object pointer in the call you found in the rviz plugin because it is not needed there to begin with

As the last option is free of further side-effects and should be simplified anyway, [here's a PR for it](https://github.com/moveit/moveit/pull/3692).
Please check how many other critical problems you can find after that...
I really do not want to spend the time on any of the other things in the list right now.

closed 2025-01-29 by rhaschke