moveit/moveit #3695: Dofbot: Unable to sample any valid states for goal tree

sathibault, 2025-02-06:
### Description

The [Dofbot](https://category.yahboom.net/products/dofbot-jetson_nano) is a 4 DOF arm.
Using the MoveGroupCommander, I receive this error almost all the time, but not every time using the same code and target:
Unable to sample any valid states for goal tree

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build? binary
* If binary, which release version? 1.0.9
* If source, which branch?

### Steps to reproduce
The script has 2 steps:
1. Move to home position with specific joint state using go(joints=...)
2. Move to a target position 1cm away using set_pos_target and go.

### Expected behaviour
Should get a successful plan

### Actual behaviour
Fails to make a plan

### Backtrace or Console output

rhaschke, 2025-02-06:
A 4DoF robot cannot reach all 6DoF Cartesian poses. You should constrain your IK solver (KDL) to position-only if you don't care about orientation.

sathibault, 2025-02-18:
Got it, it is working now!  I'm not sure why they don't preship it that way.

closed 2025-02-18 by sathibault