moveit/moveit #3700: Manipulator end-effector occasionally collides with robot cell structure

bryangd34, 2025-03-31:
### Description

We have an ABB 4600 robot arm with custom made end-effector attached. A MoveIt package was created with collision geometries included.

1.) If we send a target for which the end-effector would collide with the robot cell structure, Moveit correctly does not plan a path and reports back that the end position would result in a collision.

2.) If we send a target for which the end-effector is very near to the robot cell structure, Moveit plans a path and correctly avoids the structure. One can observe that MoveIt has created a path that very deliberately avoids the structure, as the end-effector moves _around_ the obstacle.

In light of the two points, we assumed the collision-avoidance has been correctly set up. Specifically point (2) was repeated many times.

But, during testing, whereby path-planning is repeatedly done using `set_pose_target` and `plan`, every _now and then_ the end-effector collides with the robot cell structure. I would estimate once every 30 times. 

Some more information:
- The structure is static, that is, it does not move or change. The base of the robot is also static relative to the robot cell structure.
- The structure has already been "beefed" up in the URDF. That is, it is considerably thicker in the URDF than in the real structure.
- All meshes are STLs. 
- The collision is not just in reality, but also in RVIz (see screenshot below).

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Binary install. Version: 1.1.16-1focal.20250213.212131

### Steps to reproduce
Unfortunately, the reality is it is very difficult to reliably reproduce. We might spend a whole day testing and it'll happen only once. But when it does, it is catastrophic: the end-effector collides with the structure.

### Expected behaviour
MoveIt plans paths that avoid the end-effector colliding with the robot structure, EVERY TIME.

### Actual behaviour
MoveIt occasionally plans a path that results in the end-effector colliding with the robot structure. 

### Backtrace or Console output

A screenshot of RViz showing the end-effector in collision with the robot structure. It is a highly cropped image, due to project confidentiality.

[image]

rhaschke, 2025-03-31:
I understand that you have a test procedure that reliably produces the collision in simulation (1/30 trials).
Can you please provide that test code? Thanks.

bryangd34, 2025-03-31:
Hi @rhaschke 
Thanks for your reply.

This is in fact happening on the real hardware, not simulation. 
The 1 in 30, is just an estimate, the issue has not occurred a great deal of times (which makes it all the more mysterious). I cannot run a huge amount of cycles, as it is with real hardware.

I did run numerous tests in simulation (RobotStudio + MoveIt) and did not observe the issue, but also cannot be sure, as most times I would run many cycles overnight and thus I am not constantly observing RViz for this issue. I should add, there does not appear to be any error reported. I may need to devise a way of "catching" this issue in simulation. This way I could run numerous cycles overnight and try to see it. If you have any ideas has to how one could catch this issue, please let me know.

I only have the full code for the entire application itself. I do not have a subset of the code which can then be used to repeat the error. But very briefly, the process is as follows:

`move_group.set_named_target('starting_pose')`
`plan_success, plan = move_group.plan()`
`move_group.execute(plan, True)`

_Get new target pose_

`move_group.set_pose_target(target_pose)`
`success, plan = move_group.plan()`
`move_group.execute(plan, wait=True)`

_Go back to start_

`move_group.set_named_target('starting_pose')`
`plan_success, plan = move_group.plan()`
`move_group.execute(plan, True)`

_Get new target pose_

(and so on)

The collision happens when going from the starting pose to the target.

rhaschke, 2025-03-31:
If you observe this issue only in real, not in sim, I'm afraid your robot controller is not correctly tracking the planned trajectory. This is not an issue of MoveIt then, but of the robot controller. 
You can partially protect against such issues, by limiting the allowed tracking error in your ros2 control tolerance settings.
Looks like you relaxed those settings to avoid controller cancellation, did you?

https://moveit.picknik.ai/main/doc/examples/controller_configuration/controller_configuration_tutorial.html#optional-allowed-trajectory-execution-duration-parameters
https://github.com/ros-controls/control_msgs/blob/master/control_msgs/msg/JointTolerance.msg

bryangd34, 2025-03-31:
Yes, you are correct, I did relax those tolerance settings. 
Namely `allowed_execution_duration_scaling` and `allowed_goal_duration_margin`. However, I thought those were related specifically to the _time_ to execute and not tolerances to the path that is tracked. Is that correct?

I will have a look through everything though to see if there were any relaxations made to the path tracking itself. JointTolerance.msg I did not use, but there might be something else.

Thanks a lot for these suggestions. I think you are right that the issue is related to the robot controller and not MoveIt.

bryangd34, 2025-03-31:
FYI: I am using [the new abb robot driver](https://github.com/ros-industrial/abb_robot_driver)

rhaschke, 2025-03-31:
> Namely `allowed_execution_duration_scaling` and `allowed_goal_duration_margin`. However, I thought those were related specifically to the _time_ to execute and not tolerances to the path that is tracked. Is that correct?

Yes that's correct. These parameters should be unrelated to the problem.
However, MoveIt allows to specify path tolerances and probably uses `JointTolerance.msg` to pass them to the controller.

bryangd34, 2025-04-01:
Okay thanks. I will look around for any path tolerances that may be too relaxed and let you know what I find.
I would appreciate it if we could keep this issue open for the time being, thanks.

bryangd34, 2025-04-08:
Hi @rhaschke 

I have had a look through the code base I cannot see anything that is setting the a path tolerance.
My controllers.yaml:
```
joint_position_trajectory_controller:
  type: position_controllers/JointTrajectoryController
  joints: [joint_1, joint_2, joint_3, joint_4, joint_5, joint_6]
  constraints:
    goal_time: 50.0
```

Also, I read in FollowJointTrajectory.action (which is using the `JointTolerance.msg`):

> # Tolerances for the trajectory.  If the measured joint values fall
> # outside the tolerances the trajectory goal is aborted.  Any
> # tolerances that are not specified (by being omitted or set to 0) are
> # set to the defaults for the action server (often taken from the
> # parameter server).

On the parameter server, I only see `goal_joint_tolerance`, `goal_position_tolerance` and `goal_orientation_tolerance`. So nothing related to the _path_. Incidentally, these goal tolerances are all very small (e.g. 0.0001 and 0.001 for orientation).

Do you have any other suggestions of things I could check? 
Is this perhaps more of an [abb robot driver](https://github.com/ros-industrial/abb_robot_driver) issue?

Thanks.

rhaschke, 2025-04-08:
Looking into JointTrajectoryController's source, it turns out that [path tolerances are zero by default](https://github.com/ros-controls/ros_controllers/blob/c2348e85abf35cf33cf654a84d61db206abe9a73/joint_trajectory_controller/include/joint_trajectory_controller/tolerances.h#L305-L306):
```nh.param(joint_names[i] + "/trajectory", tolerances.state_tolerance[i].position,      0.0);```

meaning that [all path errors are ignored](https://github.com/ros-controls/ros_controllers/blob/c2348e85abf35cf33cf654a84d61db206abe9a73/joint_trajectory_controller/include/joint_trajectory_controller/tolerances.h#L175-L178):
```const bool is_valid = !(tol.position     > 0.0 && abs(state_error.position[i])     > tol.position)```

simonschmeisser, 2025-04-08:
Just to clarify, these parameters do not affect the planning but instruct the driver to react to path deviations during execution. Right? Which would like be the main issue, your driver not following the planned path close enough.

rhaschke, 2025-04-08:
Yes, exactly.

bryangd34, 2025-04-08:
@rhaschke 
Thanks a lot. That looks very likely to be the source of our problem. 
I'll explicitly set limits.

closed 2025-04-08 by bryangd34

rhaschke, 2025-04-08:
As mentioned by @simonschmeisser, the actual source of your problem is that the robot controller cannot correctly track the given trajectory. Additionally, the safeguard mechanism is disabled.

bryangd34, 2025-04-08:
Yes, understood. Can you elaborate on what you mean by "safeguard mechanism" please?

rhaschke, 2025-04-08:
The safeguard mechanism of the joint trajectory controller, aborting the trajectory execution on too strong deviations, is disabled by default (if no path tolerances are specified).